rockchip-linux/kernel #141: CIF_MIPI errors

teseo-sw, 2019-01-16:
When trying to capture from camera with:
`$ v4l2-ctl --set-fmt-video=width=2112,height=1568,pixelformat=NV12 --stream-mmap --stream-count=1000 -d /dev/video3`

I get a lot of messages like these:

```
[   25.768683] cif_isp10_mipi_isr WARN: CIF_MIPI_ERR_CSI: 0x200002
[   25.768684] cif_isp10_mipi_isr WARN: CIF_MIPI_SYNC_FIFO_OVFLW: 0x200002
[   25.768686] cif_isp10_mipi_isr(6867) ERR: mipi_mis icr err: 0x2
[   25.768692] cif_isp10_mipi_isr WARN: CIF_MIPI_ERR_CSI: 0x200002
[   25.768693] cif_isp10_mipi_isr WARN: CIF_MIPI_SYNC_FIFO_OVFLW: 0x200002
[   25.775235] cif_isp10_mipi_isr WARN: CIF_MIPI_ERR_CSI: 0x1200002
[   25.775236] cif_isp10_mipi_isr WARN: CIF_MIPI_SYNC_FIFO_OVFLW: 0x1200002
[   25.775238] cif_isp10_mipi_isr(6867) ERR: mipi_mis icr err: 0x2
[   25.775244] cif_isp10_mipi_isr WARN: CIF_MIPI_ERR_CSI: 0x200002
[   25.775245] cif_isp10_mipi_isr WARN: CIF_MIPI_SYNC_FIFO_OVFLW: 0x200002
[   25.775247] cif_isp10_mipi_isr(6867) ERR: mipi_mis icr err: 0x2
[   25.775254] cif_isp10_mipi_isr WARN: CIF_MIPI_SYNC_FIFO_OVFLW: 0x2
[   25.775261] cif_isp10_mipi_isr WARN: CIF_MIPI_ERR_CSI: 0x200002
[   25.775262] cif_isp10_mipi_isr WARN: CIF_MIPI_SYNC_FIFO_OVFLW: 0x200002
[   25.775264] cif_isp10_mipi_isr(6867) ERR: mipi_mis icr err: 0x2
[   25.775270] cif_isp10_mipi_isr WARN: CIF_MIPI_SYNC_FIFO_OVFLW: 0x2
[   25.775276] cif_isp10_mipi_isr WARN: CIF_MIPI_ERR_CSI: 0x200002
```

I'm using a OV13850 sensor with the latest kernel (4.4.154) on a custom board.

sunny0917, 2019-04-03:
我也遇到了这个问题，不知道怎么解决，老感觉clk不对，或者硬件有问题

teseo-sw, 2019-04-08:
We managed to somewhat reduce the number of MIPI errors using a MIPI DPHY retimer (SN75DPHY440SS). Still, we get errors occasionally. Any hints @rkhuangtao, @zengyiqing, @wenleo, @Caesar-github ?

rkchrome, 2019-04-17:
Please check the isp clk

teseo-sw, 2019-04-19:
@rkchrome What should I check exactly? Here is a complete dmesg for a single frame capture on an OV13850:
```
$ v4l2-ctl --set-fmt-video=width=2112,height=1568,pixelformat=NV12 --stream-mmap --stream-count=1 -d /dev/video3
[  211.152174] cif_isp10_v4l2_open: video device video3.3 (rkisp10_mainpath)
[  211.152269] cif_isp10_init: 0x00000002
[  211.152331] ov13850.ov_camera_module_ioctl: cmd: 0x1
[  211.152380] ov13850.pltfrm_camera_module_ioctl: cmd: 0x1
[  211.152448] cif_isp10_s_input: setting input to CSI, [w-h]=[2112-1568], defrect=[0-0-2112-1568]
[  211.152494] ov13850.ov_camera_module_ioctl: cmd: 0x1
[  211.152536] ov13850.pltfrm_camera_module_ioctl: cmd: 0x1
[  211.152587] ov13850.ov_camera_module_enum_frameintervals: 0
[  211.152630] ov13850.ov_camera_module_ioctl: cmd: 0x2
[  211.152670] ov13850.pltfrm_camera_module_ioctl: cmd: 0x2
[  211.152716] ov13850.ov_camera_module_enum_frameintervals: 1
[  211.152758] ov13850.ov_camera_module_ioctl: cmd: 0x2
[  211.152798] ov13850.pltfrm_camera_module_ioctl: cmd: 0x2
[  211.152841] ov13850.ov_camera_module_enum_frameintervals: 2
[  211.152915] cif_isp10_img_src_select_strm_fmt: requesting format BAYER BGGR10 2112x1568(0,0,2112,1568)@30/1fps from ov13850
[  211.152972] ov13850.ov_camera_module_s_fmt: 2112x1568, fmt code 0x3007
[  211.153024] ov13850.ov_camera_module_find_config: 2112x1568, fmt code 0x3007
[  211.153070] ov13850.ov_camera_module_find_config: found matching config 2112x1568_30fps
[  211.153119] ov13850.ov_camera_module_find_config: 2112x1568, fmt code 0x3007
[  211.153166] ov13850.ov_camera_module_find_config: frame interval 1/30
[  211.153211] ov13850.ov_camera_module_find_config: found matching config 2112x1568_30fps
[  211.153251] ov13850.ov_camera_module_set_active_config: 
[  211.153304] ov13850.ov_camera_module_s_frame_interval: 1/30 (30fps)
[  211.153354] ov13850.ov_camera_module_find_config: 2112x1568, fmt code 0x3007
[  211.153401] ov13850.ov_camera_module_find_config: frame interval 1/30
[  211.153447] ov13850.ov_camera_module_find_config: found matching config 2112x1568_30fps
[  211.153497] ov13850.ov_camera_module_g_ctrl:  id 0x981981
[  211.153542] ov13850.ov_camera_module_g_ctrl: RK_V4L2_CID_VBLANKING 5000
[  211.154182] cif_isp10_v4l2_s_fmt: VIDEO_CAPTURE
[  211.154235] cif_isp10_s_fmt: MP
[  211.154300] cif_isp10_s_fmt_mp: YUV420SP 2112x1568@-17970434/-310448128fps, stride = 0, quantization: 0
[  211.154346] cif_isp10_s_fmt_mp: mp llength=0x840
[  211.154393] ov13850.ov_camera_module_enum_frameintervals: 0
[  211.154439] ov13850.ov_camera_module_ioctl: cmd: 0x2
[  211.154483] ov13850.pltfrm_camera_module_ioctl: cmd: 0x2
[  211.154529] ov13850.ov_camera_module_enum_frameintervals: 1
[  211.154570] ov13850.ov_camera_module_ioctl: cmd: 0x2
[  211.154611] ov13850.pltfrm_camera_module_ioctl: cmd: 0x2
[  211.154654] ov13850.ov_camera_module_enum_frameintervals: 2
[  211.154723] cif_isp10_img_src_select_strm_fmt: requesting format BAYER BGGR10 2112x1568(0,0,2112,1568)@30/1fps from ov13850
[  211.154778] ov13850.ov_camera_module_s_fmt: 2112x1568, fmt code 0x3007
[  211.154828] ov13850.ov_camera_module_find_config: 2112x1568, fmt code 0x3007
[  211.154874] ov13850.ov_camera_module_find_config: found matching config 2112x1568_30fps
[  211.154924] ov13850.ov_camera_module_find_config: 2112x1568, fmt code 0x3007
[  211.154969] ov13850.ov_camera_module_find_config: frame interval 1/30
[  211.155013] ov13850.ov_camera_module_find_config: found matching config 2112x1568_30fps
[  211.155053] ov13850.ov_camera_module_set_active_config: 
[  211.155103] ov13850.ov_camera_module_s_frame_interval: 1/30 (30fps)
[  211.155152] ov13850.ov_camera_module_find_config: 2112x1568, fmt code 0x3007
[  211.155197] ov13850.ov_camera_module_find_config: frame interval 1/30
[  211.155242] ov13850.ov_camera_module_find_config: found matching config 2112x1568_30fps
[  211.155290] ov13850.ov_camera_module_g_ctrl:  id 0x981981
[  211.155334] ov13850.ov_camera_module_g_ctrl: RK_V4L2_CID_VBLANKING 5000
[  211.155376] cif_isp10_calc_min_out_buff_size: MP
[  211.155418] cif_isp10_calc_min_out_buff_size: mi_path->llength: 0x840
[  211.155460] cif_isp10_calc_min_out_buff_size: calculated buffer size: 4967424
[  211.155663] cif_isp10_v4l2_vb2_queue_setup: VIDEO_CAPTURE request planes 0 count 4, size 0
[  211.155710] cif_isp10_calc_min_out_buff_size: MP
[  211.155752] cif_isp10_calc_min_out_buff_size: mi_path->llength: 0x840
[  211.155793] cif_isp10_calc_min_out_buff_size: calculated buffer size: 4967424
[  211.155841] cif_isp10_v4l2_vb2_queue_setup: VIDEO_CAPTURE count 4, size 4967424
[  211.229622] cif_isp10_v4l2_streamon: VIDEO_CAPTURE(1)
[  211.238888] cif_isp10_v4l2_vb2_queue: buffer type VIDEO_CAPTURE
[  211.238923] cif_isp10_calc_min_out_buff_size: MP
[  211.238951] cif_isp10_calc_min_out_buff_size: mi_path->llength: 0x840
[  211.238977] cif_isp10_calc_min_out_buff_size: calculated buffer size: 4967424
[  211.248176] cif_isp10_v4l2_vb2_queue: buffer type VIDEO_CAPTURE
[... 186 lines cut ...]
[  211.806854] ov13850.ov_camera_module_ioctl: cmd: 0x805876c0
[  211.856846] cif_isp10_set_pm_state: STREAMING -> STANDBY
[  211.856854] cif_isp10_qbuf: MP
[  211.856860] cif_isp10_qbuf: MP
[  211.856865] cif_isp10_qbuf: MP
[  211.856870] cif_isp10_qbuf: MP
[  211.856881] cif_isp10_stop: SP state = DISABLED, MP state = READY, DMA state = INACTIVE, img_src state = SW_STNDBY
[  211.856881]   MI_CTRL 0x007a2000
[  211.856881]   ISP_CTRL 0x00000006
[... cut ...]
[  211.878810] cif_isp10_img_src_set_state: SW_STNDBY -> OFF
[  211.878861] ov13850.ov_camera_module_s_power: 0
[  211.878906] ov13850.pltfrm_camera_module_s_power: off
[  211.879138] ov13850.tcam_camera_module_standby: standby: enable=1
[  211.879509] tcam_pca6408a_set_out: 0x10
[  211.879517] ov13850.pltfrm_camera_module_s_power: set PM state to 0 successful, camera module is off
[  211.879522] ov13850.ov_camera_module_reset: 
[  211.879528] ov13850.ov_camera_module_s_power: camera powered off
```