rockchip-linux/kernel #146: RK3399 onboard Gigabit wired network card loses response

newcoderlife, 2019-03-06:
I am holding a rk3399 development board. When I make a large traffic network access (BT download or similar), eth0 will reject any traffic. Both ICMP and TCP protocols are inaccessible.

I improved `rockchip, pwm_voltage` to 1.0v based on the issue that has been closed, but the problem remains unresolved.

After searching, the Gigabit onboard wired network card using the rk3399 solution has similar problems. Is this a problem with kernel code? Thank you!

newcoderlife, 2019-03-06:
#27 This maybe the same problem.
@jakob-tsd

rugalash, 2020-03-14:
> #27 This maybe the same problem.
> @jakob-tsd

Yes that is
You can also view here that today (years later) there is a fix coming in the kernel:
https://github.com/MichaIng/DietPi/issues/2028

MichaIng, 2020-03-14:
Note that the linked fix has only been applied to the Armbian 5.X kernel so far, AFAIK. A patch has been submitted to upstream Linux as well but not yet been accepted from what I can see.

newcoderlife, 2020-03-15:
Wow! That's a year ago. I used to think its a hardware issue but it's not. Anyway thanks a lot!

MichaIng, 2020-03-15:
@newcoderlife 
Jep, unbelievable how long such a major bug (that failing/unstable onboard Ethernet connection is, IMO) survives on such a widely used high-end SoC in all major official Linux versions/repos.

newcoderlife, 2020-03-22:
Since the bug has been fixed, I'll close this issue.

closed 2020-03-22 by newcoderlife