rockchip-linux/kernel #150: rockchip_tsp.c:43:30: fatal error: linux/rk_tsp_api.h: No such file or directory

ballsystemlord, 2019-04-02:
I hand configured my kernel using the basic rockchip config and a previous config I did for my AMD64 system as a reference.
I am getting this error and can't seem to find a fix:
```
rivers/media/platform/rockchip-tsp/rockchip_tsp.c:43:30: fatal error: linux/rk_tsp_api.h: No such file or directory
 #include <linux/rk_tsp_api.h>
                              ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/media/platform/rockchip-tsp/rockchip_tsp.o' failed
make[4]: *** [drivers/media/platform/rockchip-tsp/rockchip_tsp.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/media/platform/rockchip-tsp' failed
make[3]: *** [drivers/media/platform/rockchip-tsp] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/media/platform' failed
make[2]: *** [drivers/media/platform] Error 2
make[2]: *** Waiting for unfinished jobs....
```

Here's a copy of my kernel config:
http://dpaste.com//0594NGJ

I'm using Devuan (Debian) Linux to compile the kernel on with gcc.
```% gcc --version
gcc (Debian 6.3.0-18+deb9u1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

The commit I'm working from is the latest commit:
7532103cd08c810f95fdc71822be14fc17b455fc

If you require additional details please ask.

ballsystemlord, 2019-04-02:
If I disable ROCKCHIP_TSP then I get another problem:
```
In file included from drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.c:7:0:
drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.h:5:34: fatal error: linux/rockchip/iomap.h: No such file or directory
 #include <linux/rockchip/iomap.h>
                                  ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.o' failed
make[5]: *** [drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/hdmi/rockchip-hdmiv1' failed
make[4]: *** [drivers/video/rockchip/hdmi/rockchip-hdmiv1] Error 2
make[4]: *** Waiting for unfinished jobs....
```

ballsystemlord, 2019-04-02:
As I found that the file iomap.h was actually removed: https://github.com/rockchip-linux/kernel/commit/e85d7dbaa051106d0f4a354076c1938e8f75745a
I decided to try to just remove that line and ran into a whole new set of errors.

```In file included from include/linux/fb.h:16:0,
                 from drivers/video/rockchip/hdmi/rockchip-hdmiv1/../rockchip-hdmi.h:5,
                 from drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1.h:5,
                 from drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_cec.c:2:
drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.h: In function 'rockchip_hdmiv1_reset_pclk':
drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.h:460:29: error: 'RK_CRU_VIRT' undeclared (first use in this function)
  writel_relaxed(0x00010001, RK_CRU_VIRT + 0x128);
                             ^
./arch/arm64/include/asm/io.h:127:78: note: in definition of macro 'writel_relaxed'
 #define writel_relaxed(v,c) ((void)__raw_writel((__force u32)cpu_to_le32(v),(c)))
                                                                              ^
drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.h:460:29: note: each undeclared identifier is reported only once for each function it appears in
  writel_relaxed(0x00010001, RK_CRU_VIRT + 0x128);
                             ^
./arch/arm64/include/asm/io.h:127:78: note: in definition of macro 'writel_relaxed'
 #define writel_relaxed(v,c) ((void)__raw_writel((__force u32)cpu_to_le32(v),(c)))
                                                                              ^
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_cec.o' failed
make[5]: *** [drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_cec.o] Error 1
make[5]: *** Waiting for unfinished jobs....
  CC      drivers/video/rockchip/transmitter/rk32_lvds.o
In file included from include/linux/io.h:25:0,
                 from drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.c:3:
drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.h: In function 'rockchip_hdmiv1_reset_pclk':
drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.h:460:29: error: 'RK_CRU_VIRT' undeclared (first use in this function)
  writel_relaxed(0x00010001, RK_CRU_VIRT + 0x128);
                             ^
./arch/arm64/include/asm/io.h:127:78: note: in definition of macro 'writel_relaxed'
 #define writel_relaxed(v,c) ((void)__raw_writel((__force u32)cpu_to_le32(v),(c)))
                                                                              ^
drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.h:460:29: note: each undeclared identifier is reported only once for each function it appears in
  writel_relaxed(0x00010001, RK_CRU_VIRT + 0x128);
                             ^
./arch/arm64/include/asm/io.h:127:78: note: in definition of macro 'writel_relaxed'
 #define writel_relaxed(v,c) ((void)__raw_writel((__force u32)cpu_to_le32(v),(c)))
                                                                              ^
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.o' failed
make[5]: *** [drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.o] Error 1
In file included from include/linux/scatterlist.h:8:0,
                 from include/linux/dma-mapping.h:10,
                 from drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1.c:11:
drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.h: In function 'rockchip_hdmiv1_reset_pclk':
drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.h:460:29: error: 'RK_CRU_VIRT' undeclared (first use in this function)
  writel_relaxed(0x00010001, RK_CRU_VIRT + 0x128);
                             ^
./arch/arm64/include/asm/io.h:127:78: note: in definition of macro 'writel_relaxed'
 #define writel_relaxed(v,c) ((void)__raw_writel((__force u32)cpu_to_le32(v),(c)))
                                                                              ^
drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hw.h:460:29: note: each undeclared identifier is reported only once for each function it appears in
  writel_relaxed(0x00010001, RK_CRU_VIRT + 0x128);
                             ^
./arch/arm64/include/asm/io.h:127:78: note: in definition of macro 'writel_relaxed'
 #define writel_relaxed(v,c) ((void)__raw_writel((__force u32)cpu_to_le32(v),(c)))
                                                                              ^
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1.o' failed
make[5]: *** [drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1.o] Error 1
  CC      drivers/video/rockchip/tve/gm7122/gm7122_tve.o
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/hdmi/rockchip-hdmiv1' failed
make[4]: *** [drivers/video/rockchip/hdmi/rockchip-hdmiv1] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/hdmi' failed
make[3]: *** [drivers/video/rockchip/hdmi] Error 2
make[3]: *** Waiting for unfinished jobs....
  LD      drivers/video/rockchip/screen/built-in.o
  CC      drivers/video/rockchip/transmitter/rk31xx_lvds.o
  CC      drivers/video/rockchip/tve/rk1000/rk1000_tve.o
  CC      drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.o
  CC      drivers/video/rockchip/transmitter/dp_anx6345.o
[... 53 lines cut ...]
[... cut ...]
make[4]: *** [drivers/video/rockchip/tve/rk1000] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve' failed
make[3]: *** [drivers/video/rockchip/tve] Error 2
  LD      drivers/video/rockchip/lcdc/built-in.o
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[2]: *** [drivers/video/rockchip] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[1]: *** [drivers/video] Error 2
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2```

ballsystemlord, 2019-04-02:
If I place iomap.h back into the kernel source I still get more errors...
```  CC      drivers/video/rockchip/vpu/mpp_service.o
/tmp/cc0zRIeA.s: Assembler messages:
/tmp/cc0zRIeA.s:140: Error: missing immediate expression at operand 1 -- `dsb '
/tmp/cc0zRIeA.s:380: Error: missing immediate expression at operand 1 -- `dsb '
/tmp/cc0zRIeA.s:448: Error: missing immediate expression at operand 1 -- `dsb '
/tmp/cc0zRIeA.s:471: Error: missing immediate expression at operand 1 -- `dsb '
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/rk32_lvds.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/rk32_lvds.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/transmitter' failed
make[3]: *** [drivers/video/rockchip/transmitter] Error 2
make[3]: *** Waiting for unfinished jobs....
  CC      drivers/video/rockchip/vpu/mpp_dev_rkvenc.o
  CC      drivers/video/rockchip/vpu/mpp_dev_vepu.o
  CC      drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_cec.o
  CC      drivers/video/rockchip/hdmi/rockchip-hdmiv1/rockchip_hdmiv1_hdcp.o
  LD      drivers/video/rockchip/tve/gm7122/built-in.o
  CC      drivers/video/rockchip/tve/rk1000/rk1000_tve.o
  CC      drivers/video/rockchip/vpu/mpp_dev_h265e.o
  CC      drivers/video/rockchip/vpu/mpp_dev_common.o
  CC      drivers/video/rockchip/vpu/vpu_iommu_drm.o
  CC      drivers/video/rockchip/vpu/vpu_iommu_ion.o
drivers/video/rockchip/tve/rk1000/rk1000_tve.c:25:36: fatal error: dt-bindings/rkfb/rk_fb.h: No such file or directory
 #include <dt-bindings/rkfb/rk_fb.h>
                                    ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve/rk1000' failed
make[4]: *** [drivers/video/rockchip/tve/rk1000] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve' failed
make[3]: *** [drivers/video/rockchip/tve] Error 2
  CC      drivers/video/rockchip/vpu/vpu_iommu_ops.o
  LD      drivers/video/rockchip/hdmi/rockchip-hdmiv1/built-in.o
  LD      drivers/video/rockchip/hdmi/built-in.o
  LD      drivers/video/rockchip/vcodec/rk-vcodec.o
  LD      drivers/video/rockchip/vcodec/built-in.o
  LD      drivers/video/rockchip/vpu/built-in.o
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[2]: *** [drivers/video/rockchip] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[1]: *** [drivers/video] Error 2
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

ballsystemlord, 2019-04-02:
I'll start changing my .config back to the vanilla rockchip config and get back to you later. Right now I need to go to bed. Thanks for your help.

ballsystemlord, 2019-04-02:
Disabling CONFIG_RK_HDMI_V1 and CONFIG_RK_HDMI_V2 still gives me errors:

```drivers/video/rockchip/transmitter/rk32_lvds.c:14:34: fatal error: linux/rockchip/iomap.h: No such file or directory
 #include <linux/rockchip/iomap.h>
                                  ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/rk32_lvds.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/rk32_lvds.o] Error 1
make[4]: *** Waiting for unfinished jobs....
drivers/video/rockchip/tve/rk1000/rk1000_tve.c:25:36: fatal error: dt-bindings/rkfb/rk_fb.h: No such file or directory
 #include <dt-bindings/rkfb/rk_fb.h>
                                    ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve.o] Error 1
make[5]: *** Waiting for unfinished jobs....
drivers/video/rockchip/tve/gm7122/gm7122_tve.c:21:34: fatal error: linux/rockchip/iomap.h: No such file or directory
 #include <linux/rockchip/iomap.h>
                                  ^
compilation terminated.
drivers/video/rockchip/transmitter/rk31xx_lvds.c:27:34: fatal error: linux/rockchip/iomap.h: No such file or directory
 #include <linux/rockchip/iomap.h>
                                  ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/gm7122/gm7122_tve.o' failed
make[5]: *** [drivers/video/rockchip/tve/gm7122/gm7122_tve.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve/gm7122' failed
make[4]: *** [drivers/video/rockchip/tve/gm7122] Error 2
make[4]: *** Waiting for unfinished jobs....
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/rk31xx_lvds.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/rk31xx_lvds.o] Error 1
In file included from drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.c:4:0:
drivers/video/rockchip/tve/rk1000/rk1000_tve.h:7:30: fatal error: linux/mfd/rk1000.h: No such file or directory
 #include <linux/mfd/rk1000.h>
                              ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.o] Error 1
drivers/video/rockchip/transmitter/dp_anx6345.c: In function 'i2c_master_reg8_send':
drivers/video/rockchip/transmitter/dp_anx6345.c:43:12: error: 'struct i2c_msg' has no member named 'scl_rate'
         msg.scl_rate = scl_rate;
            ^
drivers/video/rockchip/transmitter/dp_anx6345.c: In function 'i2c_master_reg8_recv':
drivers/video/rockchip/transmitter/dp_anx6345.c:63:16: error: 'struct i2c_msg' has no member named 'scl_rate'
         msgs[0].scl_rate = scl_rate;
                ^
drivers/video/rockchip/transmitter/dp_anx6345.c:69:16: error: 'struct i2c_msg' has no member named 'scl_rate'
         msgs[1].scl_rate = scl_rate;
                ^
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/dp_anx6345.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/dp_anx6345.o] Error 1
In file included from drivers/video/rockchip/transmitter/dp501.c:12:0:
include/linux/dp501.h:6:31: fatal error: linux/earlysuspend.h: No such file or directory
 #include<linux/earlysuspend.h>
                               ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/dp501.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/dp501.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/transmitter' failed
make[3]: *** [drivers/video/rockchip/transmitter] Error 2
make[3]: *** Waiting for unfinished jobs....
In file included from drivers/video/rockchip/tve/rk1000/rk1000_tve_cvbs.c:5:0:
drivers/video/rockchip/tve/rk1000/rk1000_tve.h:7:30: fatal error: linux/mfd/rk1000.h: No such file or directory
 #include <linux/mfd/rk1000.h>
                              ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve_cvbs.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve_cvbs.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve/rk1000' failed
make[4]: *** [drivers/video/rockchip/tve/rk1000] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve' failed
make[3]: *** [drivers/video/rockchip/tve] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[2]: *** [drivers/video/rockchip] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[1]: *** [drivers/video] Error 2
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2```

ballsystemlord, 2019-04-02:
Disabling CONFIG_RK32_LVDS still is not all that is needed to get the kernel to build:

```drivers/video/rockchip/tve/gm7122/gm7122_tve.c:21:34: fatal error: linux/rockchip/iomap.h: No such file or directory
 #include <linux/rockchip/iomap.h>
                                  ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/gm7122/gm7122_tve.o' failed
make[5]: *** [drivers/video/rockchip/tve/gm7122/gm7122_tve.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve/gm7122' failed
make[4]: *** [drivers/video/rockchip/tve/gm7122] Error 2
make[4]: *** Waiting for unfinished jobs....
drivers/video/rockchip/transmitter/rk31xx_lvds.c:27:34: fatal error: linux/rockchip/iomap.h: No such file or directory
 #include <linux/rockchip/iomap.h>
                                  ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/rk31xx_lvds.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/rk31xx_lvds.o] Error 1
make[4]: *** Waiting for unfinished jobs....
drivers/video/rockchip/tve/rk1000/rk1000_tve.c:25:36: fatal error: dt-bindings/rkfb/rk_fb.h: No such file or directory
 #include <dt-bindings/rkfb/rk_fb.h>
                                    ^
compilation terminated.
drivers/video/rockchip/transmitter/dp_anx6345.c: In function 'i2c_master_reg8_send':
drivers/video/rockchip/transmitter/dp_anx6345.c:43:12: error: 'struct i2c_msg' has no member named 'scl_rate'
         msg.scl_rate = scl_rate;
            ^
drivers/video/rockchip/transmitter/dp_anx6345.c: In function 'i2c_master_reg8_recv':
drivers/video/rockchip/transmitter/dp_anx6345.c:63:16: error: 'struct i2c_msg' has no member named 'scl_rate'
         msgs[0].scl_rate = scl_rate;
                ^
drivers/video/rockchip/transmitter/dp_anx6345.c:69:16: error: 'struct i2c_msg' has no member named 'scl_rate'
         msgs[1].scl_rate = scl_rate;
                ^
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve.o] Error 1
make[5]: *** Waiting for unfinished jobs....
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/dp_anx6345.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/dp_anx6345.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/transmitter' failed
make[3]: *** [drivers/video/rockchip/transmitter] Error 2
make[3]: *** Waiting for unfinished jobs....
In file included from drivers/video/rockchip/tve/rk1000/rk1000_tve_cvbs.c:5:0:
drivers/video/rockchip/tve/rk1000/rk1000_tve.h:7:30: fatal error: linux/mfd/rk1000.h: No such file or directory
 #include <linux/mfd/rk1000.h>
                              ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve_cvbs.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve_cvbs.o] Error 1
In file included from drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.c:4:0:
drivers/video/rockchip/tve/rk1000/rk1000_tve.h:7:30: fatal error: linux/mfd/rk1000.h: No such file or directory
 #include <linux/mfd/rk1000.h>
                              ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve/rk1000' failed
make[4]: *** [drivers/video/rockchip/tve/rk1000] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve' failed
make[3]: *** [drivers/video/rockchip/tve] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[2]: *** [drivers/video/rockchip] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[1]: *** [drivers/video] Error 2
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2```

ballsystemlord, 2019-04-02:
Disabling GM7122_TV_ENCODER leads to:

```
drivers/video/rockchip/transmitter/rk31xx_lvds.c:27:34: fatal error: linux/rockchip/iomap.h: No such file or directory
 #include <linux/rockchip/iomap.h>
                                  ^
compilation terminated.
drivers/video/rockchip/tve/rk1000/rk1000_tve.c:25:36: fatal error: dt-bindings/rkfb/rk_fb.h: No such file or directory
 #include <dt-bindings/rkfb/rk_fb.h>
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/rk31xx_lvds.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/rk31xx_lvds.o] Error 1
make[4]: *** Waiting for unfinished jobs....
                                    ^
compilation terminated.
  CC      drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.o
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve.o] Error 1
make[5]: *** Waiting for unfinished jobs....
drivers/video/rockchip/transmitter/dp_anx6345.c: In function 'i2c_master_reg8_send':
drivers/video/rockchip/transmitter/dp_anx6345.c:43:12: error: 'struct i2c_msg' has no member named 'scl_rate'
         msg.scl_rate = scl_rate;
            ^
drivers/video/rockchip/transmitter/dp_anx6345.c: In function 'i2c_master_reg8_recv':
drivers/video/rockchip/transmitter/dp_anx6345.c:63:16: error: 'struct i2c_msg' has no member named 'scl_rate'
         msgs[0].scl_rate = scl_rate;
                ^
drivers/video/rockchip/transmitter/dp_anx6345.c:69:16: error: 'struct i2c_msg' has no member named 'scl_rate'
         msgs[1].scl_rate = scl_rate;
                ^
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/dp_anx6345.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/dp_anx6345.o] Error 1
In file included from drivers/video/rockchip/transmitter/dp501.c:12:0:
include/linux/dp501.h:6:31: fatal error: linux/earlysuspend.h: No such file or directory
 #include<linux/earlysuspend.h>
                               ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/dp501.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/dp501.o] Error 1
In file included from drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.c:4:0:
drivers/video/rockchip/tve/rk1000/rk1000_tve.h:7:30: fatal error: linux/mfd/rk1000.h: No such file or directory
 #include <linux/mfd/rk1000.h>
                              ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve/rk1000' failed
make[4]: *** [drivers/video/rockchip/tve/rk1000] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve' failed
make[3]: *** [drivers/video/rockchip/tve] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/transmitter' failed
make[3]: *** [drivers/video/rockchip/transmitter] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[2]: *** [drivers/video/rockchip] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[1]: *** [drivers/video] Error 2
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

ballsystemlord, 2019-04-02:
Disabling RK31XX_LVDS still throws errors:

```
drivers/video/rockchip/transmitter/dp_anx6345.c: In function 'i2c_master_reg8_send':
drivers/video/rockchip/transmitter/dp_anx6345.c:43:12: error: 'struct i2c_msg' has no member named 'scl_rate'
         msg.scl_rate = scl_rate;
            ^
drivers/video/rockchip/transmitter/dp_anx6345.c: In function 'i2c_master_reg8_recv':
drivers/video/rockchip/transmitter/dp_anx6345.c:63:16: error: 'struct i2c_msg' has no member named 'scl_rate'
         msgs[0].scl_rate = scl_rate;
                ^
drivers/video/rockchip/transmitter/dp_anx6345.c:69:16: error: 'struct i2c_msg' has no member named 'scl_rate'
         msgs[1].scl_rate = scl_rate;
                ^
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/dp_anx6345.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/dp_anx6345.o] Error 1
make[4]: *** Waiting for unfinished jobs....
drivers/video/rockchip/tve/rk1000/rk1000_tve.c:25:36: fatal error: dt-bindings/rkfb/rk_fb.h: No such file or directory
 #include <dt-bindings/rkfb/rk_fb.h>
                                    ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve/rk1000' failed
make[4]: *** [drivers/video/rockchip/tve/rk1000] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve' failed
make[3]: *** [drivers/video/rockchip/tve] Error 2
make[3]: *** Waiting for unfinished jobs....
In file included from drivers/video/rockchip/transmitter/dp501.c:12:0:
include/linux/dp501.h:6:31: fatal error: linux/earlysuspend.h: No such file or directory
 #include<linux/earlysuspend.h>
                               ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/dp501.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/dp501.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/transmitter' failed
make[3]: *** [drivers/video/rockchip/transmitter] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[2]: *** [drivers/video/rockchip] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[1]: *** [drivers/video] Error 2
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

ballsystemlord, 2019-04-02:
Disabling DP_ANX6345:

```
In file included from drivers/video/rockchip/transmitter/dp501.c:12:0:
include/linux/dp501.h:6:31: fatal error: linux/earlysuspend.h: No such file or directory
 #include<linux/earlysuspend.h>
                               ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/dp501.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/dp501.o] Error 1
make[4]: *** Waiting for unfinished jobs....
In file included from drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.c:4:0:
drivers/video/rockchip/tve/rk1000/rk1000_tve.h:7:30: fatal error: linux/mfd/rk1000.h: No such file or directory
 #include <linux/mfd/rk1000.h>
                              ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.o] Error 1
make[5]: *** Waiting for unfinished jobs....
drivers/video/rockchip/tve/rk1000/rk1000_tve.c:25:36: fatal error: dt-bindings/rkfb/rk_fb.h: No such file or directory
 #include <dt-bindings/rkfb/rk_fb.h>
                                    ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve.o] Error 1
In file included from drivers/video/rockchip/tve/rk1000/rk1000_tve_cvbs.c:5:0:
drivers/video/rockchip/tve/rk1000/rk1000_tve.h:7:30: fatal error: linux/mfd/rk1000.h: No such file or directory
 #include <linux/mfd/rk1000.h>
                              ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve_cvbs.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve_cvbs.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve/rk1000' failed
make[4]: *** [drivers/video/rockchip/tve/rk1000] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve' failed
make[3]: *** [drivers/video/rockchip/tve] Error 2
make[3]: *** Waiting for unfinished jobs....
drivers/video/rockchip/transmitter/vga.c:14:24: fatal error: ../../edid.h: No such file or directory
 #include "../../edid.h"
                        ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/vga.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/vga.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/transmitter' failed
make[3]: *** [drivers/video/rockchip/transmitter] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[2]: *** [drivers/video/rockchip] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[1]: *** [drivers/video] Error 2
make[1]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

ballsystemlord, 2019-04-02:
Disabling DP501:
```
drivers/video/rockchip/tve/rk1000/rk1000_tve.c:25:36: fatal error: dt-bindings/rkfb/rk_fb.h: No such file or directory
 #include <dt-bindings/rkfb/rk_fb.h>
                                    ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve.o] Error 1
make[5]: *** Waiting for unfinished jobs....
In file included from drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.c:4:0:
drivers/video/rockchip/tve/rk1000/rk1000_tve.h:7:30: fatal error: linux/mfd/rk1000.h: No such file or directory
 #include <linux/mfd/rk1000.h>
                              ^
compilation terminated.
drivers/video/rockchip/transmitter/vga.c:14:24: fatal error: ../../edid.h: No such file or directory
 #include "../../edid.h"
                        ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/vga.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/vga.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/transmitter' failed
make[3]: *** [drivers/video/rockchip/transmitter] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve_ypbpr.o] Error 1
In file included from drivers/video/rockchip/tve/rk1000/rk1000_tve_cvbs.c:5:0:
drivers/video/rockchip/tve/rk1000/rk1000_tve.h:7:30: fatal error: linux/mfd/rk1000.h: No such file or directory
 #include <linux/mfd/rk1000.h>
                              ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/tve/rk1000/rk1000_tve_cvbs.o' failed
make[5]: *** [drivers/video/rockchip/tve/rk1000/rk1000_tve_cvbs.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve/rk1000' failed
make[4]: *** [drivers/video/rockchip/tve/rk1000] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/tve' failed
make[3]: *** [drivers/video/rockchip/tve] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[2]: *** [drivers/video/rockchip] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[1]: *** [drivers/video] Error 2
make[1]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

ballsystemlord, 2019-04-02:
Disabling RK1000_TVOUT , RK1000_TVOUT_YPbPr , RK1000_TVOUT_CVBS still is not enough:
```
drivers/video/rockchip/transmitter/vga.c:14:24: fatal error: ../../edid.h: No such file or directory
 #include "../../edid.h"
                        ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/video/rockchip/transmitter/vga.o' failed
make[4]: *** [drivers/video/rockchip/transmitter/vga.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/transmitter' failed
make[3]: *** [drivers/video/rockchip/transmitter] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[2]: *** [drivers/video/rockchip] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[1]: *** [drivers/video] Error 2
make[1]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

ballsystemlord, 2019-04-02:
Disabling RK_VGA is still insufficient, but I'm making progress:
```
drivers/built-in.o: In function `rk32_edp_probe':
drivers/video/rockchip/transmitter/rk32_dp.c:1762: undefined reference to `rk_fb_get_prmry_screen'
drivers/video/rockchip/transmitter/rk32_dp.c:1836: undefined reference to `support_uboot_display'
drivers/video/rockchip/transmitter/rk32_dp.c:1850: undefined reference to `support_uboot_display'
drivers/video/rockchip/transmitter/rk32_dp.c:1854: undefined reference to `support_uboot_display'
drivers/video/rockchip/transmitter/rk32_dp.c:1860: undefined reference to `rk_fb_trsm_ops_register'
drivers/built-in.o: In function `rk32_edp_enable':
drivers/video/rockchip/transmitter/rk32_dp.c:133: undefined reference to `rk_fb_get_prmry_screen'
drivers/built-in.o: In function `edp_psr_state':
drivers/video/rockchip/transmitter/rk32_dp.c:1460: undefined reference to `rk_fb_poll_wait_frame_complete'
drivers/built-in.o: In function `hdmi_wq_remove':
drivers/video/rockchip/hdmi/rockchip-hdmi-core.c:404: undefined reference to `rk_fb_switch_screen'
drivers/built-in.o: In function `hdmi_work_queue':
drivers/video/rockchip/hdmi/rockchip-hdmi-core.c:513: undefined reference to `rk_fb_get_display_policy'
drivers/built-in.o: In function `rockchip_hdmi_register':
drivers/video/rockchip/hdmi/rockchip-hdmi-core.c:659: undefined reference to `rk_get_lcdc_drv'
drivers/built-in.o: In function `hdmi_set_lcdc':
drivers/video/rockchip/hdmi/rockchip-hdmi-lcdc.c:1080: undefined reference to `rk_fb_switch_screen'
drivers/built-in.o: In function `hdmi_set_scale':
drivers/video/rockchip/hdmi/rockchip-hdmi-sysfs.c:328: undefined reference to `rk_fb_disp_scale'
drivers/built-in.o: In function `hdmi_register_display_sysfs':
drivers/video/rockchip/hdmi/rockchip-hdmi-sysfs.c:666: undefined reference to `rk_display_device_register'
drivers/built-in.o: In function `hdmi_unregister_display_sysfs':
drivers/video/rockchip/hdmi/rockchip-hdmi-sysfs.c:672: undefined reference to `rk_display_device_unregister'
drivers/built-in.o: In function `kbase_platform_device_shutdown':
drivers/gpu/arm/midgard/mali_kbase_core_linux.c:4372: undefined reference to `kbase_platform_rk_shutdown'
drivers/built-in.o: In function `kbase_platform_device_probe':
drivers/gpu/arm/midgard/mali_kbase_core_linux.c:3884: undefined reference to `kbase_platform_rk_init_opp_table'
drivers/built-in.o: In function `wifi_power_write':
drivers/net/wireless/rockchip_wlan/wifi_sys/rkwifi_sys_iface.c:156: undefined reference to `rockchip_wifi_power'
drivers/built-in.o: In function `wifi_chip_read':
drivers/net/wireless/rockchip_wlan/wifi_sys/rkwifi_sys_iface.c:19: undefined reference to `get_wifi_chip_type'
drivers/built-in.o: In function `rkwifi_set_firmware':
drivers/net/wireless/rockchip_wlan/rkwifi/rk_wifi_config.c:27: undefined reference to `get_wifi_chip_type'
drivers/built-in.o: In function `dwc3_rockchip_host_testmode_write':
drivers/usb/dwc3/dwc3-rockchip.c:265: undefined reference to `xhci_set_link_state'
drivers/built-in.o: In function `dwc3_rockchip_host_testmode_write':
drivers/usb/dwc3/dwc3-rockchip-inno.c:164: undefined reference to `xhci_set_link_state'
drivers/built-in.o: In function `sfc_erase_mtd':
drivers/rkflash/sfc_nor_mtd.c:69: undefined reference to `mtd_erase_callback'
drivers/built-in.o: In function `sfc_nor_mtd_init':
drivers/rkflash/sfc_nor_mtd.c:214: undefined reference to `mtd_device_parse_register'
drivers/built-in.o:(.debug_addr+0x37550): undefined reference to `rk_fb_poll_wait_frame_complete'
drivers/built-in.o:(.debug_addr+0x37660): undefined reference to `rk_fb_trsm_ops_register'
drivers/built-in.o:(.debug_addr+0x377c0): undefined reference to `support_uboot_display'
drivers/built-in.o:(.debug_addr+0x37920): undefined reference to `rk_fb_get_prmry_screen'
drivers/built-in.o:(.debug_addr+0x397c8): undefined reference to `rk_fb_switch_screen'
drivers/built-in.o:(.debug_addr+0x399a8): undefined reference to `rk_fb_get_display_policy'
drivers/built-in.o:(.debug_addr+0x39bf0): undefined reference to `rk_get_lcdc_drv'
drivers/built-in.o:(.debug_addr+0x3a3f0): undefined reference to `rk_fb_switch_screen'
drivers/built-in.o:(.debug_addr+0x3a9f0): undefined reference to `rk_display_device_unregister'
drivers/built-in.o:(.debug_addr+0x3aa30): undefined reference to `rk_display_device_register'
drivers/built-in.o:(.debug_addr+0x3ac88): undefined reference to `rk_fb_disp_scale'
drivers/built-in.o:(.debug_addr+0xdea28): undefined reference to `kbase_platform_rk_shutdown'
drivers/built-in.o:(.debug_addr+0xdfe18): undefined reference to `kbase_platform_rk_init_opp_table'
drivers/built-in.o:(.debug_addr+0x17fe18): undefined reference to `get_wifi_chip_type'
drivers/built-in.o:(.debug_addr+0x1800a0): undefined reference to `rockchip_wifi_power'
drivers/built-in.o:(.debug_addr+0x180250): undefined reference to `get_wifi_chip_type'
drivers/built-in.o:(.debug_addr+0x19c058): undefined reference to `xhci_set_link_state'
drivers/built-in.o:(.debug_addr+0x19d500): undefined reference to `xhci_set_link_state'
drivers/built-in.o:(.debug_addr+0x27a148): undefined reference to `mtd_erase_callback'
drivers/built-in.o:(.debug_addr+0x27a278): undefined reference to `mtd_device_parse_register'
net/built-in.o: In function `wiphy_rfkill_start_polling':
net/wireless/core.c:764: undefined reference to `rfkill_resume_polling'
net/built-in.o: In function `wiphy_rfkill_stop_polling':
net/wireless/core.c:772: undefined reference to `rfkill_pause_polling'
[... cut ...]
net/built-in.o:(.debug_addr+0xc7ce8): undefined reference to `rfkill_destroy'
net/built-in.o:(.debug_addr+0xc8040): undefined reference to `rfkill_blocked'
net/built-in.o:(.debug_addr+0xc86e0): undefined reference to `rfkill_pause_polling'
net/built-in.o:(.debug_addr+0xd4510): undefined reference to `rfkill_blocked'
net/built-in.o:(.debug_addr+0xfcad0): undefined reference to `rfkill_set_sw_state'
net/built-in.o:(.debug_addr+0xfdc80): undefined reference to `rfkill_blocked'
Makefile:1009: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 1
```

ballsystemlord, 2019-04-02:
Disabling RK32_DP:
```
drivers/built-in.o: In function `hdmi_wq_remove':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/video/rockchip/hdmi/rockchip-hdmi-core.c:404: undefined reference to `rk_fb_switch_screen'
drivers/built-in.o: In function `hdmi_work_queue':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/video/rockchip/hdmi/rockchip-hdmi-core.c:513: undefined reference to `rk_fb_get_display_policy'
drivers/built-in.o: In function `rockchip_hdmi_register':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/video/rockchip/hdmi/rockchip-hdmi-core.c:659: undefined reference to `rk_get_lcdc_drv'
drivers/built-in.o: In function `hdmi_set_lcdc':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/video/rockchip/hdmi/rockchip-hdmi-lcdc.c:1080: undefined reference to `rk_fb_switch_screen'
drivers/built-in.o: In function `hdmi_set_scale':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/video/rockchip/hdmi/rockchip-hdmi-sysfs.c:328: undefined reference to `rk_fb_disp_scale'
drivers/built-in.o: In function `hdmi_register_display_sysfs':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/video/rockchip/hdmi/rockchip-hdmi-sysfs.c:666: undefined reference to `rk_display_device_register'
drivers/built-in.o: In function `hdmi_unregister_display_sysfs':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/video/rockchip/hdmi/rockchip-hdmi-sysfs.c:672: undefined reference to `rk_display_device_unregister'
drivers/built-in.o: In function `kbase_platform_device_shutdown':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:4372: undefined reference to `kbase_platform_rk_shutdown'
drivers/built-in.o: In function `kbase_platform_device_probe':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:3884: undefined reference to `kbase_platform_rk_init_opp_table'
drivers/built-in.o: In function `wifi_power_write':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/net/wireless/rockchip_wlan/wifi_sys/rkwifi_sys_iface.c:156: undefined reference to `rockchip_wifi_power'
drivers/built-in.o: In function `wifi_chip_read':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/net/wireless/rockchip_wlan/wifi_sys/rkwifi_sys_iface.c:19: undefined reference to `get_wifi_chip_type'
drivers/built-in.o: In function `rkwifi_set_firmware':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/net/wireless/rockchip_wlan/rkwifi/rk_wifi_config.c:27: undefined reference to `get_wifi_chip_type'
drivers/built-in.o: In function `dwc3_rockchip_host_testmode_write':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/usb/dwc3/dwc3-rockchip.c:265: undefined reference to `xhci_set_link_state'
drivers/built-in.o: In function `dwc3_rockchip_host_testmode_write':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/usb/dwc3/dwc3-rockchip-inno.c:164: undefined reference to `xhci_set_link_state'
drivers/built-in.o: In function `sfc_erase_mtd':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/rkflash/sfc_nor_mtd.c:69: undefined reference to `mtd_erase_callback'
drivers/built-in.o: In function `sfc_nor_mtd_init':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/rkflash/sfc_nor_mtd.c:214: undefined reference to `mtd_device_parse_register'
drivers/built-in.o:(.debug_addr+0x36fb0): undefined reference to `rk_fb_switch_screen'
drivers/built-in.o:(.debug_addr+0x37190): undefined reference to `rk_fb_get_display_policy'
drivers/built-in.o:(.debug_addr+0x373d8): undefined reference to `rk_get_lcdc_drv'
drivers/built-in.o:(.debug_addr+0x37bd8): undefined reference to `rk_fb_switch_screen'
drivers/built-in.o:(.debug_addr+0x381d8): undefined reference to `rk_display_device_unregister'
drivers/built-in.o:(.debug_addr+0x38218): undefined reference to `rk_display_device_register'
drivers/built-in.o:(.debug_addr+0x38470): undefined reference to `rk_fb_disp_scale'
drivers/built-in.o:(.debug_addr+0xdc210): undefined reference to `kbase_platform_rk_shutdown'
drivers/built-in.o:(.debug_addr+0xdd600): undefined reference to `kbase_platform_rk_init_opp_table'
drivers/built-in.o:(.debug_addr+0x17d600): undefined reference to `get_wifi_chip_type'
drivers/built-in.o:(.debug_addr+0x17d888): undefined reference to `rockchip_wifi_power'
drivers/built-in.o:(.debug_addr+0x17da38): undefined reference to `get_wifi_chip_type'
drivers/built-in.o:(.debug_addr+0x199840): undefined reference to `xhci_set_link_state'
drivers/built-in.o:(.debug_addr+0x19ace8): undefined reference to `xhci_set_link_state'
drivers/built-in.o:(.debug_addr+0x277930): undefined reference to `mtd_erase_callback'
drivers/built-in.o:(.debug_addr+0x277a60): undefined reference to `mtd_device_parse_register'
net/built-in.o: In function `wiphy_rfkill_start_polling':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/net/wireless/core.c:764: undefined reference to `rfkill_resume_polling'
net/built-in.o: In function `wiphy_rfkill_stop_polling':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/net/wireless/core.c:772: undefined reference to `rfkill_pause_polling'
net/built-in.o: In function `wiphy_unregister':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/net/wireless/core.c:788: undefined reference to `rfkill_unregister'
[... cut ...]
net/built-in.o:(.debug_addr+0xc7ce8): undefined reference to `rfkill_destroy'
net/built-in.o:(.debug_addr+0xc8040): undefined reference to `rfkill_blocked'
net/built-in.o:(.debug_addr+0xc86e0): undefined reference to `rfkill_pause_polling'
net/built-in.o:(.debug_addr+0xd4510): undefined reference to `rfkill_blocked'
net/built-in.o:(.debug_addr+0xfcad0): undefined reference to `rfkill_set_sw_state'
net/built-in.o:(.debug_addr+0xfdc80): undefined reference to `rfkill_blocked'
Makefile:1009: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 1
```

ballsystemlord, 2019-04-02:
Disabling RK_HDMI:
```
drivers/built-in.o: In function `kbase_platform_device_shutdown':
drivers/gpu/arm/midgard/mali_kbase_core_linux.c:4372: undefined reference to `kbase_platform_rk_shutdown'
drivers/built-in.o: In function `kbase_platform_device_probe':
drivers/gpu/arm/midgard/mali_kbase_core_linux.c:3884: undefined reference to `kbase_platform_rk_init_opp_table'
drivers/built-in.o: In function `wifi_power_write':
drivers/net/wireless/rockchip_wlan/wifi_sys/rkwifi_sys_iface.c:156: undefined reference to `rockchip_wifi_power'
drivers/built-in.o: In function `wifi_chip_read':
drivers/net/wireless/rockchip_wlan/wifi_sys/rkwifi_sys_iface.c:19: undefined reference to `get_wifi_chip_type'
drivers/built-in.o: In function `rkwifi_set_firmware':
drivers/net/wireless/rockchip_wlan/rkwifi/rk_wifi_config.c:27: undefined reference to `get_wifi_chip_type'
drivers/built-in.o: In function `dwc3_rockchip_host_testmode_write':
drivers/usb/dwc3/dwc3-rockchip.c:265: undefined reference to `xhci_set_link_state'
drivers/built-in.o: In function `dwc3_rockchip_host_testmode_write':
drivers/usb/dwc3/dwc3-rockchip-inno.c:164: undefined reference to `xhci_set_link_state'
drivers/built-in.o: In function `sfc_erase_mtd':
drivers/rkflash/sfc_nor_mtd.c:69: undefined reference to `mtd_erase_callback'
drivers/built-in.o: In function `sfc_nor_mtd_init':
drivers/rkflash/sfc_nor_mtd.c:214: undefined reference to `mtd_device_parse_register'
drivers/built-in.o:(.debug_addr+0xda388): undefined reference to `kbase_platform_rk_shutdown'
drivers/built-in.o:(.debug_addr+0xdb778): undefined reference to `kbase_platform_rk_init_opp_table'
drivers/built-in.o:(.debug_addr+0x17b778): undefined reference to `get_wifi_chip_type'
drivers/built-in.o:(.debug_addr+0x17ba00): undefined reference to `rockchip_wifi_power'
drivers/built-in.o:(.debug_addr+0x17bbb0): undefined reference to `get_wifi_chip_type'
drivers/built-in.o:(.debug_addr+0x1979b8): undefined reference to `xhci_set_link_state'
drivers/built-in.o:(.debug_addr+0x198e60): undefined reference to `xhci_set_link_state'
drivers/built-in.o:(.debug_addr+0x275aa8): undefined reference to `mtd_erase_callback'
drivers/built-in.o:(.debug_addr+0x275bd8): undefined reference to `mtd_device_parse_register'
net/built-in.o: In function `wiphy_rfkill_start_polling':
net/wireless/core.c:764: undefined reference to `rfkill_resume_polling'
net/built-in.o: In function `wiphy_rfkill_stop_polling':
net/wireless/core.c:772: undefined reference to `rfkill_pause_polling'
net/built-in.o: In function `wiphy_unregister':
net/wireless/core.c:788: undefined reference to `rfkill_unregister'
net/built-in.o: In function `wiphy_rfkill_set_hw_state':
net/wireless/core.c:856: undefined reference to `rfkill_set_hw_state'
net/built-in.o: In function `wiphy_new_nm':
net/wireless/core.c:445: undefined reference to `rfkill_alloc'
net/built-in.o: In function `wiphy_register':
net/wireless/core.c:745: undefined reference to `rfkill_register'
net/wireless/core.c:747: undefined reference to `rfkill_destroy'
net/built-in.o: In function `cfg80211_rfkill_sync_work':
net/wireless/core.c:272: undefined reference to `rfkill_blocked'
net/built-in.o: In function `cfg80211_dev_free':
net/wireless/core.c:836: undefined reference to `rfkill_destroy'
net/built-in.o: In function `cfg80211_netdev_notifier_call':
net/wireless/core.c:1152: undefined reference to `rfkill_blocked'
net/built-in.o: In function `nl80211_start_p2p_device':
net/wireless/nl80211.c:9951: undefined reference to `rfkill_blocked'
net/built-in.o: In function `cfg80211_wext_giwtxpower':
net/wireless/wext-compat.c:921: undefined reference to `rfkill_blocked'
net/built-in.o: In function `cfg80211_wext_siwtxpower':
net/wireless/wext-compat.c:866: undefined reference to `rfkill_set_sw_state'
net/wireless/wext-compat.c:891: undefined reference to `rfkill_set_sw_state'
net/built-in.o:(.debug_addr+0xc6bb8): undefined reference to `rfkill_unregister'
net/built-in.o:(.debug_addr+0xc7088): undefined reference to `rfkill_register'
net/built-in.o:(.debug_addr+0xc7370): undefined reference to `rfkill_alloc'
net/built-in.o:(.debug_addr+0xc7550): undefined reference to `rfkill_set_hw_state'
net/built-in.o:(.debug_addr+0xc7798): undefined reference to `rfkill_resume_polling'
net/built-in.o:(.debug_addr+0xc7ce8): undefined reference to `rfkill_destroy'
net/built-in.o:(.debug_addr+0xc8040): undefined reference to `rfkill_blocked'
net/built-in.o:(.debug_addr+0xc86e0): undefined reference to `rfkill_pause_polling'
net/built-in.o:(.debug_addr+0xd4510): undefined reference to `rfkill_blocked'
net/built-in.o:(.debug_addr+0xfcad0): undefined reference to `rfkill_set_sw_state'
net/built-in.o:(.debug_addr+0xfdc80): undefined reference to `rfkill_blocked'
Makefile:1009: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 1
```

ballsystemlord, 2019-04-02:
Disabling MALI_PLATFORM_DEVICETREE , enabling MALI_PLATFORM_THIRDPARTY , and setting MALI_PLATFORM_THIRDPARTY_NAME to rk is still not enough. I wonder why the DT platform has problems...
```
drivers/built-in.o: In function `wifi_power_write':
drivers/net/wireless/rockchip_wlan/wifi_sys/rkwifi_sys_iface.c:156: undefined reference to `rockchip_wifi_power'
drivers/built-in.o: In function `wifi_chip_read':
drivers/net/wireless/rockchip_wlan/wifi_sys/rkwifi_sys_iface.c:19: undefined reference to `get_wifi_chip_type'
drivers/built-in.o: In function `rkwifi_set_firmware':
drivers/net/wireless/rockchip_wlan/rkwifi/rk_wifi_config.c:27: undefined reference to `get_wifi_chip_type'
drivers/built-in.o: In function `dwc3_rockchip_host_testmode_write':
drivers/usb/dwc3/dwc3-rockchip.c:265: undefined reference to `xhci_set_link_state'
drivers/built-in.o: In function `dwc3_rockchip_host_testmode_write':
drivers/usb/dwc3/dwc3-rockchip-inno.c:164: undefined reference to `xhci_set_link_state'
drivers/built-in.o: In function `sfc_erase_mtd':
drivers/rkflash/sfc_nor_mtd.c:69: undefined reference to `mtd_erase_callback'
drivers/built-in.o: In function `sfc_nor_mtd_init':
drivers/rkflash/sfc_nor_mtd.c:214: undefined reference to `mtd_device_parse_register'
drivers/built-in.o:(.debug_addr+0x17bb00): undefined reference to `get_wifi_chip_type'
drivers/built-in.o:(.debug_addr+0x17bd88): undefined reference to `rockchip_wifi_power'
drivers/built-in.o:(.debug_addr+0x17bf38): undefined reference to `get_wifi_chip_type'
drivers/built-in.o:(.debug_addr+0x197d40): undefined reference to `xhci_set_link_state'
drivers/built-in.o:(.debug_addr+0x1991e8): undefined reference to `xhci_set_link_state'
drivers/built-in.o:(.debug_addr+0x275e30): undefined reference to `mtd_erase_callback'
drivers/built-in.o:(.debug_addr+0x275f60): undefined reference to `mtd_device_parse_register'
net/built-in.o: In function `wiphy_rfkill_start_polling':
net/wireless/core.c:764: undefined reference to `rfkill_resume_polling'
net/built-in.o: In function `wiphy_rfkill_stop_polling':
net/wireless/core.c:772: undefined reference to `rfkill_pause_polling'
net/built-in.o: In function `wiphy_unregister':
net/wireless/core.c:788: undefined reference to `rfkill_unregister'
net/built-in.o: In function `wiphy_rfkill_set_hw_state':
net/wireless/core.c:856: undefined reference to `rfkill_set_hw_state'
net/built-in.o: In function `wiphy_new_nm':
net/wireless/core.c:445: undefined reference to `rfkill_alloc'
net/built-in.o: In function `wiphy_register':
net/wireless/core.c:745: undefined reference to `rfkill_register'
net/wireless/core.c:747: undefined reference to `rfkill_destroy'
net/built-in.o: In function `cfg80211_rfkill_sync_work':
net/wireless/core.c:272: undefined reference to `rfkill_blocked'
net/built-in.o: In function `cfg80211_dev_free':
net/wireless/core.c:836: undefined reference to `rfkill_destroy'
net/built-in.o: In function `cfg80211_netdev_notifier_call':
net/wireless/core.c:1152: undefined reference to `rfkill_blocked'
net/built-in.o: In function `nl80211_start_p2p_device':
net/wireless/nl80211.c:9951: undefined reference to `rfkill_blocked'
net/built-in.o: In function `cfg80211_wext_giwtxpower':
net/wireless/wext-compat.c:921: undefined reference to `rfkill_blocked'
net/built-in.o: In function `cfg80211_wext_siwtxpower':
net/wireless/wext-compat.c:866: undefined reference to `rfkill_set_sw_state'
net/wireless/wext-compat.c:891: undefined reference to `rfkill_set_sw_state'
net/built-in.o:(.debug_addr+0xc6bb8): undefined reference to `rfkill_unregister'
net/built-in.o:(.debug_addr+0xc7088): undefined reference to `rfkill_register'
net/built-in.o:(.debug_addr+0xc7370): undefined reference to `rfkill_alloc'
net/built-in.o:(.debug_addr+0xc7550): undefined reference to `rfkill_set_hw_state'
net/built-in.o:(.debug_addr+0xc7798): undefined reference to `rfkill_resume_polling'
net/built-in.o:(.debug_addr+0xc7ce8): undefined reference to `rfkill_destroy'
net/built-in.o:(.debug_addr+0xc8040): undefined reference to `rfkill_blocked'
net/built-in.o:(.debug_addr+0xc86e0): undefined reference to `rfkill_pause_polling'
net/built-in.o:(.debug_addr+0xd4510): undefined reference to `rfkill_blocked'
net/built-in.o:(.debug_addr+0xfcad0): undefined reference to `rfkill_set_sw_state'
net/built-in.o:(.debug_addr+0xfdc80): undefined reference to `rfkill_blocked'
Makefile:1009: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 1

```

ballsystemlord, 2019-04-02:
Changing RFKILL from a module to built-in helps, but is still not enough to get everything to build properly. Apparently, the symbols that need exporting are not exported.
```
drivers/built-in.o: In function `dwc3_rockchip_host_testmode_write':
drivers/usb/dwc3/dwc3-rockchip.c:265: undefined reference to `xhci_set_link_state'
drivers/built-in.o: In function `dwc3_rockchip_host_testmode_write':
drivers/usb/dwc3/dwc3-rockchip-inno.c:164: undefined reference to `xhci_set_link_state'
drivers/built-in.o: In function `sfc_erase_mtd':
drivers/rkflash/sfc_nor_mtd.c:69: undefined reference to `mtd_erase_callback'
drivers/built-in.o: In function `sfc_nor_mtd_init':
drivers/rkflash/sfc_nor_mtd.c:214: undefined reference to `mtd_device_parse_register'
drivers/built-in.o:(.debug_addr+0x197d40): undefined reference to `xhci_set_link_state'
drivers/built-in.o:(.debug_addr+0x1991e8): undefined reference to `xhci_set_link_state'
drivers/built-in.o:(.debug_addr+0x275e30): undefined reference to `mtd_erase_callback'
drivers/built-in.o:(.debug_addr+0x275f60): undefined reference to `mtd_device_parse_register'
Makefile:1009: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 1
```

ballsystemlord, 2019-04-02:
Changing USB_XHCI_HCD and USB_XHCI_PLATFORM from being built as modules to built in brings me closer...
```
drivers/built-in.o: In function `sfc_erase_mtd':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/rkflash/sfc_nor_mtd.c:69: undefined reference to `mtd_erase_callback'
drivers/built-in.o: In function `sfc_nor_mtd_init':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/rkflash/sfc_nor_mtd.c:214: undefined reference to `mtd_device_parse_register'
drivers/built-in.o:(.debug_addr+0x284270): undefined reference to `mtd_erase_callback'
drivers/built-in.o:(.debug_addr+0x2843a0): undefined reference to `mtd_device_parse_register'
Makefile:1009: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 1
```

ballsystemlord, 2019-04-02:
Disabling RK_SFC_NOR uncovers still more bugs:

```
  CC [M]  drivers/cpufreq/rockchip-cpufreq.o
  CC [M]  drivers/devfreq/rockchip_dmc.o
drivers/cpufreq/rockchip-cpufreq.c:190:5: error: redefinition of 'rockchip_cpufreq_get_scale'
 int rockchip_cpufreq_get_scale(int cpu)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/cpufreq/rockchip-cpufreq.c:18:0:
include/linux/cpufreq.h:714:19: note: previous definition of 'rockchip_cpufreq_get_scale' was here
 static inline int rockchip_cpufreq_get_scale(int cpu)
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/cpufreq/rockchip-cpufreq.c:202:5: error: redefinition of 'rockchip_cpufreq_set_scale_rate'
 int rockchip_cpufreq_set_scale_rate(struct device *dev, unsigned long rate)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/cpufreq/rockchip-cpufreq.c:18:0:
include/linux/cpufreq.h:719:19: note: previous definition of 'rockchip_cpufreq_set_scale_rate' was here
 static inline int rockchip_cpufreq_set_scale_rate(struct device *dev,
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/cpufreq/rockchip-cpufreq.c:215:5: error: redefinition of 'rockchip_cpufreq_check_rate_volt'
 int rockchip_cpufreq_check_rate_volt(struct device *dev)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/cpufreq/rockchip-cpufreq.c:18:0:
include/linux/cpufreq.h:725:19: note: previous definition of 'rockchip_cpufreq_check_rate_volt' was here
 static inline int rockchip_cpufreq_check_rate_volt(struct device *dev)
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/cpufreq/rockchip-cpufreq.c:231:5: error: redefinition of 'rockchip_cpufreq_set_temp_limit_rate'
 int rockchip_cpufreq_set_temp_limit_rate(struct device *dev, unsigned long rate)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/cpufreq/rockchip-cpufreq.c:18:0:
include/linux/cpufreq.h:730:19: note: previous definition of 'rockchip_cpufreq_set_temp_limit_rate' was here
 static inline int rockchip_cpufreq_set_temp_limit_rate(struct device *dev,
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/cpufreq/rockchip-cpufreq.c:244:5: error: redefinition of 'rockchip_cpufreq_update_policy'
 int rockchip_cpufreq_update_policy(struct device *dev)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/cpufreq/rockchip-cpufreq.c:18:0:
include/linux/cpufreq.h:736:19: note: previous definition of 'rockchip_cpufreq_update_policy' was here
 static inline int rockchip_cpufreq_update_policy(struct device *dev)
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/cpufreq/rockchip-cpufreq.c:259:5: error: redefinition of 'rockchip_cpufreq_update_cur_volt'
 int rockchip_cpufreq_update_cur_volt(struct device *dev)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/cpufreq/rockchip-cpufreq.c:18:0:
include/linux/cpufreq.h:741:19: note: previous definition of 'rockchip_cpufreq_update_cur_volt' was here
 static inline int rockchip_cpufreq_update_cur_volt(struct device *dev)
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/cpufreq/rockchip-cpufreq.c:468:14: error: redefinition of 'rockchip_cpufreq_adjust_target'
 unsigned int rockchip_cpufreq_adjust_target(int cpu, unsigned int freq)
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/cpufreq/rockchip-cpufreq.c:18:0:
include/linux/cpufreq.h:708:28: note: previous definition of 'rockchip_cpufreq_adjust_target' was here
 static inline unsigned int rockchip_cpufreq_adjust_target(int cpu,
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:283: recipe for target 'drivers/cpufreq/rockchip-cpufreq.o' failed
make[3]: *** [drivers/cpufreq/rockchip-cpufreq.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/cpufreq' failed
make[2]: *** [drivers/cpufreq] Error 2
make[2]: *** Waiting for unfinished jobs....
drivers/devfreq/rockchip_dmc.c:875:6: error: redefinition of 'rockchip_dmcfreq_lock'
 void rockchip_dmcfreq_lock(void)
      ^~~~~~~~~~~~~~~~~~~~~
In file included from drivers/devfreq/rockchip_dmc.c:49:0:
include/soc/rockchip/rockchip_dmc.h:28:20: note: previous definition of 'rockchip_dmcfreq_lock' was here
 static inline void rockchip_dmcfreq_lock(void)
                    ^~~~~~~~~~~~~~~~~~~~~
drivers/devfreq/rockchip_dmc.c:881:6: error: redefinition of 'rockchip_dmcfreq_unlock'
 void rockchip_dmcfreq_unlock(void)
      ^~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/devfreq/rockchip_dmc.c:49:0:
[... 63 lines cut ...]
  CC [M]  sound/soc/rockchip/rockchip_hdmi_dp.o
sound/soc/rockchip/rockchip_vad.c: In function 'snd_pcm_vad_read':
sound/soc/rockchip/rockchip_vad.c:193:10: error: implicit declaration of function 'snd_pcm_vad_avail' [-Werror=implicit-function-declaration]
  avail = snd_pcm_vad_avail(substream);
          ^~~~~~~~~~~~~~~~~
sound/soc/rockchip/rockchip_vad.c: At top level:
sound/soc/rockchip/rockchip_vad.c:257:19: error: conflicting types for 'snd_pcm_vad_avail'
 snd_pcm_uframes_t snd_pcm_vad_avail(struct snd_pcm_substream *substream)
                   ^~~~~~~~~~~~~~~~~
sound/soc/rockchip/rockchip_vad.c:193:10: note: previous implicit declaration of 'snd_pcm_vad_avail' was here
  avail = snd_pcm_vad_avail(substream);
          ^~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:277: recipe for target 'sound/soc/rockchip/rockchip_vad.o' failed
make[4]: *** [sound/soc/rockchip/rockchip_vad.o] Error 1
make[4]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'sound/soc/rockchip' failed
make[3]: *** [sound/soc/rockchip] Error 2
scripts/Makefile.build:484: recipe for target 'sound/soc' failed
make[2]: *** [sound/soc] Error 2
Makefile:1031: recipe for target 'sound' failed
make[1]: *** [sound] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-02:
Changing CPUFREQ_DT and ARM_ROCKCHIP_CPUFREQ from modules to built-in sill leaves more errors:
```
drivers/devfreq/rockchip_dmc.c:875:6: error: redefinition of 'rockchip_dmcfreq_lock'
 void rockchip_dmcfreq_lock(void)
      ^~~~~~~~~~~~~~~~~~~~~
In file included from drivers/devfreq/rockchip_dmc.c:49:0:
include/soc/rockchip/rockchip_dmc.h:28:20: note: previous definition of 'rockchip_dmcfreq_lock' was here
 static inline void rockchip_dmcfreq_lock(void)
                    ^~~~~~~~~~~~~~~~~~~~~
drivers/devfreq/rockchip_dmc.c:881:6: error: redefinition of 'rockchip_dmcfreq_unlock'
 void rockchip_dmcfreq_unlock(void)
      ^~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/devfreq/rockchip_dmc.c:49:0:
include/soc/rockchip/rockchip_dmc.h:32:20: note: previous definition of 'rockchip_dmcfreq_unlock' was here
 static inline void rockchip_dmcfreq_unlock(void)
                    ^~~~~~~~~~~~~~~~~~~~~~~
drivers/devfreq/rockchip_dmc.c:1691:5: error: redefinition of 'rockchip_dmcfreq_wait_complete'
 int rockchip_dmcfreq_wait_complete(void)
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/devfreq/rockchip_dmc.c:49:0:
include/soc/rockchip/rockchip_dmc.h:36:19: note: previous definition of 'rockchip_dmcfreq_wait_complete' was here
 static inline int rockchip_dmcfreq_wait_complete(void)
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/devfreq/rockchip_dmc.c:2602:6: error: redefinition of 'rockchip_dmcfreq_vop_bandwidth_update'
 void rockchip_dmcfreq_vop_bandwidth_update(struct devfreq *devfreq,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/devfreq/rockchip_dmc.c:49:0:
include/soc/rockchip/rockchip_dmc.h:49:1: note: previous definition of 'rockchip_dmcfreq_vop_bandwidth_update' was here
 rockchip_dmcfreq_vop_bandwidth_update(struct devfreq *devfreq,
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/devfreq/rockchip_dmc.c:2635:5: error: redefinition of 'rockchip_dmcfreq_vop_bandwidth_request'
 int rockchip_dmcfreq_vop_bandwidth_request(struct devfreq *devfreq,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/devfreq/rockchip_dmc.c:49:0:
include/soc/rockchip/rockchip_dmc.h:42:1: note: previous definition of 'rockchip_dmcfreq_vop_bandwidth_request' was here
 rockchip_dmcfreq_vop_bandwidth_request(struct devfreq *devfreq,
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:283: recipe for target 'drivers/devfreq/rockchip_dmc.o' failed
make[3]: *** [drivers/devfreq/rockchip_dmc.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/devfreq' failed
make[2]: *** [drivers/devfreq] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC [M]  sound/soc/codecs/cx20810.o
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from sound/soc/codecs/cx20810.c:14:0:
sound/soc/codecs/cx20810.c:161:26: error: 'cx20810_id' undeclared here (not in a function)
 MODULE_DEVICE_TABLE(i2c, cx20810_id);
                          ^
include/linux/module.h:223:21: note: in definition of macro 'MODULE_DEVICE_TABLE'
 extern const typeof(name) __mod_##type##__##name##_device_table  \
                     ^~~~
include/linux/module.h:223:27: error: '__mod_i2c__cx20810_id_device_table' aliased to undefined symbol 'cx20810_id'
 extern const typeof(name) __mod_##type##__##name##_device_table  \
                           ^
sound/soc/codecs/cx20810.c:161:1: note: in expansion of macro 'MODULE_DEVICE_TABLE'
 MODULE_DEVICE_TABLE(i2c, cx20810_id);
 ^~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:277: recipe for target 'sound/soc/codecs/cx20810.o' failed
make[4]: *** [sound/soc/codecs/cx20810.o] Error 1
scripts/Makefile.build:484: recipe for target 'sound/soc/codecs' failed
make[3]: *** [sound/soc/codecs] Error 2
scripts/Makefile.build:484: recipe for target 'sound/soc' failed
make[2]: *** [sound/soc] Error 2
Makefile:1031: recipe for target 'sound' failed
make[1]: *** [sound] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-03:
Setting ARM_ROCKCHIP_DMC_DEVFREQ to enabled vs. module still leaves much to be desired:
```
In file included from sound/soc/codecs/cx20810.c:14:0:
sound/soc/codecs/cx20810.c:161:26: error: 'cx20810_id' undeclared here (not in a function)
 MODULE_DEVICE_TABLE(i2c, cx20810_id);
                          ^
include/linux/module.h:223:21: note: in definition of macro 'MODULE_DEVICE_TABLE'
 extern const typeof(name) __mod_##type##__##name##_device_table  \
                     ^~~~
include/linux/module.h:223:27: error: '__mod_i2c__cx20810_id_device_table' aliased to undefined symbol 'cx20810_id'
 extern const typeof(name) __mod_##type##__##name##_device_table  \
                           ^
sound/soc/codecs/cx20810.c:161:1: note: in expansion of macro 'MODULE_DEVICE_TABLE'
 MODULE_DEVICE_TABLE(i2c, cx20810_id);
 ^~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:277: recipe for target 'sound/soc/codecs/cx20810.o' failed
make[4]: *** [sound/soc/codecs/cx20810.o] Error 1
scripts/Makefile.build:484: recipe for target 'sound/soc/codecs' failed
make[3]: *** [sound/soc/codecs] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'sound/soc' failed
make[2]: *** [sound/soc] Error 2
Makefile:1031: recipe for target 'sound' failed
make[1]: *** [sound] Error 2
make[1]: *** Waiting for unfinished jobs....
  CC [M]  drivers/headset_observe/rockchip_headset_core.o
  CC [M]  drivers/headset_observe/rk_headset.o
  LD [M]  drivers/gpu/arm/bifrost_for_linux/bifrost_kbase.o
drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.o:(__tracepoints+0x0): multiple definition of `__tracepoint_mali_PM_WAKE_WAITERS'
drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.o:(__tracepoints+0x0): first defined here
drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.c:1910: multiple definition of `__tracepoint_mali_PM_CA_SET_POLICY'
drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.c:64: first defined here
drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.c:1913: multiple definition of `__tracepoint_mali_PM_CURRENT_POLICY_TERM'
drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.c:71: first defined here
drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.c:1918: multiple definition of `__tracepoint_mali_PM_CURRENT_POLICY_INIT'
drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.c:78: first defined here
drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.c:1924: multiple definition of `__tracepoint_mali_PM_SET_POLICY'
drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.o:drivers/gpu/arm/bifrost_for_linux/./mali_linux_kbase_trace.h:27: first defined here
drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.c:1943: multiple definition of `__tracepoint_mali_PM_GPU_OFF'
drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.o:include/linux/perf_event.h:891: first defined here
drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.c:1950: multiple definition of `__tracepoint_mali_PM_GPU_ON'
drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.o:include/linux/trace_events.h:626: first defined here
drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.c:1969: multiple definition of `__tracepoint_mali_PM_CORES_CHANGE_AVAILABLE_TILER'
drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.o:drivers/gpu/arm/bifrost_for_linux/./mali_linux_kbase_trace.h:73: first defined here
drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.c:1985: multiple definition of `__tracepoint_mali_PM_CORES_CHANGE_AVAILABLE'
drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.o:drivers/gpu/arm/bifrost_for_linux/./mali_linux_kbase_trace.h:73: first defined here
drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.c:1987: multiple definition of `__tracepoint_mali_PM_CORES_AVAILABLE_TILER'
drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.o:include/linux/perf_event.h:893: first defined here
drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.c:2002: multiple definition of `__tracepoint_mali_PM_CORES_AVAILABLE'
drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.o:drivers/gpu/arm/bifrost_for_linux/./mali_linux_kbase_trace.h:73: first defined here
drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.o:drivers/gpu/arm/bifrost_for_linux/mali_kbase_core_linux.c:2014: multiple definition of `__tracepoint_mali_PM_RELEASE_CHANGE_SHADER_INUSE'
drivers/gpu/arm/bifrost_for_linux/mali_kbase_trace_timeline.o:drivers/gpu/arm/bifrost_for_linux/./mali_linux_kbase_trace.h:109: first defined here
[... cut ...]
  .bus = BUS_USB, .vendor = (ven), .product = (prod)
                                               ^~~~
scripts/Makefile.build:283: recipe for target 'drivers/hid/hid-rkvr.o' failed
make[3]: *** [drivers/hid/hid-rkvr.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/hid' failed
make[2]: *** [drivers/hid] Error 2
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-03:
Disabling MALI_BIFROST_TRACE_TIMELINE:
```
In file included from sound/soc/codecs/cx20810.c:14:0:
sound/soc/codecs/cx20810.c:161:26: error: 'cx20810_id' undeclared here (not in a function)
 MODULE_DEVICE_TABLE(i2c, cx20810_id);
                          ^
include/linux/module.h:223:21: note: in definition of macro 'MODULE_DEVICE_TABLE'
 extern const typeof(name) __mod_##type##__##name##_device_table  \
                     ^~~~
include/linux/module.h:223:27: error: '__mod_i2c__cx20810_id_device_table' aliased to undefined symbol 'cx20810_id'
 extern const typeof(name) __mod_##type##__##name##_device_table  \
                           ^
sound/soc/codecs/cx20810.c:161:1: note: in expansion of macro 'MODULE_DEVICE_TABLE'
 MODULE_DEVICE_TABLE(i2c, cx20810_id);
 ^~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:277: recipe for target 'sound/soc/codecs/cx20810.o' failed
make[4]: *** [sound/soc/codecs/cx20810.o] Error 1
scripts/Makefile.build:484: recipe for target 'sound/soc/codecs' failed
make[3]: *** [sound/soc/codecs] Error 2
make[3]: *** Waiting for unfinished jobs....
  CC [M]  drivers/headset_observe/rockchip_headset_core.o
scripts/Makefile.build:484: recipe for target 'sound/soc' failed
make[2]: *** [sound/soc] Error 2
Makefile:1031: recipe for target 'sound' failed
make[1]: *** [sound] Error 2
make[1]: *** Waiting for unfinished jobs....
  CC [M]  drivers/gpu/rogue_m/system/rk3368/rk_init.o
drivers/headset_observe/rockchip_headset_core.c:205:20: warning: 'rockchip_headset_exit' defined but not used [-Wunused-function]
error, forbidden warning:rockchip_headset_core.c:205
 static void __exit rockchip_headset_exit(void)
                    ^~~~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:283: recipe for target 'drivers/headset_observe/rockchip_headset_core.o' failed
make[3]: *** [drivers/headset_observe/rockchip_headset_core.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/headset_observe' failed
make[2]: *** [drivers/headset_observe] Error 2
make[2]: *** Waiting for unfinished jobs....
drivers/gpu/rogue_m/system/rk3368/rk_init.c:15:33: fatal error: linux/rockchip/dvfs.h: No such file or directory
 #include <linux/rockchip/dvfs.h>
                                 ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/gpu/rogue_m/system/rk3368/rk_init.o' failed
make[4]: *** [drivers/gpu/rogue_m/system/rk3368/rk_init.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/gpu/rogue_m' failed
make[3]: *** [drivers/gpu/rogue_m] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu' failed
make[2]: *** [drivers/gpu] Error 2
  CC [M]  drivers/hid/hid-rkvr.o
  CC [M]  drivers/hwmon/pwm-fan.o
  CC [M]  drivers/hwmon/shtc1.o
  CC [M]  drivers/hwmon/sis5595.o
  CC [M]  drivers/hwmon/smm665.o
  CC [M]  drivers/hwmon/smsc47m192.o
  CC [M]  drivers/hwmon/smsc47b397.o
  CC [M]  drivers/hwmon/smsc47m1.o
In file included from include/linux/list.h:8:0,
                 from include/linux/kobject.h:20,
                 from include/linux/device.h:17,
                 from include/linux/hwmon-sysfs.h:23,
                 from drivers/hwmon/pwm-fan.c:20:
drivers/hwmon/pwm-fan.c: In function '__set_pwm':
drivers/hwmon/pwm-fan.c:50:37: error: 'struct pwm_device' has no member named 'period'
  duty = DIV_ROUND_UP(pwm * (ctx->pwm->period - 1), MAX_PWM);
                                     ^
include/linux/kernel.h:74:30: note: in definition of macro 'DIV_ROUND_UP'
 #define DIV_ROUND_UP(n,d) (((n) + (d) - 1) / (d))
                              ^
drivers/hwmon/pwm-fan.c:51:43: error: 'struct pwm_device' has no member named 'period'
  ret = pwm_config(ctx->pwm, duty, ctx->pwm->period);
                                           ^~
drivers/hwmon/pwm-fan.c: In function 'pwm_fan_probe':
[... 25 lines cut ...]
                 from include/linux/hidraw.h:16,
                 from drivers/hid/hid-rkvr.c:21:
drivers/hid/hid-rkvr.c:1404:19: error: 'USB_VENDOR_ID_ROCKCHIP' undeclared here (not in a function)
  { HID_USB_DEVICE(USB_VENDOR_ID_ROCKCHIP, USB_DEVICE_ID_NANOC) },
                   ^
include/linux/hid.h:612:29: note: in definition of macro 'HID_USB_DEVICE'
  .bus = BUS_USB, .vendor = (ven), .product = (prod)
                             ^~~
drivers/hid/hid-rkvr.c:1404:43: error: 'USB_DEVICE_ID_NANOC' undeclared here (not in a function)
  { HID_USB_DEVICE(USB_VENDOR_ID_ROCKCHIP, USB_DEVICE_ID_NANOC) },
                                           ^
include/linux/hid.h:612:47: note: in definition of macro 'HID_USB_DEVICE'
  .bus = BUS_USB, .vendor = (ven), .product = (prod)
                                               ^~~~
scripts/Makefile.build:283: recipe for target 'drivers/hid/hid-rkvr.o' failed
make[3]: *** [drivers/hid/hid-rkvr.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/hid' failed
make[2]: *** [drivers/hid] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/hwmon' failed
make[2]: *** [drivers/hwmon] Error 2
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-03:
Disabling SND_SOC_CX20810:
```
drivers/headset_observe/rockchip_headset_core.c:205:20: warning: 'rockchip_headset_exit' defined but not used [-Wunused-function]
error, forbidden warning:rockchip_headset_core.c:205
 static void __exit rockchip_headset_exit(void)
                    ^~~~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:283: recipe for target 'drivers/headset_observe/rockchip_headset_core.o' failed
make[3]: *** [drivers/headset_observe/rockchip_headset_core.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/headset_observe' failed
make[2]: *** [drivers/headset_observe] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC [M]  sound/soc/codecs/es8328-i2c.o
  CC [M]  sound/soc/codecs/es8396.o
  CC [M]  sound/soc/codecs/fm1288.o
  CC [M]  sound/soc/codecs/gtm601.o
  CC [M]  sound/soc/codecs/gva_codec.o
  CC [M]  drivers/gpu/rogue_m/system/rk3368/rk_init.o
  CC [M]  sound/soc/codecs/max98090.o
  CC [M]  sound/soc/codecs/pcm1681.o
  CC [M]  sound/soc/codecs/pcm1792a.o
  CC [M]  sound/soc/codecs/pcm512x.o
  CC [M]  sound/soc/codecs/pcm512x-i2c.o
drivers/gpu/rogue_m/system/rk3368/rk_init.c:15:33: fatal error: linux/rockchip/dvfs.h: No such file or directory
 #include <linux/rockchip/dvfs.h>
                                 ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/gpu/rogue_m/system/rk3368/rk_init.o' failed
make[4]: *** [drivers/gpu/rogue_m/system/rk3368/rk_init.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/gpu/rogue_m' failed
make[3]: *** [drivers/gpu/rogue_m] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/gpu' failed
make[2]: *** [drivers/gpu] Error 2
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
make[1]: *** Waiting for unfinished jobs....
  CC [M]  sound/soc/codecs/pcm512x-spi.o
  CC [M]  sound/soc/codecs/rk1000_codec.o
  CC [M]  sound/soc/codecs/rk312x_codec.o
  CC [M]  sound/soc/codecs/rk3228_codec.o
  CC [M]  sound/soc/codecs/rk3308_codec.o
  CC [M]  sound/soc/codecs/rk3328_codec.o
  CC [M]  sound/soc/codecs/rk817_codec.o
  CC [M]  sound/soc/codecs/rl6231.o
  CC [M]  sound/soc/codecs/rt5616.o
sound/soc/codecs/rk3308_codec.c:3549:6: error: redefinition of 'rk3308_codec_set_jack_detect'
 void rk3308_codec_set_jack_detect(struct snd_soc_codec *codec,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sound/soc/codecs/rk3308_codec.c:46:0:
sound/soc/codecs/rk3308_codec_provider.h:27:20: note: previous definition of 'rk3308_codec_set_jack_detect' was here
 static inline void rk3308_codec_set_jack_detect(struct snd_soc_codec *codec,
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:277: recipe for target 'sound/soc/codecs/rk3308_codec.o' failed
make[4]: *** [sound/soc/codecs/rk3308_codec.o] Error 1
make[4]: *** Waiting for unfinished jobs....
  LD [M]  sound/soc/rockchip/snd-soc-rockchip-i2s.o
  LD [M]  sound/soc/rockchip/snd-soc-rockchip-i2s-tdm.o
  LD [M]  sound/soc/rockchip/snd-soc-rockchip-multi-dais.o
  LD [M]  sound/soc/rockchip/snd-soc-rockchip-pdm.o
  LD [M]  sound/soc/rockchip/snd-soc-rockchip-spdif.o
  LD [M]  sound/soc/rockchip/snd-soc-rockchip-spdifrx.o
  CC [M]  sound/soc/rockchip/rockchip_vad.o
  CC [M]  sound/soc/sunxi/sun4i-codec.o
  CC [M]  sound/soc/rockchip/rockchip_da7219.o
  CC [M]  sound/soc/rockchip/rockchip_hdmi_analog.o
  CC [M]  sound/soc/xtensa/xtfpga-i2s.o
scripts/Makefile.build:484: recipe for target 'sound/soc/codecs' failed
make[3]: *** [sound/soc/codecs] Error 2
make[3]: *** Waiting for unfinished jobs....
  CC [M]  sound/soc/rockchip/rockchip_hdmi_dp.o
  CC [M]  sound/soc/rockchip/rockchip_max98090.o
sound/soc/rockchip/rockchip_vad.c: In function 'snd_pcm_vad_read':
sound/soc/rockchip/rockchip_vad.c:193:10: error: implicit declaration of function 'snd_pcm_vad_avail' [-Werror=implicit-function-declaration]
  avail = snd_pcm_vad_avail(substream);
          ^~~~~~~~~~~~~~~~~
sound/soc/rockchip/rockchip_vad.c: At top level:
sound/soc/rockchip/rockchip_vad.c:257:19: error: conflicting types for 'snd_pcm_vad_avail'
 snd_pcm_uframes_t snd_pcm_vad_avail(struct snd_pcm_substream *substream)
                   ^~~~~~~~~~~~~~~~~
sound/soc/rockchip/rockchip_vad.c:193:10: note: previous implicit declaration of 'snd_pcm_vad_avail' was here
  avail = snd_pcm_vad_avail(substream);
          ^~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:277: recipe for target 'sound/soc/rockchip/rockchip_vad.o' failed
make[4]: *** [sound/soc/rockchip/rockchip_vad.o] Error 1
make[4]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'sound/soc/rockchip' failed
make[3]: *** [sound/soc/rockchip] Error 2
  LD [M]  sound/soc/xtensa/snd-soc-xtfpga-i2s.o
scripts/Makefile.build:484: recipe for target 'sound/soc' failed
make[2]: *** [sound/soc] Error 2
Makefile:1031: recipe for target 'sound' failed
make[1]: *** [sound] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-03:
Disabling RK_HEADSET:
```
sound/soc/codecs/rk3308_codec.c:3549:6: error: redefinition of 'rk3308_codec_set_jack_detect'
 void rk3308_codec_set_jack_detect(struct snd_soc_codec *codec,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from sound/soc/codecs/rk3308_codec.c:46:0:
sound/soc/codecs/rk3308_codec_provider.h:27:20: note: previous definition of 'rk3308_codec_set_jack_detect' was here
 static inline void rk3308_codec_set_jack_detect(struct snd_soc_codec *codec,
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
scripts/Makefile.build:277: recipe for target 'sound/soc/codecs/rk3308_codec.o' failed
make[4]: *** [sound/soc/codecs/rk3308_codec.o] Error 1
scripts/Makefile.build:484: recipe for target 'sound/soc/codecs' failed
make[3]: *** [sound/soc/codecs] Error 2
scripts/Makefile.build:484: recipe for target 'sound/soc' failed
make[2]: *** [sound/soc] Error 2
Makefile:1031: recipe for target 'sound' failed
make[1]: *** [sound] Error 2
make[1]: *** Waiting for unfinished jobs....
  CC [M]  drivers/hid/hid-rkvr.o
In file included from include/uapi/linux/hidraw.h:18:0,
                 from include/linux/hidraw.h:16,
                 from drivers/hid/hid-rkvr.c:21:
drivers/hid/hid-rkvr.c:1404:19: error: 'USB_VENDOR_ID_ROCKCHIP' undeclared here (not in a function)
  { HID_USB_DEVICE(USB_VENDOR_ID_ROCKCHIP, USB_DEVICE_ID_NANOC) },
                   ^
include/linux/hid.h:612:29: note: in definition of macro 'HID_USB_DEVICE'
  .bus = BUS_USB, .vendor = (ven), .product = (prod)
                             ^~~
drivers/hid/hid-rkvr.c:1404:43: error: 'USB_DEVICE_ID_NANOC' undeclared here (not in a function)
  { HID_USB_DEVICE(USB_VENDOR_ID_ROCKCHIP, USB_DEVICE_ID_NANOC) },
                                           ^
include/linux/hid.h:612:47: note: in definition of macro 'HID_USB_DEVICE'
  .bus = BUS_USB, .vendor = (ven), .product = (prod)
                                               ^~~~
scripts/Makefile.build:283: recipe for target 'drivers/hid/hid-rkvr.o' failed
make[3]: *** [drivers/hid/hid-rkvr.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/hid' failed
make[2]: *** [drivers/hid] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC [M]  drivers/gpu/rogue_m/system/rk3368/rk_init.o
drivers/gpu/rogue_m/system/rk3368/rk_init.c:15:33: fatal error: linux/rockchip/dvfs.h: No such file or directory
 #include <linux/rockchip/dvfs.h>
                                 ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/gpu/rogue_m/system/rk3368/rk_init.o' failed
make[4]: *** [drivers/gpu/rogue_m/system/rk3368/rk_init.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/gpu/rogue_m' failed
make[3]: *** [drivers/gpu/rogue_m] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu' failed
make[2]: *** [drivers/gpu] Error 2
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-03:
Setting SND_SOC_RK3308 from module to built-in.
```
In file included from include/uapi/linux/hidraw.h:18:0,
                 from include/linux/hidraw.h:16,
                 from drivers/hid/hid-rkvr.c:21:
drivers/hid/hid-rkvr.c:1404:19: error: 'USB_VENDOR_ID_ROCKCHIP' undeclared here (not in a function)
  { HID_USB_DEVICE(USB_VENDOR_ID_ROCKCHIP, USB_DEVICE_ID_NANOC) },
                   ^
include/linux/hid.h:612:29: note: in definition of macro 'HID_USB_DEVICE'
  .bus = BUS_USB, .vendor = (ven), .product = (prod)
                             ^~~
drivers/hid/hid-rkvr.c:1404:43: error: 'USB_DEVICE_ID_NANOC' undeclared here (not in a function)
  { HID_USB_DEVICE(USB_VENDOR_ID_ROCKCHIP, USB_DEVICE_ID_NANOC) },
                                           ^
include/linux/hid.h:612:47: note: in definition of macro 'HID_USB_DEVICE'
  .bus = BUS_USB, .vendor = (ven), .product = (prod)
                                               ^~~~
scripts/Makefile.build:283: recipe for target 'drivers/hid/hid-rkvr.o' failed
make[3]: *** [drivers/hid/hid-rkvr.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/hid' failed
make[2]: *** [drivers/hid] Error 2
make[2]: *** Waiting for unfinished jobs....
In file included from include/linux/list.h:8:0,
                 from include/linux/kobject.h:20,
                 from include/linux/device.h:17,
                 from include/linux/hwmon-sysfs.h:23,
                 from drivers/hwmon/pwm-fan.c:20:
drivers/hwmon/pwm-fan.c: In function '__set_pwm':
drivers/hwmon/pwm-fan.c:50:37: error: 'struct pwm_device' has no member named 'period'
  duty = DIV_ROUND_UP(pwm * (ctx->pwm->period - 1), MAX_PWM);
                                     ^
include/linux/kernel.h:74:30: note: in definition of macro 'DIV_ROUND_UP'
 #define DIV_ROUND_UP(n,d) (((n) + (d) - 1) / (d))
                              ^
drivers/hwmon/pwm-fan.c:51:43: error: 'struct pwm_device' has no member named 'period'
  ret = pwm_config(ctx->pwm, duty, ctx->pwm->period);
                                           ^~
drivers/hwmon/pwm-fan.c: In function 'pwm_fan_probe':
drivers/hwmon/pwm-fan.c:237:23: error: 'struct pwm_device' has no member named 'period'
  duty_cycle = ctx->pwm->period - 1;
                       ^~
drivers/hwmon/pwm-fan.c:240:49: error: 'struct pwm_device' has no member named 'period'
  ret = pwm_config(ctx->pwm, duty_cycle, ctx->pwm->period);
                                                 ^~
In file included from include/linux/list.h:8:0,
                 from include/linux/kobject.h:20,
                 from include/linux/device.h:17,
                 from include/linux/hwmon-sysfs.h:23,
                 from drivers/hwmon/pwm-fan.c:20:
drivers/hwmon/pwm-fan.c: In function 'pwm_fan_resume':
drivers/hwmon/pwm-fan.c:312:48: error: 'struct pwm_device' has no member named 'period'
  duty = DIV_ROUND_UP(ctx->pwm_value * (ctx->pwm->period - 1), MAX_PWM);
                                                ^
include/linux/kernel.h:74:30: note: in definition of macro 'DIV_ROUND_UP'
 #define DIV_ROUND_UP(n,d) (((n) + (d) - 1) / (d))
                              ^
drivers/hwmon/pwm-fan.c:313:43: error: 'struct pwm_device' has no member named 'period'
  ret = pwm_config(ctx->pwm, duty, ctx->pwm->period);
                                           ^~
  LD [M]  sound/soc/rockchip/snd-soc-rockchip-da7219.o
scripts/Makefile.build:283: recipe for target 'drivers/hwmon/pwm-fan.o' failed
make[3]: *** [drivers/hwmon/pwm-fan.o] Error 1
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/hwmon' failed
make[2]: *** [drivers/hwmon] Error 2
  LD [M]  sound/soc/rockchip/snd-soc-rockchip-hdmi-analog.o
  LD [M]  sound/soc/rockchip/snd-soc-rockchip-hdmi-dp.o
  LD [M]  sound/soc/rockchip/snd-soc-rockchip-max98090.o
  CC [M]  sound/soc/rockchip/rockchip_multicodecs.o
drivers/gpu/rogue_m/system/rk3368/rk_init.c:15:33: fatal error: linux/rockchip/dvfs.h: No such file or directory
 #include <linux/rockchip/dvfs.h>
                                 ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/gpu/rogue_m/system/rk3368/rk_init.o' failed
make[4]: *** [drivers/gpu/rogue_m/system/rk3368/rk_init.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/gpu/rogue_m' failed
make[3]: *** [drivers/gpu/rogue_m] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu' failed
make[2]: *** [drivers/gpu] Error 2
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
make[1]: *** Waiting for unfinished jobs....
  CC [M]  sound/soc/rockchip/rockchip_rt5645.o
sound/soc/rockchip/rockchip_vad.c: In function 'snd_pcm_vad_read':
sound/soc/rockchip/rockchip_vad.c:193:10: error: implicit declaration of function 'snd_pcm_vad_avail' [-Werror=implicit-function-declaration]
  avail = snd_pcm_vad_avail(substream);
          ^~~~~~~~~~~~~~~~~
sound/soc/rockchip/rockchip_vad.c: At top level:
sound/soc/rockchip/rockchip_vad.c:257:19: error: conflicting types for 'snd_pcm_vad_avail'
 snd_pcm_uframes_t snd_pcm_vad_avail(struct snd_pcm_substream *substream)
                   ^~~~~~~~~~~~~~~~~
sound/soc/rockchip/rockchip_vad.c:193:10: note: previous implicit declaration of 'snd_pcm_vad_avail' was here
  avail = snd_pcm_vad_avail(substream);
          ^~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:277: recipe for target 'sound/soc/rockchip/rockchip_vad.o' failed
make[4]: *** [sound/soc/rockchip/rockchip_vad.o] Error 1
make[4]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'sound/soc/rockchip' failed
make[3]: *** [sound/soc/rockchip] Error 2
scripts/Makefile.build:484: recipe for target 'sound/soc' failed
make[2]: *** [sound/soc] Error 2
Makefile:1031: recipe for target 'sound' failed
make[1]: *** [sound] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-03:
Disabling HID_RKVR:
```
sound/soc/rockchip/rockchip_vad.c: In function 'snd_pcm_vad_read':
sound/soc/rockchip/rockchip_vad.c:193:10: error: implicit declaration of function 'snd_pcm_vad_avail' [-Werror=implicit-function-declaration]
  avail = snd_pcm_vad_avail(substream);
          ^~~~~~~~~~~~~~~~~
sound/soc/rockchip/rockchip_vad.c: At top level:
sound/soc/rockchip/rockchip_vad.c:257:19: error: conflicting types for 'snd_pcm_vad_avail'
 snd_pcm_uframes_t snd_pcm_vad_avail(struct snd_pcm_substream *substream)
                   ^~~~~~~~~~~~~~~~~
sound/soc/rockchip/rockchip_vad.c:193:10: note: previous implicit declaration of 'snd_pcm_vad_avail' was here
  avail = snd_pcm_vad_avail(substream);
          ^~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:277: recipe for target 'sound/soc/rockchip/rockchip_vad.o' failed
make[4]: *** [sound/soc/rockchip/rockchip_vad.o] Error 1
scripts/Makefile.build:484: recipe for target 'sound/soc/rockchip' failed
make[3]: *** [sound/soc/rockchip] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'sound/soc' failed
make[2]: *** [sound/soc] Error 2
Makefile:1031: recipe for target 'sound' failed
make[1]: *** [sound] Error 2
make[1]: *** Waiting for unfinished jobs....
  CC [M]  drivers/gpu/rogue_m/system/rk3368/rk_init.o
  CC [M]  drivers/hwmon/pwm-fan.o
drivers/gpu/rogue_m/system/rk3368/rk_init.c:15:33: fatal error: linux/rockchip/dvfs.h: No such file or directory
 #include <linux/rockchip/dvfs.h>
                                 ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/gpu/rogue_m/system/rk3368/rk_init.o' failed
make[4]: *** [drivers/gpu/rogue_m/system/rk3368/rk_init.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/gpu/rogue_m' failed
make[3]: *** [drivers/gpu/rogue_m] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/gpu' failed
make[2]: *** [drivers/gpu] Error 2
make[2]: *** Waiting for unfinished jobs....
In file included from include/linux/list.h:8:0,
                 from include/linux/kobject.h:20,
                 from include/linux/device.h:17,
                 from include/linux/hwmon-sysfs.h:23,
                 from drivers/hwmon/pwm-fan.c:20:
drivers/hwmon/pwm-fan.c: In function '__set_pwm':
drivers/hwmon/pwm-fan.c:50:37: error: 'struct pwm_device' has no member named 'period'
  duty = DIV_ROUND_UP(pwm * (ctx->pwm->period - 1), MAX_PWM);
                                     ^
include/linux/kernel.h:74:30: note: in definition of macro 'DIV_ROUND_UP'
 #define DIV_ROUND_UP(n,d) (((n) + (d) - 1) / (d))
                              ^
drivers/hwmon/pwm-fan.c:51:43: error: 'struct pwm_device' has no member named 'period'
  ret = pwm_config(ctx->pwm, duty, ctx->pwm->period);
                                           ^~
drivers/hwmon/pwm-fan.c: In function 'pwm_fan_probe':
drivers/hwmon/pwm-fan.c:237:23: error: 'struct pwm_device' has no member named 'period'
  duty_cycle = ctx->pwm->period - 1;
                       ^~
drivers/hwmon/pwm-fan.c:240:49: error: 'struct pwm_device' has no member named 'period'
  ret = pwm_config(ctx->pwm, duty_cycle, ctx->pwm->period);
                                                 ^~
In file included from include/linux/list.h:8:0,
                 from include/linux/kobject.h:20,
                 from include/linux/device.h:17,
                 from include/linux/hwmon-sysfs.h:23,
                 from drivers/hwmon/pwm-fan.c:20:
drivers/hwmon/pwm-fan.c: In function 'pwm_fan_resume':
drivers/hwmon/pwm-fan.c:312:48: error: 'struct pwm_device' has no member named 'period'
  duty = DIV_ROUND_UP(ctx->pwm_value * (ctx->pwm->period - 1), MAX_PWM);
                                                ^
include/linux/kernel.h:74:30: note: in definition of macro 'DIV_ROUND_UP'
 #define DIV_ROUND_UP(n,d) (((n) + (d) - 1) / (d))
                              ^
drivers/hwmon/pwm-fan.c:313:43: error: 'struct pwm_device' has no member named 'period'
  ret = pwm_config(ctx->pwm, duty, ctx->pwm->period);
                                           ^~
scripts/Makefile.build:283: recipe for target 'drivers/hwmon/pwm-fan.o' failed
make[3]: *** [drivers/hwmon/pwm-fan.o] Error 1
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/hwmon' failed
make[2]: *** [drivers/hwmon] Error 2
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-03:
Disabling SND_SOC_ROCKCHIP_VAD:
```
drivers/gpu/rogue_m/system/rk3368/rk_init.c:15:33: fatal error: linux/rockchip/dvfs.h: No such file or directory
 #include <linux/rockchip/dvfs.h>
                                 ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/gpu/rogue_m/system/rk3368/rk_init.o' failed
make[4]: *** [drivers/gpu/rogue_m/system/rk3368/rk_init.o] Error 1
make[4]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/gpu/rogue_m' failed
make[3]: *** [drivers/gpu/rogue_m] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/gpu' failed
make[2]: *** [drivers/gpu] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC [M]  drivers/hwmon/pwm-fan.o
  CC [M]  drivers/hwmon/tc74.o
  CC [M]  drivers/hwmon/thmc50.o
  CC [M]  drivers/hwmon/tmp102.o
  CC [M]  drivers/hwmon/tmp103.o
  CC [M]  drivers/hwmon/tmp401.o
  CC [M]  drivers/hwmon/vexpress.o
  CC [M]  drivers/hwmon/tmp421.o
In file included from include/linux/list.h:8:0,
                 from include/linux/kobject.h:20,
                 from include/linux/device.h:17,
                 from include/linux/hwmon-sysfs.h:23,
                 from drivers/hwmon/pwm-fan.c:20:
drivers/hwmon/pwm-fan.c: In function '__set_pwm':
drivers/hwmon/pwm-fan.c:50:37: error: 'struct pwm_device' has no member named 'period'
  duty = DIV_ROUND_UP(pwm * (ctx->pwm->period - 1), MAX_PWM);
                                     ^
include/linux/kernel.h:74:30: note: in definition of macro 'DIV_ROUND_UP'
 #define DIV_ROUND_UP(n,d) (((n) + (d) - 1) / (d))
                              ^
drivers/hwmon/pwm-fan.c:51:43: error: 'struct pwm_device' has no member named 'period'
  ret = pwm_config(ctx->pwm, duty, ctx->pwm->period);
                                           ^~
drivers/hwmon/pwm-fan.c: In function 'pwm_fan_probe':
drivers/hwmon/pwm-fan.c:237:23: error: 'struct pwm_device' has no member named 'period'
  duty_cycle = ctx->pwm->period - 1;
                       ^~
drivers/hwmon/pwm-fan.c:240:49: error: 'struct pwm_device' has no member named 'period'
  ret = pwm_config(ctx->pwm, duty_cycle, ctx->pwm->period);
                                                 ^~
In file included from include/linux/list.h:8:0,
                 from include/linux/kobject.h:20,
                 from include/linux/device.h:17,
                 from include/linux/hwmon-sysfs.h:23,
                 from drivers/hwmon/pwm-fan.c:20:
drivers/hwmon/pwm-fan.c: In function 'pwm_fan_resume':
drivers/hwmon/pwm-fan.c:312:48: error: 'struct pwm_device' has no member named 'period'
  duty = DIV_ROUND_UP(ctx->pwm_value * (ctx->pwm->period - 1), MAX_PWM);
                                                ^
include/linux/kernel.h:74:30: note: in definition of macro 'DIV_ROUND_UP'
 #define DIV_ROUND_UP(n,d) (((n) + (d) - 1) / (d))
                              ^
drivers/hwmon/pwm-fan.c:313:43: error: 'struct pwm_device' has no member named 'period'
  ret = pwm_config(ctx->pwm, duty, ctx->pwm->period);
                                           ^~
scripts/Makefile.build:283: recipe for target 'drivers/hwmon/pwm-fan.o' failed
make[3]: *** [drivers/hwmon/pwm-fan.o] Error 1
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/hwmon' failed
make[2]: *** [drivers/hwmon] Error 2
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-03:
Disabling SENSORS_PWM_FAN:
```
drivers/gpu/rogue_m/system/rk3368/rk_init.c:15:33: fatal error: linux/rockchip/dvfs.h: No such file or directory
 #include <linux/rockchip/dvfs.h>
                                 ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/gpu/rogue_m/system/rk3368/rk_init.o' failed
make[4]: *** [drivers/gpu/rogue_m/system/rk3368/rk_init.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/gpu/rogue_m' failed
make[3]: *** [drivers/gpu/rogue_m] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu' failed
make[2]: *** [drivers/gpu] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
make[1]: *** Waiting for unfinished jobs....
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

This next one is proving to be really hard to find...

ballsystemlord, 2019-04-03:
I'll have to look more tomorrow.
In general it seems that one bug report is silly for all these problems, although most of them fall under the GPU problem that I first filed this bug under. So if you'd like me to split these up just tell me how you'd like me to do it.
Thanks for your help.

ballsystemlord, 2019-04-10:
After several days of searching I found that the option was POWERVR_ROGUE_M.
That's not all the problems in this source tree though:
```
In file included from drivers/media/common/siano/smscoreapi.h:40:0,
                 from drivers/media/common/siano/sms-cards.h:23,
                 from drivers/media/common/siano/sms-cards.c:20:
drivers/media/common/siano/smsir.h:33:0: warning: "IR_DEFAULT_TIMEOUT" redefined
error, forbidden warning:smsir.h:33
 #define IR_DEFAULT_TIMEOUT  100
 
In file included from drivers/media/common/siano/smsir.h:31:0,
                 from drivers/media/common/siano/smscoreapi.h:40,
                 from drivers/media/common/siano/sms-cards.h:23,
                 from drivers/media/common/siano/sms-cards.c:20:
include/media/rc-core.h:242:0: note: this is the location of the previous definition
 #define IR_DEFAULT_TIMEOUT MS_TO_NS(125)
 
scripts/Makefile.build:277: recipe for target 'drivers/media/common/siano/sms-cards.o' failed
make[5]: *** [drivers/media/common/siano/sms-cards.o] Error 1
make[5]: *** Waiting for unfinished jobs....
In file included from drivers/media/common/siano/smscoreapi.h:40:0,
                 from drivers/media/common/siano/smsendian.c:26:
drivers/media/common/siano/smsir.h:33:0: warning: "IR_DEFAULT_TIMEOUT" redefined
error, forbidden warning:smsir.h:33
 #define IR_DEFAULT_TIMEOUT  100
 
In file included from drivers/media/common/siano/smsir.h:31:0,
                 from drivers/media/common/siano/smscoreapi.h:40,
                 from drivers/media/common/siano/smsendian.c:26:
include/media/rc-core.h:242:0: note: this is the location of the previous definition
 #define IR_DEFAULT_TIMEOUT MS_TO_NS(125)
 
scripts/Makefile.build:277: recipe for target 'drivers/media/common/siano/smsendian.o' failed
make[5]: *** [drivers/media/common/siano/smsendian.o] Error 1
In file included from drivers/media/common/siano/smscoreapi.h:40:0,
                 from drivers/media/common/siano/smscoreapi.c:24:
drivers/media/common/siano/smsir.h:33:0: warning: "IR_DEFAULT_TIMEOUT" redefined
error, forbidden warning:smsir.h:33
 #define IR_DEFAULT_TIMEOUT  100
 
In file included from drivers/media/common/siano/smsir.h:31:0,
                 from drivers/media/common/siano/smscoreapi.h:40,
                 from drivers/media/common/siano/smscoreapi.c:24:
include/media/rc-core.h:242:0: note: this is the location of the previous definition
 #define IR_DEFAULT_TIMEOUT MS_TO_NS(125)
 
scripts/Makefile.build:277: recipe for target 'drivers/media/common/siano/smscoreapi.o' failed
make[5]: *** [drivers/media/common/siano/smscoreapi.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/media/common/siano' failed
make[4]: *** [drivers/media/common/siano] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/media/common' failed
make[3]: *** [drivers/media/common] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/media' failed
make[2]: *** [drivers/media] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-10:
Disabling SMS_SDIO_DRV SMS_USB_DRV SMS_SIANO_MDTV:
```
In file included from include/linux/kernel.h:13:0,
                 from include/linux/list.h:8,
                 from include/linux/kobject.h:20,
                 from include/linux/device.h:17,
                 from include/linux/platform_device.h:14,
                 from drivers/media/platform/rockchip-vpu/rockchip_vpu_common.h:27,
                 from drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.c:17:
drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.c: In function 'rockchip_vpu_vp8e_assemble_bitstream':
drivers/media/platform/rockchip-vpu/rockchip_vpu_common.h:474:13: warning: format '%d' expects argument of type 'int', but argument 6 has type 'size_t {aka long unsigned int}' [-Wformat=]
error, forbidden warning:rockchip_vpu_common.h:474
    pr_debug("%s:%d: " fmt,                 \
             ^
include/linux/printk.h:236:21: note: in definition of macro 'pr_fmt'
 #define pr_fmt(fmt) fmt
                     ^~~
include/linux/printk.h:283:2: note: in expansion of macro 'dynamic_pr_debug'
  dynamic_pr_debug(fmt, ##__VA_ARGS__)
  ^~~~~~~~~~~~~~~~
drivers/media/platform/rockchip-vpu/rockchip_vpu_common.h:474:4: note: in expansion of macro 'pr_debug'
    pr_debug("%s:%d: " fmt,                 \
    ^~~~~~~~
drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.c:413:2: note: in expansion of macro 'vpu_debug'
  vpu_debug(1, "%s: hdr_size = %d, ext_hdr_size = %d, dct_size = %d\n",
  ^~~~~~~~~
drivers/media/platform/rockchip-vpu/rockchip_vpu_common.h:474:13: warning: format '%d' expects argument of type 'int', but argument 7 has type 'size_t {aka long unsigned int}' [-Wformat=]
error, forbidden warning:rockchip_vpu_common.h:474
    pr_debug("%s:%d: " fmt,                 \
             ^
include/linux/printk.h:236:21: note: in definition of macro 'pr_fmt'
 #define pr_fmt(fmt) fmt
                     ^~~
include/linux/printk.h:283:2: note: in expansion of macro 'dynamic_pr_debug'
  dynamic_pr_debug(fmt, ##__VA_ARGS__)
  ^~~~~~~~~~~~~~~~
drivers/media/platform/rockchip-vpu/rockchip_vpu_common.h:474:4: note: in expansion of macro 'pr_debug'
    pr_debug("%s:%d: " fmt,                 \
    ^~~~~~~~
drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.c:413:2: note: in expansion of macro 'vpu_debug'
  vpu_debug(1, "%s: hdr_size = %d, ext_hdr_size = %d, dct_size = %d\n",
  ^~~~~~~~~
drivers/media/platform/rockchip-vpu/rockchip_vpu_common.h:474:13: warning: format '%d' expects argument of type 'int', but argument 8 has type 'size_t {aka long unsigned int}' [-Wformat=]
error, forbidden warning:rockchip_vpu_common.h:474
    pr_debug("%s:%d: " fmt,                 \
             ^
include/linux/printk.h:236:21: note: in definition of macro 'pr_fmt'
 #define pr_fmt(fmt) fmt
                     ^~~
include/linux/printk.h:283:2: note: in expansion of macro 'dynamic_pr_debug'
  dynamic_pr_debug(fmt, ##__VA_ARGS__)
  ^~~~~~~~~~~~~~~~
drivers/media/platform/rockchip-vpu/rockchip_vpu_common.h:474:4: note: in expansion of macro 'pr_debug'
    pr_debug("%s:%d: " fmt,                 \
    ^~~~~~~~
drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.c:413:2: note: in expansion of macro 'vpu_debug'
  vpu_debug(1, "%s: hdr_size = %d, ext_hdr_size = %d, dct_size = %d\n",
  ^~~~~~~~~
drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.c: In function 'rockchip_vpu_h264e_assemble_bitstream':
drivers/media/platform/rockchip-vpu/rockchip_vpu_common.h:474:13: warning: format '%u' expects argument of type 'unsigned int', but argument 6 has type 'size_t {aka long unsigned int}' [-Wformat=]
error, forbidden warning:rockchip_vpu_common.h:474
    pr_debug("%s:%d: " fmt,                 \
             ^
include/linux/printk.h:236:21: note: in definition of macro 'pr_fmt'
 #define pr_fmt(fmt) fmt
                     ^~~
include/linux/printk.h:283:2: note: in expansion of macro 'dynamic_pr_debug'
  dynamic_pr_debug(fmt, ##__VA_ARGS__)
  ^~~~~~~~~~~~~~~~
drivers/media/platform/rockchip-vpu/rockchip_vpu_common.h:474:4: note: in expansion of macro 'pr_debug'
[... 27 lines cut ...]
                     ^~~
include/linux/printk.h:283:2: note: in expansion of macro 'dynamic_pr_debug'
  dynamic_pr_debug(fmt, ##__VA_ARGS__)
  ^~~~~~~~~~~~~~~~
drivers/media/platform/rockchip-vpu/rockchip_vpu_common.h:474:4: note: in expansion of macro 'pr_debug'
    pr_debug("%s:%d: " fmt,                 \
    ^~~~~~~~
drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.c:455:2: note: in expansion of macro 'vpu_debug'
  vpu_debug(1, "%s: sps_size = %u, pps_size = %u, slices_size = %u\n",
  ^~~~~~~~~
scripts/Makefile.build:277: recipe for target 'drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.o' failed
make[5]: *** [drivers/media/platform/rockchip-vpu/rockchip_vpu_hw.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/media/platform/rockchip-vpu' failed
make[4]: *** [drivers/media/platform/rockchip-vpu] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/media/platform' failed
make[3]: *** [drivers/media/platform] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/media' failed
make[2]: *** [drivers/media] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-10:
Bad news. The VPU code VIDEO_ROCKCHIP_VPU has to be disabled. Isn't that rather vital to the SoCs?
Anyway there's still more broken stuff:
```
drivers/pwm/pwm-cros-ec.c: In function 'cros_ec_pwm_set_duty':
drivers/pwm/pwm-cros-ec.c:40:33: error: field 'params' has incomplete type
   struct ec_params_pwm_set_duty params;
                                 ^~~~~~
drivers/pwm/pwm-cros-ec.c:48:17: error: 'EC_CMD_PWM_SET_DUTY' undeclared (first use in this function)
  msg->command = EC_CMD_PWM_SET_DUTY;
                 ^~~~~~~~~~~~~~~~~~~
drivers/pwm/pwm-cros-ec.c:48:17: note: each undeclared identifier is reported only once for each function it appears in
drivers/pwm/pwm-cros-ec.c:50:24: error: dereferencing pointer to incomplete type 'struct ec_params_pwm_set_duty'
  msg->outsize = sizeof(*params);
                        ^~~~~~~
drivers/pwm/pwm-cros-ec.c:53:21: error: 'EC_PWM_TYPE_GENERIC' undeclared (first use in this function)
  params->pwm_type = EC_PWM_TYPE_GENERIC;
                     ^~~~~~~~~~~~~~~~~~~
drivers/pwm/pwm-cros-ec.c: In function '__cros_ec_pwm_get_duty':
drivers/pwm/pwm-cros-ec.c:65:34: error: field 'params' has incomplete type
    struct ec_params_pwm_get_duty params;
                                  ^~~~~~
drivers/pwm/pwm-cros-ec.c:66:36: error: field 'resp' has incomplete type
    struct ec_response_pwm_get_duty resp;
                                    ^~~~
drivers/pwm/pwm-cros-ec.c:77:17: error: 'EC_CMD_PWM_GET_DUTY' undeclared (first use in this function)
  msg->command = EC_CMD_PWM_GET_DUTY;
                 ^~~~~~~~~~~~~~~~~~~
drivers/pwm/pwm-cros-ec.c:78:23: error: dereferencing pointer to incomplete type 'struct ec_params_pwm_get_duty'
  msg->insize = sizeof(*params);
                       ^~~~~~~
drivers/pwm/pwm-cros-ec.c:79:24: error: dereferencing pointer to incomplete type 'struct ec_response_pwm_get_duty'
  msg->outsize = sizeof(*resp);
                        ^~~~~
drivers/pwm/pwm-cros-ec.c:81:21: error: 'EC_PWM_TYPE_GENERIC' undeclared (first use in this function)
  params->pwm_type = EC_PWM_TYPE_GENERIC;
                     ^~~~~~~~~~~~~~~~~~~
drivers/pwm/pwm-cros-ec.c: In function 'cros_ec_pwm_apply':
drivers/pwm/pwm-cros-ec.c:105:23: error: 'EC_PWM_MAX_DUTY' undeclared (first use in this function)
  if (state->period != EC_PWM_MAX_DUTY)
                       ^~~~~~~~~~~~~~~
drivers/pwm/pwm-cros-ec.c: In function 'cros_ec_pwm_get_state':
drivers/pwm/pwm-cros-ec.c:130:18: error: 'EC_PWM_MAX_DUTY' undeclared (first use in this function)
  state->period = EC_PWM_MAX_DUTY;
                  ^~~~~~~~~~~~~~~
drivers/pwm/pwm-cros-ec.c: In function 'cros_ec_pwm_xlate':
drivers/pwm/pwm-cros-ec.c:149:21: error: 'EC_PWM_MAX_DUTY' undeclared (first use in this function)
  pwm->args.period = EC_PWM_MAX_DUTY;
                     ^~~~~~~~~~~~~~~
drivers/pwm/pwm-cros-ec.c: In function '__cros_ec_pwm_get_duty':
drivers/pwm/pwm-cros-ec.c:91:1: warning: control reaches end of non-void function [-Wreturn-type]
error, forbidden warning:pwm-cros-ec.c:91
 }
 ^
scripts/Makefile.build:283: recipe for target 'drivers/pwm/pwm-cros-ec.o' failed
make[3]: *** [drivers/pwm/pwm-cros-ec.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/pwm' failed
make[2]: *** [drivers/pwm] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC [M]  drivers/media/video/rk30_camera_pingpong.o
  CC [M]  drivers/media/video/rk_camsys/camsys_drv.o
drivers/media/video/rk30_camera_pingpong.c:40:34: fatal error: linux/rockchip/iomap.h: No such file or directory
 #include <linux/rockchip/iomap.h>
                                  ^
compilation terminated.
scripts/Makefile.build:283: recipe for target 'drivers/media/video/rk30_camera_pingpong.o' failed
make[4]: *** [drivers/media/video/rk30_camera_pingpong.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/media/video' failed
make[3]: *** [drivers/media/video] Error 2
make[3]: *** Waiting for unfinished jobs....
  CC [M]  drivers/media/video/rk_camsys/camsys_marvin.o
[... 103 lines cut ...]
error, forbidden warning:camsys_cif.c:219
    camsys_dev->devmems.registermem->vir_base +
    ^~~~~~~~~~
In file included from include/linux/io.h:25:0,
                 from drivers/media/video/rk_camsys/camsys_internal.h:13,
                 from drivers/media/video/rk_camsys/camsys_cif.h:5,
                 from drivers/media/video/rk_camsys/camsys_cif.c:2:
./arch/arm64/include/asm/io.h:52:22: note: expected 'volatile void *' but argument is of type 'long unsigned int'
 #define __raw_writel __raw_writel
                      ^
./arch/arm64/include/asm/io.h:53:20: note: in expansion of macro '__raw_writel'
 static inline void __raw_writel(u32 val, volatile void __iomem *addr)
                    ^~~~~~~~~~~~
scripts/Makefile.build:283: recipe for target 'drivers/media/video/rk_camsys/camsys_cif.o' failed
make[4]: *** [drivers/media/video/rk_camsys/camsys_cif.o] Error 1
make[4]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/media/video/rk_camsys' failed
make[3]: *** [drivers/media/video/rk_camsys] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/media' failed
make[2]: *** [drivers/media] Error 2
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-10:
Disabling CROS_EC_CHARDEV , CROS_EC_LPC , MFD_CROS_EC_I2C , MFD_CROS_EC_SPI , CHROME_PLATFORMS:
```
drivers/media/video/rk30_camera_pingpong.c:40:34: fatal error: linux/rockchip/iomap.h: No such file or directory
 #include <linux/rockchip/iomap.h>
                                  ^
compilation terminated.
scripts/Makefile.build:283: recipe for target 'drivers/media/video/rk30_camera_pingpong.o' failed
make[4]: *** [drivers/media/video/rk30_camera_pingpong.o] Error 1
make[4]: *** Waiting for unfinished jobs....
drivers/media/video/rk_camsys/camsys_cif.c: In function 'camsys_cif_iomux_cb':
drivers/media/video/rk_camsys/camsys_cif.c:86:37: error: 'RK_GRF_VIRT' undeclared (first use in this function)
  __raw_writel(((1<<1)|(1<<(1+16))), RK_GRF_VIRT+0x0380);
                                     ^~~~~~~~~~~
drivers/media/video/rk_camsys/camsys_cif.c:86:37: note: each undeclared identifier is reported only once for each function it appears in
drivers/media/video/rk_camsys/camsys_cif.c:9:15: warning: unused variable 'cif_vol_sel' [-Wunused-variable]
error, forbidden warning:camsys_cif.c:9
  unsigned int cif_vol_sel;
               ^~~~~~~~~~~
drivers/media/video/rk_camsys/camsys_cif.c: In function 'camsys_cif_irq':
drivers/media/video/rk_camsys/camsys_cif.c:196:26: warning: passing argument 1 of '__raw_readl' makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:camsys_cif.c:196
     intsta = __raw_readl(camsys_dev->devmems.registermem->vir_base +
                          ^~~~~~~~~~
In file included from include/linux/io.h:25:0,
                 from drivers/media/video/rk_camsys/camsys_internal.h:13,
                 from drivers/media/video/rk_camsys/camsys_cif.h:5,
                 from drivers/media/video/rk_camsys/camsys_cif.c:2:
./arch/arm64/include/asm/io.h:87:21: note: expected 'const volatile void *' but argument is of type 'long unsigned int'
 #define __raw_readl __raw_readl
                     ^
./arch/arm64/include/asm/io.h:88:19: note: in expansion of macro '__raw_readl'
 static inline u32 __raw_readl(const volatile void __iomem *addr)
                   ^~~~~~~~~~~
drivers/media/video/rk_camsys/camsys_cif.c:198:26: warning: passing argument 1 of '__raw_readl' makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:camsys_cif.c:198
     frmsta = __raw_readl(camsys_dev->devmems.registermem->vir_base +
                          ^~~~~~~~~~
In file included from include/linux/io.h:25:0,
                 from drivers/media/video/rk_camsys/camsys_internal.h:13,
                 from drivers/media/video/rk_camsys/camsys_cif.h:5,
                 from drivers/media/video/rk_camsys/camsys_cif.c:2:
./arch/arm64/include/asm/io.h:87:21: note: expected 'const volatile void *' but argument is of type 'long unsigned int'
 #define __raw_readl __raw_readl
                     ^
./arch/arm64/include/asm/io.h:88:19: note: in expansion of macro '__raw_readl'
 static inline u32 __raw_readl(const volatile void __iomem *addr)
                   ^~~~~~~~~~~
drivers/media/video/rk_camsys/camsys_cif.c:204:4: warning: passing argument 2 of '__raw_writel' makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:camsys_cif.c:204
    camsys_dev->devmems.registermem->vir_base +
    ^~~~~~~~~~
In file included from include/linux/io.h:25:0,
                 from drivers/media/video/rk_camsys/camsys_internal.h:13,
                 from drivers/media/video/rk_camsys/camsys_cif.h:5,
                 from drivers/media/video/rk_camsys/camsys_cif.c:2:
./arch/arm64/include/asm/io.h:52:22: note: expected 'volatile void *' but argument is of type 'long unsigned int'
 #define __raw_writel __raw_writel
                      ^
./arch/arm64/include/asm/io.h:53:20: note: in expansion of macro '__raw_writel'
 static inline void __raw_writel(u32 val, volatile void __iomem *addr)
                    ^~~~~~~~~~~~
drivers/media/video/rk_camsys/camsys_cif.c:207:4: warning: passing argument 2 of '__raw_writel' makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:camsys_cif.c:207
    camsys_dev->devmems.registermem->vir_base +
    ^~~~~~~~~~
In file included from include/linux/io.h:25:0,
                 from drivers/media/video/rk_camsys/camsys_internal.h:13,
                 from drivers/media/video/rk_camsys/camsys_cif.h:5,
                 from drivers/media/video/rk_camsys/camsys_cif.c:2:
./arch/arm64/include/asm/io.h:52:22: note: expected 'volatile void *' but argument is of type 'long unsigned int'
[... 37 lines cut ...]
In file included from include/linux/io.h:25:0,
                 from drivers/media/video/rk_camsys/camsys_internal.h:13,
                 from drivers/media/video/rk_camsys/camsys_cif.h:5,
                 from drivers/media/video/rk_camsys/camsys_cif.c:2:
./arch/arm64/include/asm/io.h:52:22: note: expected 'volatile void *' but argument is of type 'long unsigned int'
 #define __raw_writel __raw_writel
                      ^
./arch/arm64/include/asm/io.h:53:20: note: in expansion of macro '__raw_writel'
 static inline void __raw_writel(u32 val, volatile void __iomem *addr)
                    ^~~~~~~~~~~~
scripts/Makefile.build:283: recipe for target 'drivers/media/video/rk_camsys/camsys_cif.o' failed
make[4]: *** [drivers/media/video/rk_camsys/camsys_cif.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/media/video/rk_camsys' failed
make[3]: *** [drivers/media/video/rk_camsys] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/media/video' failed
make[3]: *** [drivers/media/video] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/media' failed
make[2]: *** [drivers/media] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-10:
Disabling RK30_CAMERA_PINGPONG (Don't I need that for my cameras?):
```
  CC [M]  drivers/media/video/rk_camsys/camsys_cif.o
drivers/media/video/rk_camsys/camsys_cif.c: In function 'camsys_cif_iomux_cb':
drivers/media/video/rk_camsys/camsys_cif.c:86:37: error: 'RK_GRF_VIRT' undeclared (first use in this function)
  __raw_writel(((1<<1)|(1<<(1+16))), RK_GRF_VIRT+0x0380);
                                     ^~~~~~~~~~~
drivers/media/video/rk_camsys/camsys_cif.c:86:37: note: each undeclared identifier is reported only once for each function it appears in
drivers/media/video/rk_camsys/camsys_cif.c:9:15: warning: unused variable 'cif_vol_sel' [-Wunused-variable]
error, forbidden warning:camsys_cif.c:9
  unsigned int cif_vol_sel;
               ^~~~~~~~~~~
drivers/media/video/rk_camsys/camsys_cif.c: In function 'camsys_cif_irq':
drivers/media/video/rk_camsys/camsys_cif.c:196:26: warning: passing argument 1 of '__raw_readl' makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:camsys_cif.c:196
     intsta = __raw_readl(camsys_dev->devmems.registermem->vir_base +
                          ^~~~~~~~~~
In file included from include/linux/io.h:25:0,
                 from drivers/media/video/rk_camsys/camsys_internal.h:13,
                 from drivers/media/video/rk_camsys/camsys_cif.h:5,
                 from drivers/media/video/rk_camsys/camsys_cif.c:2:
./arch/arm64/include/asm/io.h:87:21: note: expected 'const volatile void *' but argument is of type 'long unsigned int'
 #define __raw_readl __raw_readl
                     ^
./arch/arm64/include/asm/io.h:88:19: note: in expansion of macro '__raw_readl'
 static inline u32 __raw_readl(const volatile void __iomem *addr)
                   ^~~~~~~~~~~
drivers/media/video/rk_camsys/camsys_cif.c:198:26: warning: passing argument 1 of '__raw_readl' makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:camsys_cif.c:198
     frmsta = __raw_readl(camsys_dev->devmems.registermem->vir_base +
                          ^~~~~~~~~~
In file included from include/linux/io.h:25:0,
                 from drivers/media/video/rk_camsys/camsys_internal.h:13,
                 from drivers/media/video/rk_camsys/camsys_cif.h:5,
                 from drivers/media/video/rk_camsys/camsys_cif.c:2:
./arch/arm64/include/asm/io.h:87:21: note: expected 'const volatile void *' but argument is of type 'long unsigned int'
 #define __raw_readl __raw_readl
                     ^
./arch/arm64/include/asm/io.h:88:19: note: in expansion of macro '__raw_readl'
 static inline u32 __raw_readl(const volatile void __iomem *addr)
                   ^~~~~~~~~~~
drivers/media/video/rk_camsys/camsys_cif.c:204:4: warning: passing argument 2 of '__raw_writel' makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:camsys_cif.c:204
    camsys_dev->devmems.registermem->vir_base +
    ^~~~~~~~~~
In file included from include/linux/io.h:25:0,
                 from drivers/media/video/rk_camsys/camsys_internal.h:13,
                 from drivers/media/video/rk_camsys/camsys_cif.h:5,
                 from drivers/media/video/rk_camsys/camsys_cif.c:2:
./arch/arm64/include/asm/io.h:52:22: note: expected 'volatile void *' but argument is of type 'long unsigned int'
 #define __raw_writel __raw_writel
                      ^
./arch/arm64/include/asm/io.h:53:20: note: in expansion of macro '__raw_writel'
 static inline void __raw_writel(u32 val, volatile void __iomem *addr)
                    ^~~~~~~~~~~~
drivers/media/video/rk_camsys/camsys_cif.c:207:4: warning: passing argument 2 of '__raw_writel' makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:camsys_cif.c:207
    camsys_dev->devmems.registermem->vir_base +
    ^~~~~~~~~~
In file included from include/linux/io.h:25:0,
                 from drivers/media/video/rk_camsys/camsys_internal.h:13,
                 from drivers/media/video/rk_camsys/camsys_cif.h:5,
                 from drivers/media/video/rk_camsys/camsys_cif.c:2:
./arch/arm64/include/asm/io.h:52:22: note: expected 'volatile void *' but argument is of type 'long unsigned int'
 #define __raw_writel __raw_writel
                      ^
./arch/arm64/include/asm/io.h:53:20: note: in expansion of macro '__raw_writel'
 static inline void __raw_writel(u32 val, volatile void __iomem *addr)
                    ^~~~~~~~~~~~
drivers/media/video/rk_camsys/camsys_cif.c:213:4: warning: passing argument 2 of '__raw_writel' makes pointer from integer without a cast [-Wint-conversion]
[... 29 lines cut ...]
    camsys_dev->devmems.registermem->vir_base +
    ^~~~~~~~~~
In file included from include/linux/io.h:25:0,
                 from drivers/media/video/rk_camsys/camsys_internal.h:13,
                 from drivers/media/video/rk_camsys/camsys_cif.h:5,
                 from drivers/media/video/rk_camsys/camsys_cif.c:2:
./arch/arm64/include/asm/io.h:52:22: note: expected 'volatile void *' but argument is of type 'long unsigned int'
 #define __raw_writel __raw_writel
                      ^
./arch/arm64/include/asm/io.h:53:20: note: in expansion of macro '__raw_writel'
 static inline void __raw_writel(u32 val, volatile void __iomem *addr)
                    ^~~~~~~~~~~~
scripts/Makefile.build:283: recipe for target 'drivers/media/video/rk_camsys/camsys_cif.o' failed
make[4]: *** [drivers/media/video/rk_camsys/camsys_cif.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/media/video/rk_camsys' failed
make[3]: *** [drivers/media/video/rk_camsys] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/media' failed
make[2]: *** [drivers/media] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-10:
Disabling CAMSYS_CIF (Don't I need this also?):
```
drivers/staging/iio/imu/inv_mpu/inv_mpu_i2c.o: In function `init_module':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_i2c.c:744: multiple definition of `init_module'
drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.o:/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.c:585: first defined here
drivers/staging/iio/imu/inv_mpu/inv_mpu_i2c.o: In function `cleanup_module':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_i2c.c:628: multiple definition of `cleanup_module'
drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.o:/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.c:101: first defined here
drivers/staging/iio/imu/inv_mpu/inv_mpu_i2c.o:/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_i2c.c:640: multiple definition of `__mod_of__inv_mpu_of_match_device_table'
drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.o:/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.c:143: first defined here
drivers/staging/iio/imu/inv_mpu/inv_mpu_i2c.o:/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_i2c.c:135: multiple definition of `__mod_i2c__inv_mpu_id_device_table'
drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.o:/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.c:243: first defined here
./../prebuilts/gcc/linux-x86/aarch64/gcc-linaro-6.3.1-2017.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-ld: Warning: size of symbol `__mod_i2c__inv_mpu_id_device_table' changed from 360 in drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.o to 288 in drivers/staging/iio/imu/inv_mpu/inv_mpu_i2c.o
drivers/staging/iio/imu/inv_mpu/inv_mpu_hid.o: In function `init_module':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_hid.c:298: multiple definition of `init_module'
drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.o:/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.c:101: first defined here
drivers/staging/iio/imu/inv_mpu/inv_mpu_hid.o: In function `cleanup_module':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_hid.c:36: multiple definition of `cleanup_module'
drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.o:/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.c:101: first defined here
drivers/staging/iio/imu/inv_mpu/inv_mpu_hid.o:/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_hid.c:222: multiple definition of `__mod_of__inv_mpu_of_match_device_table'
drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.o:/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/staging/iio/imu/inv_mpu/inv_mpu_spi.c:143: first defined here
scripts/Makefile.build:472: recipe for target 'drivers/staging/iio/imu/inv_mpu/inv-mpu-iio.o' failed
make[5]: *** [drivers/staging/iio/imu/inv_mpu/inv-mpu-iio.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/staging/iio/imu/inv_mpu' failed
make[4]: *** [drivers/staging/iio/imu/inv_mpu] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/staging/iio' failed
make[3]: *** [drivers/staging/iio] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/staging' failed
make[2]: *** [drivers/staging] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-10:
Disabling INV_MPU_IIO (which is enabled in your own original config (do I need it?)):
```
  CC [M]  drivers/usb/dwc3/dwc3-pci.o
drivers/usb/dwc3/dwc3-pci.c: In function 'dwc3_pci_quirks':
drivers/usb/dwc3/dwc3-pci.c:57:4: error: implicit declaration of function 'PROPERTY_ENTRY_BOOL' [-Werror=implicit-function-declaration]
    PROPERTY_ENTRY_BOOL("snps,has-lpm-erratum"),
    ^~~~~~~~~~~~~~~~~~~
drivers/usb/dwc3/dwc3-pci.c:57:4: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:dwc3-pci.c:57
drivers/usb/dwc3/dwc3-pci.c:57:4: note: (near initialization for 'properties[0].name')
drivers/usb/dwc3/dwc3-pci.c:58:4: error: implicit declaration of function 'PROPERTY_ENTRY_U8' [-Werror=implicit-function-declaration]
    PROPERTY_ENTRY_U8("snps,lpm-nyet-threshold", 0xf),
    ^~~~~~~~~~~~~~~~~
drivers/usb/dwc3/dwc3-pci.c:61:4: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:dwc3-pci.c:61
    PROPERTY_ENTRY_BOOL("snps,req_p1p2p3_quirk"),
    ^~~~~~~~~~~~~~~~~~~
drivers/usb/dwc3/dwc3-pci.c:61:4: note: (near initialization for 'properties[0].<anonymous>.pointer.raw_data')
drivers/usb/dwc3/dwc3-pci.c:62:4: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:dwc3-pci.c:62
    PROPERTY_ENTRY_BOOL("snps,del_p1p2p3_quirk"),
    ^~~~~~~~~~~~~~~~~~~
drivers/usb/dwc3/dwc3-pci.c:62:4: note: (near initialization for 'properties[1].name')
drivers/usb/dwc3/dwc3-pci.c:66:4: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:dwc3-pci.c:66
    PROPERTY_ENTRY_BOOL("snps,tx_de_emphasis_quirk"),
    ^~~~~~~~~~~~~~~~~~~
drivers/usb/dwc3/dwc3-pci.c:66:4: note: (near initialization for 'properties[1].<anonymous>.pointer.raw_data')
drivers/usb/dwc3/dwc3-pci.c:67:4: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:dwc3-pci.c:67
    PROPERTY_ENTRY_U8("snps,tx_de_emphasis", 1),
    ^~~~~~~~~~~~~~~~~
drivers/usb/dwc3/dwc3-pci.c:67:4: note: (near initialization for 'properties[2].name')
drivers/usb/dwc3/dwc3-pci.c:56:40: warning: missing braces around initializer [-Wmissing-braces]
error, forbidden warning:dwc3-pci.c:56
   struct property_entry properties[] = {
                                        ^
drivers/usb/dwc3/dwc3-pci.c:56:40: note: (near initialization for 'properties')
drivers/usb/dwc3/dwc3-pci.c:78:10: error: implicit declaration of function 'platform_device_add_properties' [-Werror=implicit-function-declaration]
   return platform_device_add_properties(dwc3, properties);
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/usb/dwc3/dwc3-pci.c:85:4: error: implicit declaration of function 'PROPERTY_ENTRY_STRING' [-Werror=implicit-function-declaration]
    PROPERTY_ENTRY_STRING("dr-mode", "peripheral"),
    ^~~~~~~~~~~~~~~~~~~~~
drivers/usb/dwc3/dwc3-pci.c:85:4: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:dwc3-pci.c:85
drivers/usb/dwc3/dwc3-pci.c:85:4: note: (near initialization for 'properties[0].name')
drivers/usb/dwc3/dwc3-pci.c:86:4: warning: braces around scalar initializer
error, forbidden warning:dwc3-pci.c:86
    { }
    ^
drivers/usb/dwc3/dwc3-pci.c:86:4: note: (near initialization for 'properties[0].length')
drivers/usb/dwc3/dwc3-pci.c:86:4: error: empty scalar initializer
drivers/usb/dwc3/dwc3-pci.c:86:4: note: (near initialization for 'properties[0].length')
drivers/usb/dwc3/dwc3-pci.c:128:4: warning: initialization makes pointer from integer without a cast [-Wint-conversion]
error, forbidden warning:dwc3-pci.c:128
    PROPERTY_ENTRY_BOOL("snps,usb3_lpm_capable"),
    ^~~~~~~~~~~~~~~~~~~
drivers/usb/dwc3/dwc3-pci.c:128:4: note: (near initialization for 'properties[0].name')
drivers/usb/dwc3/dwc3-pci.c:131:4: warning: braces around scalar initializer
error, forbidden warning:dwc3-pci.c:131
    { },
    ^
drivers/usb/dwc3/dwc3-pci.c:131:4: note: (near initialization for 'properties[0].is_string')
drivers/usb/dwc3/dwc3-pci.c:131:4: error: empty scalar initializer
drivers/usb/dwc3/dwc3-pci.c:131:4: note: (near initialization for 'properties[0].is_string')
cc1: some warnings being treated as errors
scripts/Makefile.build:283: recipe for target 'drivers/usb/dwc3/dwc3-pci.o' failed
make[4]: *** [drivers/usb/dwc3/dwc3-pci.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/usb/dwc3' failed
[... 310 lines cut ...]
 static int rk_fb_alloc_buffer(struct fb_info *fbi)
            ^~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:283: recipe for target 'drivers/video/rockchip/rk_fb.o' failed
make[4]: *** [drivers/video/rockchip/rk_fb.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[3]: *** [drivers/video/rockchip] Error 2
make[3]: *** Waiting for unfinished jobs....
  CC [M]  drivers/video/backlight/lm3630a_bl.o
drivers/video/backlight/lm3630a_bl.c: In function 'lm3630a_probe':
drivers/video/backlight/lm3630a_bl.c:428:13: error: 'struct pwm_device' has no member named 'period'
  pchip->pwmd->period = pdata->pwm_period;
             ^~
scripts/Makefile.build:283: recipe for target 'drivers/video/backlight/lm3630a_bl.o' failed
make[4]: *** [drivers/video/backlight/lm3630a_bl.o] Error 1
make[4]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/video/backlight' failed
make[3]: *** [drivers/video/backlight] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[2]: *** [drivers/video] Error 2
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-10:
Disabling USB_DWC3_PCI:
```
  CC [M]  drivers/video/rockchip/rk_fb.o
drivers/video/rockchip/rk_fb.c: In function 'rk_fb_video_mode_from_timing':
drivers/video/rockchip/rk_fb.c:504:19: error: 'const struct display_timing' has no member named 'screen_type'
  screen->type = dt->screen_type;
                   ^~
drivers/video/rockchip/rk_fb.c:505:27: error: 'const struct display_timing' has no member named 'refresh_mode'
  screen->refresh_mode = dt->refresh_mode;
                           ^~
drivers/video/rockchip/rk_fb.c:506:26: error: 'const struct display_timing' has no member named 'lvds_format'
  screen->lvds_format = dt->lvds_format;
                          ^~
drivers/video/rockchip/rk_fb.c:507:19: error: 'const struct display_timing' has no member named 'face'
  screen->face = dt->face;
                   ^~
drivers/video/rockchip/rk_fb.c:508:25: error: 'const struct display_timing' has no member named 'color_mode'
  screen->color_mode = dt->color_mode;
                         ^~
drivers/video/rockchip/rk_fb.c:509:20: error: 'const struct display_timing' has no member named 'screen_widt'
  screen->width = dt->screen_widt;
                    ^~
drivers/video/rockchip/rk_fb.c:510:21: error: 'const struct display_timing' has no member named 'screen_hight'
  screen->height = dt->screen_hight;
                     ^~
drivers/video/rockchip/rk_fb.c:511:22: error: 'const struct display_timing' has no member named 'dsp_lut'
  screen->dsp_lut = dt->dsp_lut;
                      ^~
drivers/video/rockchip/rk_fb.c:512:23: error: 'const struct display_timing' has no member named 'cabc_lut'
  screen->cabc_lut = dt->cabc_lut;
                       ^~
drivers/video/rockchip/rk_fb.c:513:30: error: 'const struct display_timing' has no member named 'cabc_gamma_base'
  screen->cabc_gamma_base = dt->cabc_gamma_base;
                              ^~
drivers/video/rockchip/rk_fb.c: At top level:
drivers/video/rockchip/rk_fb.c:1569:16: warning: 'enum rk_iommu_inttype' declared inside parameter list will not be visible outside of this definition or declaration
error, forbidden warning:rk_fb.c:1569
           enum rk_iommu_inttype itype,
                ^~~~~~~~~~~~~~~~
drivers/video/rockchip/rk_fb.c:1569:33: error: parameter 2 ('itype') has incomplete type
           enum rk_iommu_inttype itype,
                                 ^~~~~
drivers/video/rockchip/rk_fb.c:1568:5: error: function declaration isn't a prototype [-Werror=strict-prototypes]
 int rk_fb_sysmmu_fault_handler(struct device *dev,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/video/rockchip/rk_fb.c: In function 'rk_fb_free_wb_buf':
drivers/video/rockchip/rk_fb.c:1615:3: error: implicit declaration of function 'ion_unmap_iommu' [-Werror=implicit-function-declaration]
   ion_unmap_iommu(dev_drv->dev, rk_fb->ion_client,
   ^~~~~~~~~~~~~~~
drivers/video/rockchip/rk_fb.c:1615:38: error: 'struct rk_fb' has no member named 'ion_client'
   ion_unmap_iommu(dev_drv->dev, rk_fb->ion_client,
                                      ^~
drivers/video/rockchip/rk_fb.c:1618:3: error: implicit declaration of function 'ion_free' [-Werror=implicit-function-declaration]
   ion_free(rk_fb->ion_client, wb_data->ion_handle);
   ^~~~~~~~
drivers/video/rockchip/rk_fb.c:1618:17: error: 'struct rk_fb' has no member named 'ion_client'
   ion_free(rk_fb->ion_client, wb_data->ion_handle);
                 ^~
drivers/video/rockchip/rk_fb.c: In function 'rk_fb_free_dma_buf':
drivers/video/rockchip/rk_fb.c:1634:40: error: 'struct rk_fb' has no member named 'ion_client'
     ion_unmap_iommu(dev_drv->dev, rk_fb->ion_client,
                                        ^~
drivers/video/rockchip/rk_fb.c:1639:18: error: 'struct rk_fb' has no member named 'ion_client'
    ion_free(rk_fb->ion_client, area_data->ion_handle);
                  ^~
drivers/video/rockchip/rk_fb.c: In function 'rk_fb_update_win':
drivers/video/rockchip/rk_fb.c:1698:17: error: 'struct rk_lcdc_win_area' has no member named 'ion_hdl'
     win->area[i].ion_hdl =
                 ^
drivers/video/rockchip/rk_fb.c: In function 'rk_fb_set_wb_buffer':
[... 246 lines cut ...]
make[4]: *** [drivers/usb/dwc_otg_310/dwc_otg_driver.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/usb/dwc_otg_310' failed
make[3]: *** [drivers/usb/dwc_otg_310] Error 2
make[3]: *** Waiting for unfinished jobs....
  CC [M]  drivers/video/backlight/lp8788_bl.o
  CC [M]  drivers/video/backlight/lv5207lp.o
drivers/video/backlight/lm3630a_bl.c: In function 'lm3630a_probe':
drivers/video/backlight/lm3630a_bl.c:428:13: error: 'struct pwm_device' has no member named 'period'
  pchip->pwmd->period = pdata->pwm_period;
             ^~
scripts/Makefile.build:283: recipe for target 'drivers/video/backlight/lm3630a_bl.o' failed
make[4]: *** [drivers/video/backlight/lm3630a_bl.o] Error 1
make[4]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/video/backlight' failed
make[3]: *** [drivers/video/backlight] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/usb' failed
make[2]: *** [drivers/usb] Error 2
make[2]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[2]: *** [drivers/video] Error 2
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-11:
Disabling something (no idea what it was, make menuconfig crashed), inside the Bifrost section fixed the first few errors.
Disabling USB20_HOST , USB20_OTG:
```
drivers/video/backlight/lm3630a_bl.c: In function 'lm3630a_probe':
drivers/video/backlight/lm3630a_bl.c:428:13: error: 'struct pwm_device' has no member named 'period'
  pchip->pwmd->period = pdata->pwm_period;
             ^~
scripts/Makefile.build:283: recipe for target 'drivers/video/backlight/lm3630a_bl.o' failed
make[4]: *** [drivers/video/backlight/lm3630a_bl.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/backlight' failed
make[3]: *** [drivers/video/backlight] Error 2
make[3]: *** Waiting for unfinished jobs....
  CC [M]  drivers/usb/dwc_otg_310/dwc_otg_driver.o
  CC [M]  drivers/usb/dwc_otg_310/dwc_otg_attr.o
  CC [M]  drivers/usb/dwc_otg_310/dwc_otg_cil.o
  CC [M]  drivers/usb/dwc_otg_310/dwc_otg_cil_intr.o
drivers/usb/dwc_otg_310/dwc_otg_driver.c: In function 'dwc_otg_driver_init':
drivers/usb/dwc_otg_310/dwc_otg_driver.c:1690:6: warning: unused variable 'error' [-Wunused-variable]
error, forbidden warning:dwc_otg_driver.c:1690
  int error;
      ^~~~~
At top level:
drivers/usb/dwc_otg_310/dwc_otg_driver.c:1181:13: warning: 'dwc_otg_driver_shutdown' defined but not used [-Wunused-function]
error, forbidden warning:dwc_otg_driver.c:1181
 static void dwc_otg_driver_shutdown(struct platform_device *_dev)
             ^~~~~~~~~~~~~~~~~~~~~~~
drivers/usb/dwc_otg_310/dwc_otg_driver.c:1176:12: warning: 'dwc_otg_driver_resume' defined but not used [-Wunused-function]
error, forbidden warning:dwc_otg_driver.c:1176
 static int dwc_otg_driver_resume(struct platform_device *_dev)
            ^~~~~~~~~~~~~~~~~~~~~
drivers/usb/dwc_otg_310/dwc_otg_driver.c:1170:12: warning: 'dwc_otg_driver_suspend' defined but not used [-Wunused-function]
error, forbidden warning:dwc_otg_driver.c:1170
 static int dwc_otg_driver_suspend(struct platform_device *_dev,
            ^~~~~~~~~~~~~~~~~~~~~~
drivers/usb/dwc_otg_310/dwc_otg_driver.c:844:20: warning: 'dwc_otg_common_irq' defined but not used [-Wunused-function]
error, forbidden warning:dwc_otg_driver.c:844
 static irqreturn_t dwc_otg_common_irq(int irq, void *dev)
                    ^~~~~~~~~~~~~~~~~~
drivers/usb/dwc_otg_310/dwc_otg_driver.c:592:12: warning: 'set_parameters' defined but not used [-Wunused-function]
error, forbidden warning:dwc_otg_driver.c:592
 static int set_parameters(dwc_otg_core_if_t *core_if,
            ^~~~~~~~~~~~~~
In file included from drivers/usb/dwc_otg_310/dwc_otg_os_dep.h:16:0,
                 from drivers/usb/dwc_otg_310/dwc_otg_driver.c:51:
include/linux/device.h:304:26: warning: 'driver_attr_vbus_status' defined but not used [-Wunused-variable]
error, forbidden warning:device.h:304
  struct driver_attribute driver_attr_##_name = __ATTR(_name, _mode, _show, _store)
                          ^
drivers/usb/dwc_otg_310/dwc_otg_driver.c:586:8: note: in expansion of macro 'DRIVER_ATTR'
 static DRIVER_ATTR(vbus_status, S_IRUGO, vbus_status_show, NULL);
        ^~~~~~~~~~~
include/linux/device.h:304:26: warning: 'driver_attr_op_state' defined but not used [-Wunused-variable]
error, forbidden warning:device.h:304
  struct driver_attribute driver_attr_##_name = __ATTR(_name, _mode, _show, _store)
                          ^
drivers/usb/dwc_otg_310/dwc_otg_driver.c:574:8: note: in expansion of macro 'DRIVER_ATTR'
 static DRIVER_ATTR(op_state, S_IRUGO, dwc_otg_op_state_show, NULL);
        ^~~~~~~~~~~
include/linux/device.h:304:26: warning: 'driver_attr_dwc_otg_conn_en' defined but not used [-Wunused-variable]
error, forbidden warning:device.h:304
  struct driver_attribute driver_attr_##_name = __ATTR(_name, _mode, _show, _store)
                          ^
drivers/usb/dwc_otg_310/dwc_otg_driver.c:538:8: note: in expansion of macro 'DRIVER_ATTR'
 static DRIVER_ATTR(dwc_otg_conn_en, S_IRUGO | S_IWUSR, dwc_otg_conn_en_show,
        ^~~~~~~~~~~
include/linux/device.h:304:26: warning: 'driver_attr_force_usb_mode' defined but not used [-Wunused-variable]
error, forbidden warning:device.h:304
  struct driver_attribute driver_attr_##_name = __ATTR(_name, _mode, _show, _store)
                          ^
drivers/usb/dwc_otg_310/dwc_otg_driver.c:509:8: note: in expansion of macro 'DRIVER_ATTR'
[... 6 lines cut ...]
drivers/usb/dwc_otg_310/dwc_otg_driver.c:345:8: note: in expansion of macro 'DRIVER_ATTR'
 static DRIVER_ATTR(debuglevel, S_IRUGO | S_IWUSR, dbg_level_show,
        ^~~~~~~~~~~
include/linux/device.h:304:26: warning: 'driver_attr_version' defined but not used [-Wunused-variable]
error, forbidden warning:device.h:304
  struct driver_attribute driver_attr_##_name = __ATTR(_name, _mode, _show, _store)
                          ^
drivers/usb/dwc_otg_310/dwc_otg_driver.c:320:8: note: in expansion of macro 'DRIVER_ATTR'
 static DRIVER_ATTR(version, S_IRUGO, version_show, NULL);
        ^~~~~~~~~~~
scripts/Makefile.build:277: recipe for target 'drivers/usb/dwc_otg_310/dwc_otg_driver.o' failed
make[4]: *** [drivers/usb/dwc_otg_310/dwc_otg_driver.o] Error 1
make[4]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[2]: *** [drivers/video] Error 2
make[2]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/usb/dwc_otg_310' failed
make[3]: *** [drivers/usb/dwc_otg_310] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/usb' failed
make[2]: *** [drivers/usb] Error 2
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-11:
Disabling MALI_BIFROST_DEVFREQ MALI_BIFROST_DMA_FENCE MALI_BIFROST_DEBUG MALI_2MB_ALLOC MALI_PWRSOFT_765 MALI_BIFROST:
```
drivers/video/backlight/lm3630a_bl.c: In function 'lm3630a_probe':
drivers/video/backlight/lm3630a_bl.c:428:13: error: 'struct pwm_device' has no member named 'period'
  pchip->pwmd->period = pdata->pwm_period;
             ^~
scripts/Makefile.build:283: recipe for target 'drivers/video/backlight/lm3630a_bl.o' failed
make[4]: *** [drivers/video/backlight/lm3630a_bl.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/backlight' failed
make[3]: *** [drivers/video/backlight] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[2]: *** [drivers/video] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```
I suspect that the problems with the Mali drivers will be back because ion is not mentioned above.

ballsystemlord, 2019-04-11:
Disabling BACKLIGHT_LM3630A:
```
  CC [M]  drivers/video/rockchip/rk_fb.o
drivers/video/rockchip/rk_fb.c: In function 'rk_fb_video_mode_from_timing':
drivers/video/rockchip/rk_fb.c:504:19: error: 'const struct display_timing' has no member named 'screen_type'
  screen->type = dt->screen_type;
                   ^~
drivers/video/rockchip/rk_fb.c:505:27: error: 'const struct display_timing' has no member named 'refresh_mode'
  screen->refresh_mode = dt->refresh_mode;
                           ^~
drivers/video/rockchip/rk_fb.c:506:26: error: 'const struct display_timing' has no member named 'lvds_format'
  screen->lvds_format = dt->lvds_format;
                          ^~
drivers/video/rockchip/rk_fb.c:507:19: error: 'const struct display_timing' has no member named 'face'
  screen->face = dt->face;
                   ^~
drivers/video/rockchip/rk_fb.c:508:25: error: 'const struct display_timing' has no member named 'color_mode'
  screen->color_mode = dt->color_mode;
                         ^~
drivers/video/rockchip/rk_fb.c:509:20: error: 'const struct display_timing' has no member named 'screen_widt'
  screen->width = dt->screen_widt;
                    ^~
drivers/video/rockchip/rk_fb.c:510:21: error: 'const struct display_timing' has no member named 'screen_hight'
  screen->height = dt->screen_hight;
                     ^~
drivers/video/rockchip/rk_fb.c:511:22: error: 'const struct display_timing' has no member named 'dsp_lut'
  screen->dsp_lut = dt->dsp_lut;
                      ^~
drivers/video/rockchip/rk_fb.c:512:23: error: 'const struct display_timing' has no member named 'cabc_lut'
  screen->cabc_lut = dt->cabc_lut;
                       ^~
drivers/video/rockchip/rk_fb.c:513:30: error: 'const struct display_timing' has no member named 'cabc_gamma_base'
  screen->cabc_gamma_base = dt->cabc_gamma_base;
                              ^~
drivers/video/rockchip/rk_fb.c: At top level:
drivers/video/rockchip/rk_fb.c:1569:16: warning: 'enum rk_iommu_inttype' declared inside parameter list will not be visible outside of this definition or declaration
error, forbidden warning:rk_fb.c:1569
           enum rk_iommu_inttype itype,
                ^~~~~~~~~~~~~~~~
drivers/video/rockchip/rk_fb.c:1569:33: error: parameter 2 ('itype') has incomplete type
           enum rk_iommu_inttype itype,
                                 ^~~~~
drivers/video/rockchip/rk_fb.c:1568:5: error: function declaration isn't a prototype [-Werror=strict-prototypes]
 int rk_fb_sysmmu_fault_handler(struct device *dev,
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/video/rockchip/rk_fb.c: In function 'rk_fb_free_wb_buf':
drivers/video/rockchip/rk_fb.c:1615:3: error: implicit declaration of function 'ion_unmap_iommu' [-Werror=implicit-function-declaration]
   ion_unmap_iommu(dev_drv->dev, rk_fb->ion_client,
   ^~~~~~~~~~~~~~~
drivers/video/rockchip/rk_fb.c:1615:38: error: 'struct rk_fb' has no member named 'ion_client'
   ion_unmap_iommu(dev_drv->dev, rk_fb->ion_client,
                                      ^~
drivers/video/rockchip/rk_fb.c:1618:3: error: implicit declaration of function 'ion_free' [-Werror=implicit-function-declaration]
   ion_free(rk_fb->ion_client, wb_data->ion_handle);
   ^~~~~~~~
drivers/video/rockchip/rk_fb.c:1618:17: error: 'struct rk_fb' has no member named 'ion_client'
   ion_free(rk_fb->ion_client, wb_data->ion_handle);
                 ^~
drivers/video/rockchip/rk_fb.c: In function 'rk_fb_free_dma_buf':
drivers/video/rockchip/rk_fb.c:1634:40: error: 'struct rk_fb' has no member named 'ion_client'
     ion_unmap_iommu(dev_drv->dev, rk_fb->ion_client,
                                        ^~
drivers/video/rockchip/rk_fb.c:1639:18: error: 'struct rk_fb' has no member named 'ion_client'
    ion_free(rk_fb->ion_client, area_data->ion_handle);
                  ^~
drivers/video/rockchip/rk_fb.c: In function 'rk_fb_update_win':
drivers/video/rockchip/rk_fb.c:1698:17: error: 'struct rk_lcdc_win_area' has no member named 'ion_hdl'
     win->area[i].ion_hdl =
                 ^
drivers/video/rockchip/rk_fb.c: In function 'rk_fb_set_wb_buffer':
[... 158 lines cut ...]
drivers/video/rockchip/rk_fb.c:4231:1: note: in expansion of macro 'late_initcall'
 late_initcall(rockchip_uboot_mem_late_init);
 ^~~~~~~~~~~~~
drivers/video/rockchip/rk_fb.c:3947:12: warning: 'rk_fb_alloc_buffer' defined but not used [-Wunused-function]
error, forbidden warning:rk_fb.c:3947
 static int rk_fb_alloc_buffer(struct fb_info *fbi)
            ^~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:283: recipe for target 'drivers/video/rockchip/rk_fb.o' failed
make[4]: *** [drivers/video/rockchip/rk_fb.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[3]: *** [drivers/video/rockchip] Error 2
make[3]: *** Waiting for unfinished jobs....
  CC [M]  drivers/video/backlight/sky81452-backlight.o
  CC [M]  drivers/video/backlight/tps65217_bl.o
  CC [M]  drivers/video/backlight/wm831x_bl.o
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[2]: *** [drivers/video] Error 2
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

Yup, there the rk_fb problem is again, rearing it's ulgy head...

ballsystemlord, 2019-04-11:
Disabling FB_MIRRORING and FB_ROCKCHIP appears to have fixed it, but don't hold your breath:
```
drivers/video/rockchip/transmitter/rk32_mipi_dsi.c:19:20: fatal error: common.h: No such file or directory
 #include <common.h>
                    ^
compilation terminated.
scripts/Makefile.build:283: recipe for target 'drivers/video/rockchip/transmitter/rk32_mipi_dsi.o' failed
make[5]: *** [drivers/video/rockchip/transmitter/rk32_mipi_dsi.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/transmitter' failed
make[4]: *** [drivers/video/rockchip/transmitter] Error 2
make[4]: *** Waiting for unfinished jobs....
  CC [M]  drivers/video/rockchip/rga/rga_mmu_info.o
  CC [M]  drivers/video/rockchip/rga/rga_reg_info.o
  CC [M]  drivers/video/rockchip/rga2/rga2_reg_info.o
  CC [M]  drivers/video/rockchip/rga2/RGA2_API.o
  CC [M]  drivers/video/rockchip/rga/RGA_API.o
  LD [M]  drivers/video/rockchip/rga2/rga2.o
  LD [M]  drivers/video/rockchip/rga/rga.o
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[3]: *** [drivers/video/rockchip] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[2]: *** [drivers/video] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-11:
Disabling RK32_MIPI_DSI:
```
drivers/video/rockchip/transmitter/tc358768.c:18:23: fatal error: mach/gpio.h: No such file or directory
 #include <mach/gpio.h>
                       ^
compilation terminated.
scripts/Makefile.build:283: recipe for target 'drivers/video/rockchip/transmitter/tc358768.o' failed
make[5]: *** [drivers/video/rockchip/transmitter/tc358768.o] Error 1
make[5]: *** Waiting for unfinished jobs....
drivers/video/rockchip/transmitter/ssd2828.c:18:23: fatal error: mach/gpio.h: No such file or directory
 #include <mach/gpio.h>
                       ^
compilation terminated.
scripts/Makefile.build:283: recipe for target 'drivers/video/rockchip/transmitter/ssd2828.o' failed
make[5]: *** [drivers/video/rockchip/transmitter/ssd2828.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/transmitter' failed
make[4]: *** [drivers/video/rockchip/transmitter] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[3]: *** [drivers/video/rockchip] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[2]: *** [drivers/video] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-11:
Disabling TC358768_RGB2MIPI (Don't I need this for the cameras?):
```
drivers/video/rockchip/transmitter/ssd2828.c:18:23: fatal error: mach/gpio.h: No such file or directory
 #include <mach/gpio.h>
                       ^
compilation terminated.
scripts/Makefile.build:283: recipe for target 'drivers/video/rockchip/transmitter/ssd2828.o' failed
make[5]: *** [drivers/video/rockchip/transmitter/ssd2828.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip/transmitter' failed
make[4]: *** [drivers/video/rockchip/transmitter] Error 2
make[4]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/video/rockchip' failed
make[3]: *** [drivers/video/rockchip] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[2]: *** [drivers/video] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-11:
Disabling SSD2828_RGB2MIPI (Don't I need this for the cameras?):
```
WARNING: modpost: missing MODULE_LICENSE() in drivers/media/video/generic_sensor.o
see include/linux/module.h for more information
FATAL: drivers/rtc/rtc-rk-timer: sizeof(struct platform_device_id)=32 is not a modulo of the size of section __mod_platform__<identifier>_device_table=400.
Fix definition of struct platform_device_id in mod_devicetable.h
scripts/Makefile.modpost:91: recipe for target '__modpost' failed
make[2]: *** [__modpost] Error 1
Makefile:1188: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-17:
Switching RTC_DRV_RK_TIMER from module to built-in leaves still more problems to solve:
```
  Building modules, stage 2.
  MODPOST 2689 modules
WARNING: modpost: missing MODULE_LICENSE() in drivers/media/video/generic_sensor.o
see include/linux/module.h for more information
WARNING: modpost: missing MODULE_LICENSE() in drivers/soc/rockchip/rk_camera.o
see include/linux/module.h for more information
ERROR: "snd_card_locked" [sound/soc/rockchip/snd-soc-rockchip-multicodecs.ko] undefined!
ERROR: "__dma_flush_range" [drivers/video/rockchip/rga2/rga2.ko] undefined!
ERROR: "__dma_flush_range" [drivers/video/rockchip/rga/rga.ko] undefined!
ERROR: "iep_iommu_drm_set_ops" [drivers/video/rockchip/iep/iep_iommu_ops.ko] undefined!
ERROR: "rockchip_ion_client_create" [drivers/video/rockchip/iep/iep_iommu_ion.ko] undefined!
ERROR: "ion_import_dma_buf" [drivers/video/rockchip/iep/iep_iommu_ion.ko] undefined!
ERROR: "ion_phys" [drivers/video/rockchip/iep/iep_iommu_ion.ko] undefined!
ERROR: "ion_map_iommu" [drivers/video/rockchip/iep/iep_iommu_ion.ko] undefined!
ERROR: "ion_free" [drivers/video/rockchip/iep/iep_iommu_ion.ko] undefined!
ERROR: "iep_iommu_clear" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_iommu_info_create" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_iommu_detach" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_config_frame_end_int_clr" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_probe_int" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "devm_ioremap_release" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_iommu_info_destroy" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_config" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_config_frm_start" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_config_done" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_config_frame_end_int_en" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_set_deinterlace_mode" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_get_deinterlace_mode" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_get_status" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_iommu_free" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_iommu_unmap_iommu" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_iommu_attach" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_service" [drivers/video/rockchip/iep/hw_iep_reg.ko] undefined!
ERROR: "iep_iommu_free" [drivers/video/rockchip/iep/hw_iep_reg.ko] undefined!
ERROR: "iep_iommu_map_iommu" [drivers/video/rockchip/iep/hw_iep_reg.ko] undefined!
ERROR: "iep_iommu_import" [drivers/video/rockchip/iep/hw_iep_reg.ko] undefined!
ERROR: "xhci_set_link_state" [drivers/usb/dwc3/dwc3-rockchip.ko] undefined!
ERROR: "dwc3_trace" [drivers/usb/dwc3/dwc3-rockchip.ko] undefined!
ERROR: "dwc3_set_mode" [drivers/usb/dwc3/dwc3-rockchip.ko] undefined!
ERROR: "__tracepoint_dwc3_readl" [drivers/usb/dwc3/dwc3-rockchip.ko] undefined!
ERROR: "xhci_set_link_state" [drivers/usb/dwc3/dwc3-rockchip-inno.ko] undefined!
ERROR: "pvr_sync_init" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "rockchip_ion_client_create" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_get_update_fd" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_append_fences" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_deinit" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_free_append_fences_data" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_get_updates" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_rollback_append_fences" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "ion_client_destroy" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_get_checks" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "saved_command_line" [drivers/nvme/host/nvme.ko] undefined!
ERROR: "generic_sensor_s_power" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_ioctl" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_s_ext_controls" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_g_ext_controls" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_s_control" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_g_control" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_init" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_s_fmt" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_try_fmt" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_g_fmt" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_enum_fmt" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_enum_frameintervals" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_enum_framesizes" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_cropcap" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_s_stream" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_query_bus_param" [drivers/media/video/tp2825.ko] undefined!
[... 815 lines cut ...]
ERROR: "adv_camera_module_g_frame_interval" [drivers/media/i2c/soc_camera/rockchip/adv7181_v4l2-i2c-subdev.ko] undefined!
[... cut ...]
ERROR: "kbase_gpu_irq_handler" [drivers/gpu/arm/midgard/tests/mali_kutf_irq_test/mali_kutf_irq_test.ko] undefined!
ERROR: "kbase_release_device" [drivers/gpu/arm/midgard/tests/mali_kutf_irq_test/mali_kutf_irq_test.ko] undefined!
ERROR: "sip_smc_soc_bus_div" [drivers/devfreq/rockchip_bus.ko] undefined!
scripts/Makefile.modpost:91: recipe for target '__modpost' failed
make[2]: *** [__modpost] Error 1
Makefile:1188: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-17:
In reference to https://github.com/rockchip-linux/kernel/issues/150#issuecomment-481944547
I had hoped to fix this problem by studying this https://marc.info/?l=linux-kbuild&m=136767800809256&w=2 patch for the same problem in a different part of the code base without success. But maybe it will help you.

ballsystemlord, 2019-04-17:
Setting SND_SOC_ROCKCHIP and SND_SOC_ROCKCHIP_MULTICODECS to built-in helps, but the RGA options are trouble some. They need to be copiled as modules or else:
```drivers/media/built-in.o: In function `rga_start':
drivers/media/platform/rockchip/rga/rga-hw.c:653: multiple definition of `rga_start'
drivers/video/built-in.o:(.bss+0x2740): first defined here
./../prebuilts/gcc/linux-x86/aarch64/gcc-linaro-6.3.1-2017.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-ld: Warning: size of symbol `rga_start' changed from 8 in drivers/video/built-in.o to 156 in drivers/media/built-in.o
./../prebuilts/gcc/linux-x86/aarch64/gcc-linaro-6.3.1-2017.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-ld: Warning: type of symbol `rga_start' changed from 1 to 2 in drivers/media/built-in.o
scripts/Makefile.build:418: recipe for target 'drivers/built-in.o' failed
make[1]: *** [drivers/built-in.o] Error 1
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2

```
Disabling ROCKCHIP_RGA2 generates the same error (with ROCKCHIP_RGA built-in), but if I disable ROCKCHIP_RGA and set ROCKCHIP_RGA2 to built-in then I once again am getting closer...
```
  Building modules, stage 2.
  MODPOST 2686 modules
WARNING: modpost: missing MODULE_LICENSE() in drivers/media/video/generic_sensor.o
see include/linux/module.h for more information
WARNING: modpost: missing MODULE_LICENSE() in drivers/soc/rockchip/rk_camera.o
see include/linux/module.h for more information
ERROR: "iep_iommu_drm_set_ops" [drivers/video/rockchip/iep/iep_iommu_ops.ko] undefined!
ERROR: "rockchip_ion_client_create" [drivers/video/rockchip/iep/iep_iommu_ion.ko] undefined!
ERROR: "ion_import_dma_buf" [drivers/video/rockchip/iep/iep_iommu_ion.ko] undefined!
ERROR: "ion_phys" [drivers/video/rockchip/iep/iep_iommu_ion.ko] undefined!
ERROR: "ion_map_iommu" [drivers/video/rockchip/iep/iep_iommu_ion.ko] undefined!
ERROR: "ion_free" [drivers/video/rockchip/iep/iep_iommu_ion.ko] undefined!
ERROR: "iep_iommu_clear" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_iommu_info_create" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_iommu_detach" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_config_frame_end_int_clr" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_probe_int" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "devm_ioremap_release" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_iommu_info_destroy" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_config" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_config_frm_start" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_config_done" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_config_frame_end_int_en" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_set_deinterlace_mode" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_get_deinterlace_mode" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_get_status" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_iommu_free" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_iommu_unmap_iommu" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_iommu_attach" [drivers/video/rockchip/iep/iep_drv.ko] undefined!
ERROR: "iep_service" [drivers/video/rockchip/iep/hw_iep_reg.ko] undefined!
ERROR: "iep_iommu_free" [drivers/video/rockchip/iep/hw_iep_reg.ko] undefined!
ERROR: "iep_iommu_map_iommu" [drivers/video/rockchip/iep/hw_iep_reg.ko] undefined!
ERROR: "iep_iommu_import" [drivers/video/rockchip/iep/hw_iep_reg.ko] undefined!
ERROR: "xhci_set_link_state" [drivers/usb/dwc3/dwc3-rockchip.ko] undefined!
ERROR: "dwc3_trace" [drivers/usb/dwc3/dwc3-rockchip.ko] undefined!
ERROR: "dwc3_set_mode" [drivers/usb/dwc3/dwc3-rockchip.ko] undefined!
ERROR: "__tracepoint_dwc3_readl" [drivers/usb/dwc3/dwc3-rockchip.ko] undefined!
ERROR: "xhci_set_link_state" [drivers/usb/dwc3/dwc3-rockchip-inno.ko] undefined!
ERROR: "pvr_sync_init" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "rockchip_ion_client_create" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_get_update_fd" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_append_fences" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_deinit" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_free_append_fences_data" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_get_updates" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_rollback_append_fences" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "ion_client_destroy" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "pvr_sync_get_checks" [drivers/staging/imgtec/rogue/pvrsrvkm.ko] undefined!
ERROR: "saved_command_line" [drivers/nvme/host/nvme.ko] undefined!
ERROR: "generic_sensor_s_power" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_ioctl" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_s_ext_controls" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_g_ext_controls" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_s_control" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_g_control" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_init" [drivers/media/video/tp2825.ko] undefined!
[... cut ...]
ERROR: "kbase_gpu_irq_handler" [drivers/gpu/arm/midgard/tests/mali_kutf_irq_test/mali_kutf_irq_test.ko] undefined!
ERROR: "kbase_release_device" [drivers/gpu/arm/midgard/tests/mali_kutf_irq_test/mali_kutf_irq_test.ko] undefined!
ERROR: "sip_smc_soc_bus_div" [drivers/devfreq/rockchip_bus.ko] undefined!
scripts/Makefile.modpost:91: recipe for target '__modpost' failed
make[2]: *** [__modpost] Error 1
Makefile:1188: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-17:
Setting IEP to built-in seems to indicate that it needs ION which is an Android feature, so I'll just disable IEP.
```
drivers/built-in.o: In function `iep_ion_unmap_iommu':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/video/rockchip/iep/iep_iommu_ion.c:153: undefined reference to `ion_free'
drivers/built-in.o: In function `iep_ion_map_iommu':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/video/rockchip/iep/iep_iommu_ion.c:182: undefined reference to `ion_map_iommu'
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/video/rockchip/iep/iep_iommu_ion.c:185: undefined reference to `ion_phys'
drivers/built-in.o: In function `iep_ion_import':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/video/rockchip/iep/iep_iommu_ion.c:202: undefined reference to `ion_import_dma_buf'
drivers/built-in.o: In function `iep_ion_create':
/mnt/md1/to-go-through/Downloads/ff-linux/linux-sdk/kernel/drivers/video/rockchip/iep/iep_iommu_ion.c:226: undefined reference to `rockchip_ion_client_create'
drivers/built-in.o:(.debug_addr+0x3d678): undefined reference to `ion_free'
drivers/built-in.o:(.debug_addr+0x3d688): undefined reference to `rockchip_ion_client_create'
drivers/built-in.o:(.debug_addr+0x3d770): undefined reference to `ion_map_iommu'
drivers/built-in.o:(.debug_addr+0x3d7e8): undefined reference to `ion_phys'
drivers/built-in.o:(.debug_addr+0x3d960): undefined reference to `ion_import_dma_buf'
Makefile:1009: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 1
```

ballsystemlord, 2019-04-17:
Setting POWERVR_ROGUE_N to built-in leads to errors:
```
drivers/built-in.o: In function `PVRSRVCommonDeviceInit':
drivers/staging/imgtec/rogue/module_common.c:298: undefined reference to `pvr_sync_init'
drivers/built-in.o: In function `PVRSRVCommonDeviceDeinit':
drivers/staging/imgtec/rogue/module_common.c:404: undefined reference to `pvr_sync_deinit'
drivers/built-in.o: In function `OSConnectionPrivateDataInit':
drivers/staging/imgtec/rogue/osconnection_server.c:110: undefined reference to `rockchip_ion_client_create'
drivers/built-in.o: In function `OSConnectionPrivateDataDeInit':
drivers/staging/imgtec/rogue/env_connection.h:110: undefined reference to `ion_client_destroy'
drivers/built-in.o: In function `PVRSRVRGXKickCDMKM':
drivers/staging/imgtec/rogue/rgxcompute.c:365: undefined reference to `pvr_sync_append_fences'
drivers/staging/imgtec/rogue/rgxcompute.c:380: undefined reference to `pvr_sync_get_updates'
drivers/staging/imgtec/rogue/rgxcompute.c:383: undefined reference to `pvr_sync_get_checks'
drivers/staging/imgtec/rogue/rgxcompute.c:495: undefined reference to `pvr_sync_get_update_fd'
drivers/staging/imgtec/rogue/rgxcompute.c:511: undefined reference to `pvr_sync_free_append_fences_data'
drivers/staging/imgtec/rogue/rgxcompute.c:521: undefined reference to `pvr_sync_rollback_append_fences'
drivers/staging/imgtec/rogue/rgxcompute.c:522: undefined reference to `pvr_sync_free_append_fences_data'
drivers/built-in.o: In function `PVRSRVRGXKickSyncKM':
drivers/staging/imgtec/rogue/rgxkicksync.c:273: undefined reference to `pvr_sync_append_fences'
drivers/staging/imgtec/rogue/rgxkicksync.c:288: undefined reference to `pvr_sync_get_updates'
drivers/staging/imgtec/rogue/rgxkicksync.c:290: undefined reference to `pvr_sync_get_checks'
drivers/staging/imgtec/rogue/rgxkicksync.c:397: undefined reference to `pvr_sync_get_update_fd'
drivers/staging/imgtec/rogue/rgxkicksync.c:414: undefined reference to `pvr_sync_free_append_fences_data'
drivers/staging/imgtec/rogue/rgxkicksync.c:424: undefined reference to `pvr_sync_rollback_append_fences'
drivers/staging/imgtec/rogue/rgxkicksync.c:426: undefined reference to `pvr_sync_free_append_fences_data'
drivers/built-in.o: In function `PVRSRVRGXKickTA3DKM':
drivers/staging/imgtec/rogue/rgxta3d.c:3110: undefined reference to `pvr_sync_append_fences'
drivers/staging/imgtec/rogue/rgxta3d.c:3133: undefined reference to `pvr_sync_get_updates'
drivers/staging/imgtec/rogue/rgxta3d.c:3136: undefined reference to `pvr_sync_get_checks'
drivers/staging/imgtec/rogue/rgxta3d.c:3578: undefined reference to `pvr_sync_get_update_fd'
drivers/staging/imgtec/rogue/rgxta3d.c:3594: undefined reference to `pvr_sync_free_append_fences_data'
drivers/staging/imgtec/rogue/rgxta3d.c:3616: undefined reference to `pvr_sync_rollback_append_fences'
drivers/staging/imgtec/rogue/rgxta3d.c:3617: undefined reference to `pvr_sync_free_append_fences_data'
drivers/built-in.o: In function `PVRSRVRGXTDMSubmitTransferKM':
drivers/staging/imgtec/rogue/rgxtdmtransfer.c:484: undefined reference to `pvr_sync_append_fences'
drivers/staging/imgtec/rogue/rgxtdmtransfer.c:499: undefined reference to `pvr_sync_get_updates'
drivers/staging/imgtec/rogue/rgxtdmtransfer.c:501: undefined reference to `pvr_sync_get_checks'
drivers/staging/imgtec/rogue/rgxtdmtransfer.c:630: undefined reference to `pvr_sync_get_update_fd'
drivers/staging/imgtec/rogue/rgxtdmtransfer.c:649: undefined reference to `pvr_sync_free_append_fences_data'
drivers/staging/imgtec/rogue/rgxtdmtransfer.c:685: undefined reference to `pvr_sync_rollback_append_fences'
drivers/staging/imgtec/rogue/rgxtdmtransfer.c:687: undefined reference to `pvr_sync_free_append_fences_data'
drivers/built-in.o: In function `PVRSRVRGXSubmitTransferKM':
drivers/staging/imgtec/rogue/rgxtransfer.c:761: undefined reference to `pvr_sync_append_fences'
drivers/staging/imgtec/rogue/rgxtransfer.c:776: undefined reference to `pvr_sync_get_updates'
drivers/staging/imgtec/rogue/rgxtransfer.c:778: undefined reference to `pvr_sync_get_checks'
drivers/staging/imgtec/rogue/rgxtransfer.c:978: undefined reference to `pvr_sync_get_update_fd'
drivers/staging/imgtec/rogue/rgxtransfer.c:997: undefined reference to `pvr_sync_free_append_fences_data'
drivers/staging/imgtec/rogue/rgxtransfer.c:1035: undefined reference to `pvr_sync_rollback_append_fences'
drivers/staging/imgtec/rogue/rgxtransfer.c:1037: undefined reference to `pvr_sync_free_append_fences_data'
drivers/built-in.o:(.debug_addr+0x25e3d8): undefined reference to `pvr_sync_init'
drivers/built-in.o:(.debug_addr+0x25e4d0): undefined reference to `pvr_sync_deinit'
drivers/built-in.o:(.debug_addr+0x25e5c8): undefined reference to `rockchip_ion_client_create'
drivers/built-in.o:(.debug_addr+0x25e610): undefined reference to `ion_client_destroy'
drivers/built-in.o:(.debug_addr+0x267be8): undefined reference to `pvr_sync_append_fences'
drivers/built-in.o:(.debug_addr+0x267d60): undefined reference to `pvr_sync_get_checks'
drivers/built-in.o:(.debug_addr+0x267ea0): undefined reference to `pvr_sync_get_updates'
drivers/built-in.o:(.debug_addr+0x268000): undefined reference to `pvr_sync_rollback_append_fences'
drivers/built-in.o:(.debug_addr+0x268220): undefined reference to `pvr_sync_get_update_fd'
drivers/built-in.o:(.debug_addr+0x268250): undefined reference to `pvr_sync_free_append_fences_data'
drivers/built-in.o:(.debug_addr+0x26fec8): undefined reference to `pvr_sync_free_append_fences_data'
[... cut ...]
drivers/built-in.o:(.debug_addr+0x275998): undefined reference to `pvr_sync_get_checks'
drivers/built-in.o:(.debug_addr+0x275b00): undefined reference to `pvr_sync_get_updates'
drivers/built-in.o:(.debug_addr+0x275ef8): undefined reference to `pvr_sync_rollback_append_fences'
drivers/built-in.o:(.debug_addr+0x275ff8): undefined reference to `pvr_sync_get_update_fd'
drivers/built-in.o:(.debug_addr+0x276050): undefined reference to `pvr_sync_free_append_fences_data'
Makefile:1009: recipe for target 'vmlinux' failed
make: *** [vmlinux] Error 1
```
So, I'll have to disable it.

ballsystemlord, 2019-04-17:
Setting USB_DWC3_ROCKCHIP , USB_DWC3_ROCKCHIP_INNO , BLK_DEV_NVME to built-in helps but there's still much more to do:
```
  MODPOST 2677 modules
WARNING: modpost: missing MODULE_LICENSE() in drivers/media/video/generic_sensor.o
see include/linux/module.h for more information
WARNING: modpost: missing MODULE_LICENSE() in drivers/soc/rockchip/rk_camera.o
see include/linux/module.h for more information
ERROR: "generic_sensor_s_power" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_ioctl" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_s_ext_controls" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_g_ext_controls" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_s_control" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_g_control" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_init" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_s_fmt" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_try_fmt" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_g_fmt" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_enum_fmt" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_enum_frameintervals" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_enum_framesizes" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_cropcap" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_s_stream" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_query_bus_param" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_set_bus_param" [drivers/media/video/tp2825.ko] undefined!
ERROR: "deinit_sensor_gpios" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_check_id" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_softreset" [drivers/media/video/tp2825.ko] undefined!
ERROR: "sensor_read_reg1val1" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_get_max_min_res" [drivers/media/video/tp2825.ko] undefined!
ERROR: "sensor_write_reg1val1" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_af_workqueue_set" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_ioctrl" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_write_array" [drivers/media/video/tp2825.ko] undefined!
ERROR: "generic_sensor_s_power" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_ioctl" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_s_ext_controls" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_g_ext_controls" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_s_control" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_g_control" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_init" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_s_fmt" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_try_fmt" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_g_fmt" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_enum_fmt" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_enum_frameintervals" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_enum_framesizes" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_cropcap" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_s_stream" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_query_bus_param" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_set_bus_param" [drivers/media/video/sp2518.ko] undefined!
ERROR: "deinit_sensor_gpios" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_check_id" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_softreset" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_get_max_min_res" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_af_workqueue_set" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_ioctrl" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_write_array" [drivers/media/video/sp2518.ko] undefined!
ERROR: "generic_sensor_s_power" [drivers/media/video/sp0838.ko] undefined!
ERROR: "generic_sensor_ioctl" [drivers/media/video/sp0838.ko] undefined!
ERROR: "generic_sensor_s_ext_controls" [drivers/media/video/sp0838.ko] undefined!
ERROR: "generic_sensor_g_ext_controls" [drivers/media/video/sp0838.ko] undefined!
ERROR: "generic_sensor_s_control" [drivers/media/video/sp0838.ko] undefined!
ERROR: "generic_sensor_g_control" [drivers/media/video/sp0838.ko] undefined!
ERROR: "generic_sensor_init" [drivers/media/video/sp0838.ko] undefined!
ERROR: "generic_sensor_s_fmt" [drivers/media/video/sp0838.ko] undefined!
ERROR: "generic_sensor_try_fmt" [drivers/media/video/sp0838.ko] undefined!
ERROR: "generic_sensor_g_fmt" [drivers/media/video/sp0838.ko] undefined!
ERROR: "generic_sensor_enum_fmt" [drivers/media/video/sp0838.ko] undefined!
ERROR: "generic_sensor_enum_frameintervals" [drivers/media/video/sp0838.ko] undefined!
ERROR: "generic_sensor_enum_framesizes" [drivers/media/video/sp0838.ko] undefined!
[... 768 lines cut ...]
ERROR: "adv_camera_module_g_frame_interval" [drivers/media/i2c/soc_camera/rockchip/adv7181_v4l2-i2c-subdev.ko] undefined!
[... cut ...]
ERROR: "kbase_gpu_irq_handler" [drivers/gpu/arm/midgard/tests/mali_kutf_irq_test/mali_kutf_irq_test.ko] undefined!
ERROR: "kbase_release_device" [drivers/gpu/arm/midgard/tests/mali_kutf_irq_test/mali_kutf_irq_test.ko] undefined!
ERROR: "sip_smc_soc_bus_div" [drivers/devfreq/rockchip_bus.ko] undefined!
scripts/Makefile.modpost:91: recipe for target '__modpost' failed
make[2]: *** [__modpost] Error 1
Makefile:1188: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-17:
I had to fall back to an earlier version of my config before I disabled some input stuff to try to get the kernel to build. As I've listed everything else that's wrong with this thing why not this also?

```
  CC      drivers/input/sensors/accel/dmard10.o
drivers/input/sensors/accel/dmard10.c:169:0: warning: "DMARD10_RANGE" redefined
error, forbidden warning:dmard10.c:169
 #define DMARD10_RANGE   4000000
 
drivers/input/sensors/accel/dmard10.c:110:0: note: this is the location of the previous definition
 #define DMARD10_RANGE      2000000
 
drivers/input/sensors/accel/dmard10.c: In function 'gsensor_reset':
drivers/input/sensors/accel/dmard10.c:184:30: warning: unused variable 'sensor' [-Wunused-variable]
error, forbidden warning:dmard10.c:184
  struct sensor_private_data *sensor =
                              ^~~~~~
  CC      drivers/input/sensors/angle/angle_kxtik.o
scripts/Makefile.build:277: recipe for target 'drivers/input/sensors/accel/dmard10.o' failed
make[4]: *** [drivers/input/sensors/accel/dmard10.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors/accel' failed
make[3]: *** [drivers/input/sensors/accel] Error 2
make[3]: *** Waiting for unfinished jobs....
  CC      drivers/media/cec/cec-core.o
  CC      drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.o
  CC      drivers/media/cec/cec-adap.o
drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.c:582:22: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
error, forbidden warning:mali_kbase_ipa.c:582
  .get_static_power = &kbase_get_static_power,
                      ^
drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.c:582:22: note: (near initialization for 'kbase_ipa_power_model_ops.get_static_power')
drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.c:583:23: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
error, forbidden warning:mali_kbase_ipa.c:583
  .get_dynamic_power = &kbase_get_dynamic_power,
                       ^
drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.c:583:23: note: (near initialization for 'kbase_ipa_power_model_ops.get_dynamic_power')
scripts/Makefile.build:277: recipe for target 'drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.o' failed
make[4]: *** [drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/gpu/arm/midgard' failed
make[3]: *** [drivers/gpu/arm/midgard] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu/arm' failed
make[2]: *** [drivers/gpu/arm] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu' failed
make[1]: *** [drivers/gpu] Error 2
make[1]: *** Waiting for unfinished jobs....
  CC      drivers/input/sensors/angle/angle_lis3dh.o
drivers/input/sensors/angle/angle_kxtik.c: In function 'sensor_convert_data':
drivers/input/sensors/angle/angle_kxtik.c:259:7: warning: this 'else' clause does not guard... [-Wmisleading-indentation]
error, forbidden warning:angle_kxtik.c:259
       else
       ^~~~
drivers/input/sensors/angle/angle_kxtik.c:262:4: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'else'
    break;
    ^~~~~
scripts/Makefile.build:277: recipe for target 'drivers/input/sensors/angle/angle_kxtik.o' failed
make[4]: *** [drivers/input/sensors/angle/angle_kxtik.o] Error 1
make[4]: *** Waiting for unfinished jobs....
  CC      drivers/media/cec/cec-api.o
drivers/media/cec/cec-core.c: In function 'cec_allocate_adapter':
drivers/media/cec/cec-core.c:266:13: error: too many arguments to function 'rc_allocate_device'
  adap->rc = rc_allocate_device(RC_DRIVER_SCANCODE);
             ^~~~~~~~~~~~~~~~~~
In file included from include/media/cec.h:31:0,
                 from drivers/media/cec/cec-priv.h:24,
                 from drivers/media/cec/cec-core.c:30:
include/media/rc-core.h:186:16: note: declared here
 struct rc_dev *rc_allocate_device(void);
                ^~~~~~~~~~~~~~~~~~
drivers/media/cec/cec-core.c:282:31: error: 'BUS_CEC' undeclared (first use in this function)
  adap->rc->input_id.bustype = BUS_CEC;
                               ^~~~~~~
[... 67 lines cut ...]
make[2]: *** [drivers/input/touchscreen] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input' failed
make[1]: *** [drivers/input] Error 2
  CC      drivers/media/rc/ir-nec-decoder.o
  LD      drivers/media/dvb-core/dvb-core.o
  LD      drivers/media/dvb-core/built-in.o
  CC      drivers/media/platform/rockchip/isp1/regs.o
  CC      drivers/media/platform/rockchip/isp1/isp_stats.o
  LD      drivers/media/rc/keymaps/built-in.o
  CC      drivers/media/platform/rockchip/isp1/isp_params.o
  CC      drivers/media/platform/rockchip/isp1/capture.o
  CC      drivers/media/tuners/tda9887.o
  LD      drivers/media/rc/rc-core.o
  CC      drivers/media/platform/rockchip/isp1/mipi_dphy_sy.o
  LD      drivers/media/rc/built-in.o
  LD      drivers/media/tuners/built-in.o
  LD      drivers/media/platform/rockchip/isp1/video_rkisp1.o
  LD      drivers/media/platform/rockchip/isp1/built-in.o
scripts/Makefile.build:484: recipe for target 'drivers/media/platform' failed
make[2]: *** [drivers/media/platform] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/media' failed
make[1]: *** [drivers/media] Error 2
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

ballsystemlord, 2019-04-17:
Disabling GS_DMT10 helps:
```
drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.c:582:22: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
error, forbidden warning:mali_kbase_ipa.c:582
  .get_static_power = &kbase_get_static_power,
                      ^
drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.c:582:22: note: (near initialization for 'kbase_ipa_power_model_ops.get_static_power')
drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.c:583:23: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
error, forbidden warning:mali_kbase_ipa.c:583
  .get_dynamic_power = &kbase_get_dynamic_power,
                       ^
drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.c:583:23: note: (near initialization for 'kbase_ipa_power_model_ops.get_dynamic_power')
scripts/Makefile.build:277: recipe for target 'drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.o' failed
make[4]: *** [drivers/gpu/arm/midgard/ipa/mali_kbase_ipa.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/gpu/arm/midgard' failed
make[3]: *** [drivers/gpu/arm/midgard] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu/arm' failed
make[2]: *** [drivers/gpu/arm] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC      drivers/input/sensors/angle/angle_kxtik.o
drivers/media/cec/cec-core.c: In function 'cec_allocate_adapter':
drivers/media/cec/cec-core.c:266:13: error: too many arguments to function 'rc_allocate_device'
  adap->rc = rc_allocate_device(RC_DRIVER_SCANCODE);
             ^~~~~~~~~~~~~~~~~~
In file included from include/media/cec.h:31:0,
                 from drivers/media/cec/cec-priv.h:24,
                 from drivers/media/cec/cec-core.c:30:
include/media/rc-core.h:186:16: note: declared here
 struct rc_dev *rc_allocate_device(void);
                ^~~~~~~~~~~~~~~~~~
drivers/media/cec/cec-core.c:282:31: error: 'BUS_CEC' undeclared (first use in this function)
  adap->rc->input_id.bustype = BUS_CEC;
                               ^~~~~~~
drivers/media/cec/cec-core.c:282:31: note: each undeclared identifier is reported only once for each function it appears in
drivers/media/cec/cec-core.c:287:32: error: 'RC_BIT_CEC' undeclared (first use in this function)
  adap->rc->allowed_protocols = RC_BIT_CEC;
                                ^~~~~~~~~~
drivers/media/cec/cec-core.c:289:23: error: 'RC_MAP_CEC' undeclared (first use in this function)
  adap->rc->map_name = RC_MAP_CEC;
                       ^~~~~~~~~~
scripts/Makefile.build:277: recipe for target 'drivers/media/cec/cec-core.o' failed
make[3]: *** [drivers/media/cec/cec-core.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/media/cec' failed
make[2]: *** [drivers/media/cec] Error 2
make[2]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/media' failed
make[1]: *** [drivers/media] Error 2
make[1]: *** Waiting for unfinished jobs....
drivers/input/sensors/angle/angle_kxtik.c: In function 'sensor_convert_data':
drivers/input/sensors/angle/angle_kxtik.c:259:7: warning: this 'else' clause does not guard... [-Wmisleading-indentation]
error, forbidden warning:angle_kxtik.c:259
       else
       ^~~~
drivers/input/sensors/angle/angle_kxtik.c:262:4: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'else'
    break;
    ^~~~~
scripts/Makefile.build:277: recipe for target 'drivers/input/sensors/angle/angle_kxtik.o' failed
make[4]: *** [drivers/input/sensors/angle/angle_kxtik.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors/angle' failed
make[3]: *** [drivers/input/sensors/angle] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/gpu' failed
make[1]: *** [drivers/gpu] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors' failed
make[2]: *** [drivers/input/sensors] Error 2
make[2]: *** Waiting for unfinished jobs....
drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.c:571:11: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
error, forbidden warning:rockchip_gslX680.c:571
  .write = gsl_config_write_proc,
           ^~~~~~~~~~~~~~~~~~~~~
drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.c:571:11: note: (near initialization for 'gsl_seq_fops.write')
scripts/Makefile.build:277: recipe for target 'drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.o' failed
make[4]: *** [drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/input/touchscreen/gslx680_d708' failed
make[3]: *** [drivers/input/touchscreen/gslx680_d708] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input/touchscreen' failed
make[2]: *** [drivers/input/touchscreen] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input' failed
make[1]: *** [drivers/input] Error 2
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

ballsystemlord, 2019-04-17:
It seems that I failed to mention when I first started trouble shooting that you also need to disable MEDIA_CEC_RC:
```
drivers/input/sensors/angle/angle_kxtik.c: In function 'sensor_convert_data':
drivers/input/sensors/angle/angle_kxtik.c:259:7: warning: this 'else' clause does not guard... [-Wmisleading-indentation]
error, forbidden warning:angle_kxtik.c:259
       else
       ^~~~
drivers/input/sensors/angle/angle_kxtik.c:262:4: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'else'
    break;
    ^~~~~
scripts/Makefile.build:277: recipe for target 'drivers/input/sensors/angle/angle_kxtik.o' failed
make[4]: *** [drivers/input/sensors/angle/angle_kxtik.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors/angle' failed
make[3]: *** [drivers/input/sensors/angle] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors' failed
make[2]: *** [drivers/input/sensors] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC      drivers/media/cec/cec-adap.o
drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.c:571:11: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
error, forbidden warning:rockchip_gslX680.c:571
  .write = gsl_config_write_proc,
           ^~~~~~~~~~~~~~~~~~~~~
drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.c:571:11: note: (near initialization for 'gsl_seq_fops.write')
scripts/Makefile.build:277: recipe for target 'drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.o' failed
make[4]: *** [drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/input/touchscreen/gslx680_d708' failed
make[3]: *** [drivers/input/touchscreen/gslx680_d708] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input/touchscreen' failed
make[2]: *** [drivers/input/touchscreen] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input' failed
make[1]: *** [drivers/input] Error 2
make[1]: *** Waiting for unfinished jobs....
  LD      drivers/media/cec/cec.o
  LD      drivers/media/cec/built-in.o
  LD      drivers/media/built-in.o
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

ballsystemlord, 2019-04-17:
Disabling ANGLE_KXTIK:
```
drivers/input/sensors/angle/angle_lis3dh.c: In function 'sensor_convert_data':
drivers/input/sensors/angle/angle_lis3dh.c:192:7: warning: this 'else' clause does not guard... [-Wmisleading-indentation]
error, forbidden warning:angle_lis3dh.c:192
       else
       ^~~~
drivers/input/sensors/angle/angle_lis3dh.c:195:4: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the 'else'
    break;
    ^~~~~
scripts/Makefile.build:277: recipe for target 'drivers/input/sensors/angle/angle_lis3dh.o' failed
make[4]: *** [drivers/input/sensors/angle/angle_lis3dh.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors/angle' failed
make[3]: *** [drivers/input/sensors/angle] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors' failed
make[2]: *** [drivers/input/sensors] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC      drivers/phy/rockchip/phy-rockchip-usb.o
  CC      drivers/pinctrl/pinctrl-amd.o
  CC      drivers/power/cw2015_battery.o
drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.c:571:11: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
error, forbidden warning:rockchip_gslX680.c:571
  .write = gsl_config_write_proc,
           ^~~~~~~~~~~~~~~~~~~~~
drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.c:571:11: note: (near initialization for 'gsl_seq_fops.write')
scripts/Makefile.build:277: recipe for target 'drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.o' failed
make[4]: *** [drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/input/touchscreen/gslx680_d708' failed
make[3]: *** [drivers/input/touchscreen/gslx680_d708] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input/touchscreen' failed
make[2]: *** [drivers/input/touchscreen] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input' failed
make[1]: *** [drivers/input] Error 2
make[1]: *** Waiting for unfinished jobs....
```

ballsystemlord, 2019-04-17:
Disabling ANGLE_LIS3DH:
```
drivers/input/sensors/lsensor/ls_ap321xx.c:28:23: fatal error: mach/gpio.h: No such file or directory
 #include <mach/gpio.h>
                       ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/input/sensors/lsensor/ls_ap321xx.o' failed
make[4]: *** [drivers/input/sensors/lsensor/ls_ap321xx.o] Error 1
make[4]: *** Waiting for unfinished jobs....
drivers/input/sensors/lsensor/ls_us5152.c:14:23: fatal error: mach/gpio.h: No such file or directory
 #include <mach/gpio.h>
                       ^
compilation terminated.
  CC      drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.o
scripts/Makefile.build:277: recipe for target 'drivers/input/sensors/lsensor/ls_us5152.o' failed
make[4]: *** [drivers/input/sensors/lsensor/ls_us5152.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors/lsensor' failed
make[3]: *** [drivers/input/sensors/lsensor] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors' failed
make[2]: *** [drivers/input/sensors] Error 2
make[2]: *** Waiting for unfinished jobs....
drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.c:571:11: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
error, forbidden warning:rockchip_gslX680.c:571
  .write = gsl_config_write_proc,
           ^~~~~~~~~~~~~~~~~~~~~
drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.c:571:11: note: (near initialization for 'gsl_seq_fops.write')
scripts/Makefile.build:277: recipe for target 'drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.o' failed
make[4]: *** [drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/input/touchscreen/gslx680_d708' failed
make[3]: *** [drivers/input/touchscreen/gslx680_d708] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input/touchscreen' failed
make[2]: *** [drivers/input/touchscreen] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input' failed
make[1]: *** [drivers/input] Error 2
make[1]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

ballsystemlord, 2019-04-17:
Disabling LS_AP321XX:
```
drivers/input/sensors/lsensor/ls_us5152.c:14:23: fatal error: mach/gpio.h: No such file or directory
 #include <mach/gpio.h>
                       ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/input/sensors/lsensor/ls_us5152.o' failed
make[4]: *** [drivers/input/sensors/lsensor/ls_us5152.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors/lsensor' failed
make[3]: *** [drivers/input/sensors/lsensor] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors' failed
make[2]: *** [drivers/input/sensors] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input' failed
make[1]: *** [drivers/input] Error 2
make[1]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
make: *** Waiting for unfinished jobs....
```

ballsystemlord, 2019-04-17:
Disabling LS_US5152:
```
drivers/input/sensors/lsensor/ls_us5152.c:14:23: fatal error: mach/gpio.h: No such file or directory
 #include <mach/gpio.h>
                       ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/input/sensors/lsensor/ls_us5152.o' failed
make[4]: *** [drivers/input/sensors/lsensor/ls_us5152.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors/lsensor' failed
make[3]: *** [drivers/input/sensors/lsensor] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors' failed
make[2]: *** [drivers/input/sensors] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input' failed
make[1]: *** [drivers/input] Error 2
make[1]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
make: *** Waiting for unfinished jobs....
```

ballsystemlord, 2019-04-17:
Disabling PS_AP321XX:
```
drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.c:571:11: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
error, forbidden warning:rockchip_gslX680.c:571
  .write = gsl_config_write_proc,
           ^~~~~~~~~~~~~~~~~~~~~
drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.c:571:11: note: (near initialization for 'gsl_seq_fops.write')
scripts/Makefile.build:277: recipe for target 'drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.o' failed
make[4]: *** [drivers/input/touchscreen/gslx680_d708/rockchip_gslX680.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/input/touchscreen/gslx680_d708' failed
make[3]: *** [drivers/input/touchscreen/gslx680_d708] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input/touchscreen' failed
make[2]: *** [drivers/input/touchscreen] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input' failed
make[1]: *** [drivers/input] Error 2
make[1]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```

ballsystemlord, 2019-04-17:
Disabling TOUCHSCREEN_GSLX680_D708:
```
In file included from drivers/gpu/arm/mali400/mali/common/mali_session.h:17:0,
                 from drivers/gpu/arm/mali400/mali/common/mali_gp_job.h:17,
                 from drivers/gpu/arm/mali400/mali/common/mali_gp.h:15,
                 from drivers/gpu/arm/mali400/mali/common/mali_group.h:17,
                 from drivers/gpu/arm/mali400/mali/common/mali_pm_metrics.h:16,
                 from drivers/gpu/arm/mali400/mali/include/linux/mali/mali_utgard.h:22,
                 from drivers/gpu/arm/mali400/mali/platform/rk/rk.c:40:
drivers/gpu/arm/mali400/mali/linux/mali_memory_types.h:17:34: fatal error: ump_kernel_interface.h: No such file or directory
 #include "ump_kernel_interface.h"
                                  ^
compilation terminated.
scripts/Makefile.build:277: recipe for target 'drivers/gpu/arm/mali400/mali/platform/rk/rk.o' failed
make[6]: *** [drivers/gpu/arm/mali400/mali/platform/rk/rk.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/gpu/arm/mali400/mali' failed
make[5]: *** [drivers/gpu/arm/mali400/mali] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu/arm/mali400' failed
make[4]: *** [drivers/gpu/arm/mali400] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu/arm' failed
make[3]: *** [drivers/gpu/arm] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu' failed
make[2]: *** [drivers/gpu] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
make[1]: *** Waiting for unfinished jobs....
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-17:
Disabling MALI400_UMP (another thing I must have forgotten to mention):
```
make[5]: *** No rule to make target 'drivers/input/sensors/compass/mmc328x.c', needed by 'drivers/input/sensors/compass/mmc328x.o'.  Stop.
make[5]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors/compass' failed
make[4]: *** [drivers/input/sensors/compass] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/input/sensors' failed
make[3]: *** [drivers/input/sensors] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/input' failed
make[2]: *** [drivers/input] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

ballsystemlord, 2019-04-17:
Disabling COMPASS_MMC328X is still insufficient:
```
drivers/video/backlight/lm3630a_bl.c: In function 'lm3630a_probe':
drivers/video/backlight/lm3630a_bl.c:428:13: error: 'struct pwm_device' has no member named 'period'
  pchip->pwmd->period = pdata->pwm_period;
             ^~
scripts/Makefile.build:283: recipe for target 'drivers/video/backlight/lm3630a_bl.o' failed
make[4]: *** [drivers/video/backlight/lm3630a_bl.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/video/backlight' failed
make[3]: *** [drivers/video/backlight] Error 2
make[3]: *** Waiting for unfinished jobs....
scripts/Makefile.build:484: recipe for target 'drivers/video' failed
make[2]: *** [drivers/video] Error 2
make[2]: *** Waiting for unfinished jobs....
Makefile:1031: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
arch/arm64/Makefile:167: recipe for target 'kernel.img' failed
make: *** [kernel.img] Error 2
```

[30 more replies]