rockchip-linux/kernel #151: Enabling CONFIG_MALI_TRACE_TIMELINE leads to many errors

ballsystemlord, 2019-04-02:
I thought I would enable tracing in the mali driver so that I could help diagnose any problems with the driver. Unfortunately, the kernel will not build with this option enabled.
I'm using the latest commit AFAIK 7532103cd08c810f95fdc71822be14fc17b455fc .

```
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:(__tracepoints+0x0): multiple definition of `__tracepoint_mali_PM_WAKE_WAITERS'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:(__tracepoints+0x0): first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:1978: multiple definition of `__tracepoint_mali_PM_CA_SET_POLICY'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_trace_timeline.c:64: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:1981: multiple definition of `__tracepoint_mali_PM_CURRENT_POLICY_TERM'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_trace_timeline.c:71: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:1986: multiple definition of `__tracepoint_mali_PM_CURRENT_POLICY_INIT'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:27: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:1992: multiple definition of `__tracepoint_mali_PM_SET_POLICY'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:27: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2011: multiple definition of `__tracepoint_mali_PM_GPU_OFF'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/include/linux/perf_event.h:891: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2018: multiple definition of `__tracepoint_mali_PM_GPU_ON'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/include/linux/trace_events.h:626: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2037: multiple definition of `__tracepoint_mali_PM_CORES_CHANGE_AVAILABLE_TILER'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:73: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2053: multiple definition of `__tracepoint_mali_PM_CORES_CHANGE_AVAILABLE'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:73: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2055: multiple definition of `__tracepoint_mali_PM_CORES_AVAILABLE_TILER'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/include/linux/perf_event.h:893: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2070: multiple definition of `__tracepoint_mali_PM_CORES_AVAILABLE'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:109: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2082: multiple definition of `__tracepoint_mali_PM_RELEASE_CHANGE_SHADER_INUSE'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:109: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2019: multiple definition of `__tracepoint_mali_PM_REGISTER_CHANGE_SHADER_INUSE'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/include/linux/perf_event.h:891: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:3964: multiple definition of `__tracepoint_mali_PM_REGISTER_CHANGE_SHADER_NEEDED'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/include/linux/trace_events.h:626: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:3965: multiple definition of `__tracepoint_mali_PM_REQUEST_CHANGE_SHADER_NEEDED'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/include/linux/trace_events.h:450: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:3966: multiple definition of `__tracepoint_mali_PM_UNREQUEST_CHANGE_SHADER_NEEDED'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/include/linux/trace_events.h:458: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:3967: multiple definition of `__tracepoint_mali_PM_DESIRED_REACHED_TILER'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:73: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:4785: multiple definition of `__tracepoint_mali_PM_DESIRED_REACHED'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/include/linux/trace_events.h:454: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2109: multiple definition of `__tracepoint_mali_PM_CORES_POWERED_L2'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:73: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2131: multiple definition of `__tracepoint_mali_PM_CORES_POWERED_TILER'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:109: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2140: multiple definition of `__tracepoint_mali_PM_CORES_POWERED'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:109: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2116: multiple definition of `__tracepoint_mali_PM_PWROFF_L2'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:27: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2162: multiple definition of `__tracepoint_mali_PM_PWROFF_TILER'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/include/linux/trace_seq.h:68: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2184: multiple definition of `__tracepoint_mali_PM_PWROFF'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:73: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2188: multiple definition of `__tracepoint_mali_PM_PWRON_L2'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/include/linux/trace_seq.h:68: first defined here
drivers/gpu/arm/midgard/mali_kbase_core_linux.o:linux-sdk/kernel/drivers/gpu/arm/midgard/mali_kbase_core_linux.c:2195: multiple definition of `__tracepoint_mali_PM_PWRON_TILER'
drivers/gpu/arm/midgard/mali_kbase_trace_timeline.o:linux-sdk/kernel/drivers/gpu/arm/midgard/./mali_linux_kbase_trace.h:109: first defined here
[... cut ...]
scripts/Makefile.build:468: recipe for target 'drivers/gpu/arm/midgard/midgard_kbase.o' failed
make[4]: *** [drivers/gpu/arm/midgard/midgard_kbase.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/gpu/arm/midgard' failed
make[3]: *** [drivers/gpu/arm/midgard] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu/arm' failed
make[2]: *** [drivers/gpu/arm] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu' failed
make[1]: *** [drivers/gpu] Error 2
make[1]: *** Waiting for unfinished jobs....
```

Caesar-github, 2019-04-19:
what are you going to enable this config?  I don't think we need enable it.

Caesar-github, 2019-04-19:
The lastest code is fine with enabling this config, can you test it again with lastest code?

--- a/arch/arm64/configs/rockchip_linux_defconfig
+++ b/arch/arm64/configs/rockchip_linux_defconfig
@@ -338,6 +338,7 @@ CONFIG_MALI_EXPERT=y
 CONFIG_MALI_PLATFORM_THIRDPARTY=y
 CONFIG_MALI_PLATFORM_THIRDPARTY_NAME="rk"
 CONFIG_MALI_DEBUG=y
+CONFIG_MALI_TRACE_TIMELINE=y

ballsystemlord, 2019-04-23:
I wanted to be able to give an intelligent report if I encounter any bugs with the MALI code, so I enabled: CONFIG_MALI_TRACE_TIMELINE . I don't strictly need it in order to use the kernel.
Using the latest commit: b43ed1b265244eb5d1d9cbd2bf780709746a23ef I'm still getting errors:

```
drivers/gpu/arm/midgard_for_linux/backend/gpu/mali_kbase_power_model_simple.o
  LD      drivers/gpu/arm/midgard_for_linux/midgard_kbase.o
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:(__tracepoints+0x0): multiple definition of `__tracepoint_mali_PM_WAKE_WAITERS'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:(__tracepoints+0x0): first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:2932: multiple definition of `__tracepoint_mali_PM_CA_SET_POLICY'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.c:64: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:3163: multiple definition of `__tracepoint_mali_PM_CURRENT_POLICY_TERM'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.c:71: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:3164: multiple definition of `__tracepoint_mali_PM_CURRENT_POLICY_INIT'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/drivers/gpu/arm/midgard_for_linux/./mali_linux_kbase_trace.h:27: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:3165: multiple definition of `__tracepoint_mali_PM_SET_POLICY'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/drivers/gpu/arm/midgard_for_linux/./mali_linux_kbase_trace.h:27: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:3918: multiple definition of `__tracepoint_mali_PM_GPU_OFF'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/include/linux/perf_event.h:891: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:1104: multiple definition of `__tracepoint_mali_PM_GPU_ON'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/include/linux/trace_events.h:626: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:1113: multiple definition of `__tracepoint_mali_PM_CORES_CHANGE_AVAILABLE_TILER'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/drivers/gpu/arm/midgard_for_linux/./mali_linux_kbase_trace.h:73: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:1126: multiple definition of `__tracepoint_mali_PM_CORES_CHANGE_AVAILABLE'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/drivers/gpu/arm/midgard_for_linux/./mali_linux_kbase_trace.h:73: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:3880: multiple definition of `__tracepoint_mali_PM_CORES_AVAILABLE_TILER'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/include/linux/perf_event.h:893: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:3889: multiple definition of `__tracepoint_mali_PM_CORES_AVAILABLE'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/drivers/gpu/arm/midgard_for_linux/./mali_linux_kbase_trace.h:109: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:3850: multiple definition of `__tracepoint_mali_PM_RELEASE_CHANGE_SHADER_INUSE'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/drivers/gpu/arm/midgard_for_linux/./mali_linux_kbase_trace.h:109: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:3862: multiple definition of `__tracepoint_mali_PM_REGISTER_CHANGE_SHADER_INUSE'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/include/linux/perf_event.h:891: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:3866: multiple definition of `__tracepoint_mali_PM_REGISTER_CHANGE_SHADER_NEEDED'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/include/linux/trace_events.h:626: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:3836: multiple definition of `__tracepoint_mali_PM_REQUEST_CHANGE_SHADER_NEEDED'
drivers/gpu/arm/midgard_for_linux/mali_kbase_trace_timeline.o:/include/linux/trace_events.h:450: first defined here
drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.o:/drivers/gpu/arm/midgard_for_linux/mali_kbase_core_linux.c:3808: multiple definition of `__tracepoint_mali_PM_UNREQUEST_CHANGE_SHADER_NEEDED'
[... cut ...]
make[4]: *** [drivers/gpu/arm/midgard_for_linux/midgard_kbase.o] Error 1
scripts/Makefile.build:484: recipe for target 'drivers/gpu/arm/midgard_for_linux' failed
make[3]: *** [drivers/gpu/arm/midgard_for_linux] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu/arm' failed
make[2]: *** [drivers/gpu/arm] Error 2
scripts/Makefile.build:484: recipe for target 'drivers/gpu' failed
make[1]: *** [drivers/gpu] Error 2
Makefile:1036: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```