rockchip-linux/kernel #153: rtl8812au support status

teseo-sw, 2019-04-08:
I have a USB WiFi dongle equipped with rtl8812au. I noticed that source code for the required chipset is in `drivers/net/wireless/rockchip_wlan/rtl8812au` but I am unable to even make it compile. Can I expect a working driver in the near future?

Caesar-github, 2019-04-19:
That rtl8812au should be supported by wifi driver. can you upload the error log by your compile.

teseo-sw, 2019-04-19:
rkl9912au in not included in: https://github.com/rockchip-linux/kernel/blob/develop-4.4/drivers/net/wireless/rockchip_wlan/Kconfig#L46
I added it manually to the Kconfig and set CONFIG_RTL8812AU=y. There is no error, just the code for the driver never gets built for some reason.

Caesar-github, 2019-04-19:
Maybe mrfixit3000's commit is helpful for you.

https://github.com/mrfixit3000/kernel/commit/24cc9735238f00919dbabb91142a420bcd54a086

teseo-sw, 2019-04-19:
I applied changes from that config to mine but nothing happened, rtl8812au still not getting in.