rockchip-linux/kernel #154: Can't find /dev/rga driver

delavadiya, 2019-04-09:
I compiled latest kernel and select rockchip-rga as Module.

When I list modules, it showed as below.
Module                  Size  Used by
rockchip_rga           24576  0
videobuf2_dma_sg       20480  1 rockchip_rga
v4l2_mem2mem           20480  1 rockchip_rga
mali                  262144  1
dw_hdmi_i2s_audio      16384  0
rtc_ds1307             20480  0
ip_tables              24576  0
x_tables               32768  1 ip_tables
autofs4                40960  2
uas                    20480  0
usb_storage            61440  1 uas

But I can't find that how to use this rockchip-rga module and how to installed rga driver in /dev directory?

Caesar-github, 2019-04-19:
/dev/rga should enable the rga2 with "CONFIG_ROCKCHIP_RGA2=y"

  rga: rga@ff680000 {
+               compatible = "rockchip,rga2";
+               dev_mode = <1>;
+               reg = <0x0 0xff680000 0x0 0x1000>;
+               interrupts = <GIC_SPI 55 IRQ_TYPE_LEVEL_HIGH 0>;
+               clocks = <&cru ACLK_RGA>, <&cru HCLK_RGA>, <&cru SCLK_RGA_CORE>;
+               clock-names = "aclk_rga", "hclk_rga", "clk_rga";
+               power-domains = <&power RK3399_PD_RGA>;
+               dma-coherent;
+               status = "okay";
+       };

Caesar-github, 2019-04-19:
maybe you can refer to the rk3399/rk3399pro configure.

delavadiya, 2019-05-06:
When compile RGA2 module give this warning "__dma_flush_range" [drivers/video/rockchip/rga2/rga2.ko] undefined! and when load rga2 module, dmesg gives "__dma_flush_range" undefined. because of this error module cannot be loaded.

delavadiya, 2019-05-06:
After removing this error-> Unknown symbol __dma_flush_range.
I compiled RGA2 driver and it's generate rga2.ko module without any warning.
I also set CONFIG_ROCKCHIP_RGA2=m and load module using modprobe rga2. 
After checked in /dev node but not found rga. Also dmesg nothing shown any error related to rga2.

Varsha1807, 2025-05-21:
I have erorr while run on luckfox lyra rk3506 board!

8   4075   4075 E im2d_rga_context: failed to open /dev/rga:No such file or directory.
import dma_fd error!

how to rectify this problem.