rockchip-linux/kernel #166: CIF_ISP_PIC_SIZE_ERROR error on rkisp10

ozone999, 2019-06-28:
I am developing a camera application using nanopc-t4 (Friendly desktop - ubuntu18).

Camera streaming stops with the following error while streaming camera.

[45828.009220] rkisp10 ff910000.cif_isp: CIF_ISP_PIC_SIZE_ERROR (0x00000001)

Please help me.

davidmarshalla, 2020-03-10:
have you resolved this problem? I met the same promble with nanopi-neo4,CIF_PIC_SIZE_ERROR，and my camera output image is not right

hjpark-rdv, 2020-03-24:
@davidmarshalla 
You must check your cif_isp10 version.
In my case, upgrading the version of cif_isp10 from 2.1 to 2.3 solved the problem.

https://github.com/friendlyarm/kernel-rockchip/blob/nanopi4-linux-v4.4.y/drivers/media/platform/rk-isp10/cif_isp10_version.h

davidmarshalla, 2020-03-31:
> @davidmarshalla
> You must check your cif_isp10 version.
> In my case, upgrading the version of cif_isp10 from 2.1 to 2.3 solved the problem.
> 
> https://github.com/friendlyarm/kernel-rockchip/blob/nanopi4-linux-v4.4.y/drivers/media/platform/rk-isp10/cif_isp10_version.h

thanks  for giving advice, but i use rkisp instead of cif_isp , so is there any other method?