rockchip-linux/kernel #167: firefly rk3399 uvc-gadget 无法正常工作

herochengm, 2019-06-28:
启动uvc-gadget 图像数据吐不出来。usb 总线上也没报文

[ 2001.552024] fusb302 4-0022: connection has disconnected
[ 2001.675553] configfs-gadget gadget: uvc_function_disable
[ 2001.676556] android_work: sent uevent USB_STATE=DISCONNECTED
[ 2001.899830] fusb302 4-0022: connection has disconnected
[ 2002.568333] rockchip-dwc3 usb0: Peripheral disconnect timeout
[ 2002.568856] rockchip-dwc3 usb0: USB unconnected
[ 2003.099616] wl_run_escan: LEGACY_SCAN sync ID: 20, bssidx: 0
[ 2006.857098] fusb302 4-0022: CC connected in CC2 as UFP
[ 2006.860848] rockchip-dwc3 usb0: USB peripheral connected
[ 2007.216517] phy phy-ff770000.syscon:usb2-phy@e450.7: charger = USB_SDP_CHARGER
[ 2007.419878] android_work: sent uevent USB_STATE=CONNECTED
[ 2007.642941] android_work: sent uevent USB_STATE=DISCONNECTED
[ 2007.742529] android_work: sent uevent USB_STATE=CONNECTED
[ 2007.789420] configfs-gadget gadget: high-speed config #1: c
[ 2007.789954] configfs-gadget gadget: uvc_function_set_alt(0, 0)
[ 2007.790483] configfs-gadget gadget: reset UVC Control
[ 2007.790962] configfs-gadget gadget: uvc_function_set_alt(1, 0)
[ 2007.791802] configfs-gadget gadget: uvc_function_set_alt(1, 0)
[ 2007.792758] android_work: sent uevent USB_STATE=CONFIGURED
[ 2007.816771] configfs-gadget gadget: uvc_function_set_alt(1, 0)
[ 2009.790380] fusb302 4-0022: PD disabled
[ 2017.081964] configfs-gadget gadget: uvc_function_set_alt(1, 1)
[ 2017.082923] configfs-gadget gadget: reset UVC
[ 2017.251525] uvcg_queue_enable 283
[ 2017.324460] uvcg_video_pump 330
[ 2017.324773] uvc_video_encode_isoc 111
[ 2017.325119] uvc_video_encode_data 51
[ 2017.325557] uvcg_video_pump 330
[ 2017.325858] uvc_video_encode_isoc 111
[ 2017.326203] uvc_video_encode_data 51
[ 2017.326573] uvcg_video_pump 330
[ 2017.326874] uvc_video_encode_isoc 111
[ 2017.327218] uvc_video_encode_data 51
[ 2017.327586] uvcg_video_pump 330
[ 2017.327886] uvc_video_encode_isoc 111
[ 2017.328231] uvc_video_encode_data 51
[ 2017.328597] uvcg_video_pump 312
[ 2017.357910] uvcg_video_pump 312
[ 2017.391567] uvcg_video_pump 312
[ 2018.393847] VS request completed with status -104.
[ 2018.394622] VS request completed with status -104.
[ 2018.395063] VS request completed with status -104.
[ 2018.395544] VS request completed with status -104.
[ 2018.395994] uvcg_queue_enable 292
[ 2037.979188] configfs-gadget gadget: uvc_function_set_alt(1, 0)
[ 2037.981121] uvcg_queue_enable 292
[ 2123.103513] wl_run_escan: LEGACY_SCAN sync ID: 21, bssidx: 0
[ 2243.100610] wl_run_escan: LEGACY_SCAN sync ID: 22, bssidx: 0
[ 2328.530855] uvcg_queue_enable 292

xiaoshangpeng, 2020-05-26:
请问该问题解决了吗？

Learner379, 2020-11-25:
Hi All,

I am also facing the same issue. Can any one please help me out to resolve this issue. 

Thanks.