rockchip-linux/kernel #169: Camera picture is green-ish with rkisp10

teseo-sw, 2019-07-01:
I'm using an OV13850 sensor with rkisp10 framework. If I try to capture a stream the image colors are biased towards a green-ish tone and more generally the colors seems to not be accurately reproduced. I'm 100% sure the sensor's registers configuration and image conversion process on my side is correct (the same exact setup under Android produces correct images) so there has to be a problem with rockchip's ISP driver. Steps to reproduce:

    v4l2-ctl --set-fmt-video=width=2112,height=1568,pixelformat=NV12 --stream-mmap --stream-count=1 -d /dev/video2 --stream-to=/tmp/image.nv12

Resulting image (captured from latest Linux):
[image]
Expected image (captured from Android):
[image]

teseo-sw, 2019-07-04:
Any hints anyone? @rkhuangtao

teseo-sw, 2019-07-08:
Is this issue-tracker maintained? Who should I contact to get ISP/camera support?

rkhuangtao, 2019-07-08:
You can contact ov13850 author, william.hu at rock-chips.com. Try send email.
The best way is report bug through https://redmine.rockchip.com.cn/.

teseo-sw, 2019-07-08:
Thank you so much, I'll try to contact him.

DecksLabs, 2019-07-11:
@teseo-sw 
Any success? I managed to enable isp1 with this camera, but still I get green images :(

teseo-sw, 2019-07-12:
You will need camera_engine_cifisp or camera_engine_rkisp along with an IQ calibration file suited for your sensor. We have one for OV13850 R1A which produces good colors:

[image]

DecksLabs, 2019-07-12:
@teseo-sw Thank you!
I tried to avoid it as much as possible as the camera_engine_rkisp contains loads of binary libs and generally seems to be pain to build :(

teseo-sw, 2019-07-12:
Yes, it is a pain to build and contains binary only stuff but AFAIK it's the only way to get good images out of camera pipeline.

Caesar-github, 2019-07-12:
To be honest,  the refer document found on https://github.com/rockchip-linux/docs/blob/master/Linux%20reference%20documents/Camera/Rockchip_Developer_Guide_Linux_Camera_EN.pdf

Indeed, the camera_engine_cifisp or camera_engine_rkisp was used for rkis10 or rkisp1.
We have used the rkisp1 with camera_engine_rkisp on linux platform by default.

DecksLabs, 2019-07-25:
@teseo-sw @Caesar-github 
Although I built camera_engine_rkisp and I'm able to take pictures using it, it still produce somewhat greenish pictures. I got ov13850 cam module and imx219. For the imx219 don't have required calibration iq files.

Any tips here? Which plugin for gstreamer to use, which kernel version? I'm a bit lost since I think I tried pretty much everything that is available on rockchip github.

teseo-sw, 2019-07-26:
FWIW I use latest develop-4.4 and cifisp_demo to capture images. I feel you pain, I tried pretty much everything myself. The only thing I'm sure about is that you'll need proper IQ calib for you specific sensors in order to get acceptable results.

DecksLabs, 2019-07-26:
@teseo-sw Oh, so you actually used camera_engine_cifisp not rkisp one, right? That might be game changer :)
Another question, if I might ask. I also use ov13850 module. Did you use [calibration file](https://github.com/rockchip-linux/camera_engine_rkisp/blob/master/iqfiles/ov13850_CMK-CT0116_Largan-50013A1.xml) or some other? If other, then how can you create one, it's kinda hard to get any data on the topic.

teseo-sw, 2019-07-26:
I use OV13850 with the attached calibration file. I do not know how to generate them, I asked our chip seller to cooperate with Rockchip in order to get a working IQ file for our specific sensor. We are still waiting.
[OV13850_V0.1.7.txt](https://github.com/rockchip-linux/kernel/files/3436078/OV13850_V0.1.7.txt)

rothn, 2022-09-12:
I think it might have something to do with the default initialization in the rkisp1 driver. The function that initializes streams, `rkisp1_params_configure()`, sets defaults for white balance and exposure control, but not lens shade control (LSC). It's possible that we need to initialize this to a more sensible default to avoid the green color.