rockchip-linux/kernel #183: Missing UART messages while using camera

barzelshir, 2019-09-09:
I'm reading UART messages through /dev/ttyS4 while reading frames from /dev/video0
From time to time I've noticed that messages get lost
What can be the cause?
Thanks

barzelshir, 2019-10-30:
Hi @masf7g, why you referenced this commit?
Why is it relevant ?

masf7g, 2019-10-30:
Hi @barzelshir, I did not mention that commit in any of my comments. I just did merge and that commit was already referenced in the comments of some other commits which I merged in and github detected it.

barzelshir, 2019-10-31:
> Hi @barzelshir, I did not mention that commit in any of my comments. I just did merge and that commit was already referenced in the comments of some other commits which I merged in and github detected it.

Hi @masf7g I can't see which commits also referenced this issue, can you tell me what are the commits numbers?