rockchip-linux/kernel #197: Compilation failed `drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.o` with `rockchip_linux_defconfig`

DamonBlais, 2019-12-02:
I am using `rockchip_linux_defconfig` and the latest `develop-4.4` OR `stable-4.4-rk3399-linux`

e.g: " **This repository doesn't build.** "

# stable-4.4-rk3399-linux

```
  CC [M]  drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.o
drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.c: In function 'dhd_attach':
drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.c:9454:4: warning: 'strncat' specified bound 2 equals source length [-Wstringop-overflow=]
error, forbidden warning:dhd_linux.c:9454
 9454 |    strncat(if_name, "%d", 2);
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~
make[6]: *** [scripts/Makefile.build:278: drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.o] Error 1
make[6]: *** Deleting file 'drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.o'
make[5]: *** [scripts/Makefile.build:484: drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd] Error 2
make[4]: *** [scripts/Makefile.build:484: drivers/net/wireless/rockchip_wlan/rkwifi] Error 2
make[3]: *** [scripts/Makefile.build:484: drivers/net/wireless/rockchip_wlan] Error 2
make[2]: *** [scripts/Makefile.build:484: drivers/net/wireless] Error 2
make[1]: *** [scripts/Makefile.build:484: drivers/net] Error 2
make: *** [Makefile:1036: drivers] Error 2
```

# develop-4.4 

```
  CC [M]  drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.o
drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.c: In function 'dhd_attach':
drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.c:9454:4: warning: 'strncat' specified bound 2 equals source length [-Wstringop-overflow=]
error, forbidden warning:dhd_linux.c:9454
 9454 |    strncat(if_name, "%d", 2);
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~
make[6]: *** [scripts/Makefile.build:278: drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.o] Error 1
make[6]: *** Deleting file 'drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_linux.o'
make[5]: *** [scripts/Makefile.build:484: drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd] Error 2
make[4]: *** [scripts/Makefile.build:484: drivers/net/wireless/rockchip_wlan/rkwifi] Error 2
make[3]: *** [scripts/Makefile.build:484: drivers/net/wireless/rockchip_wlan] Error 2
make[2]: *** [scripts/Makefile.build:484: drivers/net/wireless] Error 2
make[1]: *** [scripts/Makefile.build:484: drivers/net] Error 2
make: *** [Makefile:1036: drivers] Error 2
```

DamonBlais, 2019-12-02:
Commands used to compile (failed):

```
$ make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- rockchip_linux_defconfig
$ make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- -j9
```

DamonBlais, 2019-12-02:
cross-compiler is `aarch64-linux-gnu-gcc`

```
$ aarch64-linux-gnu-gcc --version
aarch64-linux-gnu-gcc (GCC) 9.2.1 20190827 (Red Hat Cross 9.2.1-1)
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Provided by:

```
$ dnf info gcc-aarch64-linux-gnu
Installed Packages
Name         : gcc-aarch64-linux-gnu
Version      : 9.2.1
Release      : 1.fc31
Architecture : x86_64
Size         : 55 M
Source       : cross-gcc-9.2.1-1.fc31.src.rpm
```

trecetp, 2020-02-27:
A bit late, but can help someone else. Try adding `-Wno-stringop-overflow` to `KCFLAGS`:
```
$ KCFLAGS="-Wno-stringop-overflow" make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- -j9 rockchip_linux_defconfig all
```