rockchip-linux/kernel #202: rkisp1: MIPI mis error: 0x00010000

hkurj, 2020-01-02:
run shell
`v4l2-ctl -d /dev/video0 --set-selection=target=crop,width=640,height=320 --set-fmt-video=width=640,height=320,pixelformat=NV21 --stream-mmap --stream-to=mp.out`

when run over start_stream ，an error occurred “rkisp1: MIPI mis error: 0x00010000”
It is read from the register, but I can't see the detailed register information in the datasheet

jaydcarlson, 2020-10-05:
Did you ever figure out what was going on?

I can confirm I'm having the same issue with an OV2740 image sensor operating 1920x1080x10-bit RAW @ 60 fps. I'm using 2-lane mode with a 400 MHz clock (800 Mbps transfer rate). I've verified the correct number of lanes and correct frequencies are being output.

```
[  364.391006] rkisp1: waiting on params stream on event timeout
[  364.391597] rkisp1 ff4a0000.rkisp1: can not get first iq setting in stream on
[b][  364.392419] rkisp1: MIPI error: err2: 0x00010000[/b]
[  364.394693] rockchip-mipi-dphy-rx ff2f0000.mipi-dphy-rx0: data rate hsfreq range set to 899 mbps
[  364.395541] rockchip-mipi-dphy-rx ff2f0000.mipi-dphy-rx0: enabled 2 lanes for reception
[  364.396334] ov2740 2-0030: __ov2740_start_stream() called
[  364.396898] ov2740 2-0030: ov2740_set_ctrl() V4L2_CID_VBLANK called
[  364.397504] ov2740 2-0030: ov2740_set_ctrl() V4L2_CID_VBLANK=8
[  364.399508] ov2740 2-0030: ov2740_set_ctrl() V4L2_CID_EXPOSURE=1080
[  364.401722] ov2740 2-0030: ov2740_set_ctrl() V4L2_CID_ANALOGUE_GAIN=16
[  364.402412] ov2740 2-0030: ov2740_set_ctrl() V4L2_CID_TEST_PATTERN called
[  364.403109] ov2740 2-0030: Disabling test pattern
[b]select timeout[/b]
[b][  367.410019] rkisp1: waiting on event return error 0[/b]
[  367.411561] rockchip-mipi-dphy-rx ff2f0000.mipi-dphy-rx0: disabled reception
[  367.412254] ov2740 2-0030: __ov2740_stop_stream() called
[  367.430933] ov2740 2-0030: ov2740_s_power() called
```

hkurj, 2020-10-12:
It may be a clock problem

rageworx, 2023-04-12:
This MIS errors are must belong to your camera driver register configured to different clock base.
You need right clock matching with rk3399 ( 40MHz ? ) for MIPI register configuration of camera driver.