rockchip-linux/kernel #214: rk3328 can't playback rate up to 176.4k/192k via SPDIF

willardting, 2020-06-01:
Hello, contributors. @sjoerd-ccu @rkhuangtao @Sugar Zhang
     These days I've been testing spdif transmitting from rk3328, using firefly aio-rk3328-jd4 core board. I get stuck on playback of 88.2k/24bit, 96k/24bit, 176.4k/16bit(24bit) and 192k/16bit(24bit), as it sent nothing from spdifm1_tx pin.
     I used dmesg to get the kernel log as follows:
`[ 8252.009035] dma-pl330 ff1f0000.dmac: Reset Channel-4  CS-20400f FTC-1`
`[ 8267.250622] dma-pl330 ff1f0000.dmac: Reset Channel-4  CS-20400f FTC-2000`
     I used wav file with aplay command:
`aplay -D plughw:2,0 192000-16.WAV`
     and got the feedback message:
`Playing WAVE '192000-16.WAV' : Signed 16 bit Little Endian, Rate 192000 Hz, Stereo`
`^CAborted by signal Interrupt...`
`aplay: pcm_write:1939: write error: Input/output error`
     So what's the problem? Could somebody kindly test with the audio files as I mentioned above?
Sincerely,
     Willard.

willardting, 2020-06-02:
@sugaroneone  I ultimately found out a way to fix the problem, eliminating any errors:
```
--- a/sound/soc/rockchip/rockchip_spdif.c
+++ b/sound/soc/rockchip/rockchip_spdif.c
@@ -118,6 +118,7 @@ static int rk_spdif_hw_params(struct snd_pcm_substream *substream,
                val |= SPDIF_CFGR_VDW_20;
                break;
        case SNDRV_PCM_FORMAT_S24_LE:
+       case SNDRV_PCM_FORMAT_S24_3LE:
                val |= SPDIF_CFGR_VDW_24;
                break;
        default:
@@ -201,11 +202,14 @@ static struct snd_soc_dai_driver rk_spdif_dai = {
                .rates = (SNDRV_PCM_RATE_32000 |
                          SNDRV_PCM_RATE_44100 |
                          SNDRV_PCM_RATE_48000 |
+                         SNDRV_PCM_RATE_88200 |
                          SNDRV_PCM_RATE_96000 |
+                         SNDRV_PCM_RATE_176400 |
                          SNDRV_PCM_RATE_192000),
                .formats = (SNDRV_PCM_FMTBIT_S16_LE |
                            SNDRV_PCM_FMTBIT_S20_3LE |
-                           SNDRV_PCM_FMTBIT_S24_LE),
+                           SNDRV_PCM_FMTBIT_S24_LE |
+                           SNDRV_PCM_FMTBIT_S24_3LE),
        },
        .ops = &rk_spdif_dai_ops,
 };
@@ -338,7 +342,7 @@ static int rk_spdif_probe(struct platform_device *pdev)
 
        spdif->playback_dma_data.addr = res->start + SPDIF_SMPDR;
        spdif->playback_dma_data.addr_width = DMA_SLAVE_BUSWIDTH_4_BYTES;
-       spdif->playback_dma_data.maxburst = 4;
+       spdif->playback_dma_data.maxburst = 8;
 
        spdif->dev = &pdev->dev;
        dev_set_drvdata(&pdev->dev, spdif);
```
I simply changed the `spdif->playback_dma_data.maxburst` from 4 to 8, plus adding params supporting 88200 and 176400. Now I can smoothly "SPDIF out" 44100/16(24)bit, 48000/16(24)bit, 88200/16(24)bit, 96000/16(24)bit, 176400/16(24)bit, 192000/16(24)bit!!! I also tested with 32bit wav files, luckily it still works as well!

willardting, 2020-06-02:
After compiling the spdif module, I did the recommended steps as you said, playback 88200-24.wav, and this is the outcome:

1. 
```
closed
card: 0
device: 0
subdevice: 0
stream: PLAYBACK
id: ff010000.i2s-rk3328-hifi rk3328-hifi-0
name:
subname: subdevice #0
class: 0
subclass: 0
subdevices_count: 1
subdevices_avail: 1
512
18014398509481983
closed
closed
closed
card: 1
device: 0
subdevice: 0
stream: PLAYBACK
id: ff000000.i2s-i2s-hifi i2s-hifi-0
name:
subname: subdevice #0
class: 0
subclass: 0
subdevices_count: 1
subdevices_avail: 1
512
18014398509481983
closed
closed
access: MMAP_INTERLEAVED
format: S24_LE
subformat: STD
channels: 2
rate: 88200 (88200/1)
period_size: 11025
buffer_size: 44100
card: 2
device: 0
subdevice: 0
stream: PLAYBACK
id: ff030000.spdif-dit-hifi dit-hifi-0
name:
subname: subdevice #0
class: 0
subclass: 0
subdevices_count: 1
subdevices_avail: 0
512
18014398509481983
state: RUNNING
owner_pid   : 10590
trigger_time: 4092.515362153
tstamp      : 0.000000000
delay       : 35876
avail       : 8224
avail_max   : 11213
-----
hw_ptr      : 1452499
appl_ptr    : 1488375
tstamp_mode: NONE
period_step: 1
avail_min: 11025
start_threshold: 44100
[... 256 lines cut ...]
                      i2s1_out            0            0    11289600          0 0
          clk_i2s0_div                    0            0    75000000          0 0
             i2s0_pre                     0            0    75000000          0 0
                clk_i2s0                  0            0    75000000          0 0
             clk_i2s0_frac                0            0     3750000          0 0
          clk_tsp                         0            0    50000000          0 0
    pll_dpll                              1            1   664000000          0 0
       dpll                               1            1   664000000          0 0
          sclk_ddrc                       2            2   332000000          0 0
             aclk_ddrupctl                0            0   332000000          0 0
             clk_ddrupctl                 1            1   332000000          0 0
             clk_ddrmsch                  1            1   332000000          0 0
          dpll_core                       0            0   664000000          0 0
    pll_apll                              0            0    61439999          0 0
       apll                               0            0    61439999          0 0
          clk_pdm                         0            0    61439999          0 0
          apll_core                       0            0    61439999          0 0
 clk_hsadc_tsp                            0            0           0          0 0
 clk_jtag                                 1            1           0          0 0
```

So are these messages enough to spark any bug fix?

Thanks a lot,
Willard.

sugaroneone, 2020-06-03:
I think the root cause is the unalinged bursts xfer，plz enable the pl330 dmac debug and then dump the full kmsg.
`diff --git a/drivers/dma/pl330.c b/drivers/dma/pl330.c
index 5572334..35d715a 100644
--- a/drivers/dma/pl330.c
+++ b/drivers/dma/pl330.c
@@ -253,7 +253,7 @@ enum pl330_byteswap {
 /* Use this _only_ to wait on transient states */
 #define UNTIL(t, s)    while (!(_state(t) & (s))) cpu_relax();

-#ifdef PL330_DEBUG_MCGEN
+#if 1//def PL330_DEBUG_MCGEN
 static unsigned cmd_line;
 `