rockchip-linux/kernel #223: RK3328 HWRNG output Low quality random numbers

wevsty, 2020-08-25:
I was trying to use HWRNG with the RK3328, but when I tested /dev/hwrng I found that the output didn't pass the random number test.
This means that the quality of the output random numbers is very low and I'm not sure if this is a problem with the HWRNG code or a flaw in the chip itself.
The test results are as follows.
```bash
root@rockchip-linux:~# hexdump -n 128 /dev/hwrng
0000000 b978 e14c 78fc ec33 b501 35b3 6470 2724
0000010 2918 084d 2a07 461f 81d9 6f60 3899 ace4
0000020 b4c0 54c9 110e 5097 9be9 244a d7c8 e18e
0000030 665a 9e60 d80c f820 cf15 7ecc e660 8f67
0000040 99fc 44aa a6d2 0224 588a 00fe 4a8d 44d8
0000050 987f 6d4b 4921 021c 079c da28 44a0 ac64
0000060 1e5b d97c ccff 5092 5489 38ca 72b2 6ba1
0000070 3f66 1d08 2dc9 4b58 e161 cec4 5303 cb02
0000080

root@rockchip-linux:~# cat /dev/hwrng | rngtest -c 100000
rngtest 6.7
Copyright (c) 2004 by Henrique de Moraes Holschuh
This is free software; see the source for copying conditions.  There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

rngtest: starting FIPS tests...
rngtest: bits received from input: 2000000032
rngtest: FIPS 140-2 successes: 0
rngtest: FIPS 140-2 failures: 100000
rngtest: FIPS 140-2(2001-10-10) Monobit: 100000
rngtest: FIPS 140-2(2001-10-10) Poker: 100000
rngtest: FIPS 140-2(2001-10-10) Runs: 100000
rngtest: FIPS 140-2(2001-10-10) Long run: 313
rngtest: FIPS 140-2(2001-10-10) Continuous run: 0
rngtest: input channel speed: (min=404.541; avg=1073.877; max=4882812.500)Kibits/s
rngtest: FIPS tests speed: (min=4.523; avg=18.105; max=59.605)Mibits/s
rngtest: Program run time: 1924394866 microseconds
```

From the hexdump results, the driver is working properly, but the quality of the output is not high.

If it is a driver problem is there a way to fix it?

wevsty, 2020-08-28:
After an email exchange, the advice I got was
Adjust the value of CRYPTO_V1_TRNG_SAMPLE_PERIOD.

File path :  kernel/drivers/char/hw_random/rockchip-rng.c 
Expanding CRYPTO_V1_TRNG_SAMPLE_PERIOD can strengthen the quality of the random number.

closed 2020-08-28 by wevsty

tstevens, 2021-04-06:
@wevsty was there a value of CRYPTO_V1_TRNG_SAMPLE_PERIOD that worked for you?

wevsty, 2021-04-07:
> 
> 
> @wevsty was there a value of CRYPTO_V1_TRNG_SAMPLE_PERIOD that worked for you?

I suggest you try adjusting it to above 1000.