rockchip-linux/kernel #242: RK3288 linux with LVDS display

msmsubair, 2021-03-03:
Hello,
Anybody who got success in getting LVDS display working on RK3288 ?
my display is working but backlight is not switching on except when the board boots.
any help is appreciated.

christianhalter, 2021-07-20:
Hi @msmsubair ,
Can I see your dts? I can't even get it work on boot. The display is OK, I've tested it with other board.
Thanks in advance

msmsubair, 2021-07-21:
Thank you Christian, it worked, it as an issue of the gpio used in the config  , this is a board which bought from china, they do have good boards but very few provide details like this.

christianhalter, 2021-07-21:
I'm glad. I asked because I am trying to get a LVDS display working in my board, and it's not working for me. I'm sure something is missing in my dts, that's why I asked for your's. I'll check also the gpio config. Thanks!

msmsubair, 2021-07-21:
i can share my dts,  see below, share  your email, i can sent you the dts file as it is,  also which kernel you use ? 4.4. or 5.12

	backlight: backlight {
		compatible = "pwm-backlight";
		brightness-levels =  @<
			  0   1   2   3   4   5   6   7
			  8   9  10  11  12  13  14  15
			 16  17  18  19  20  21  22  23
			 24  25  26  27  28  29  30  31
			 32  33  34  35  36  37  38  39
			 40  41  42  43  44  45  46  47
			 48  49  50  51  52  53  54  55
			 56  57  58  59  60  61  62  63
			 64  65  66  67  68  69  70  71
			 72  73  74  75  76  77  78  79
			 80  81  82  83  84  85  86  87
			 88  89  90  91  92  93  94  95
			 96  97  98  99 100 101 102 103
			104 105 106 107 108 109 110 111
			112 113 114 115 116 117 118 119
			120 121 122 123 124 125 126 127
			128 129 130 131 132 133 134 135
			136 137 138 139 140 141 142 143
			144 145 146 147 148 149 150 151
			152 153 154 155 156 157 158 159
			160 161 162 163 164 165 166 167
			168 169 170 171 172 173 174 175
			176 177 178 179 180 181 182 183
			184 185 186 187 188 189 190 191
			192 193 194 195 196 197 198 199
			200 201 202 203 204 205 206 207
			208 209 210 211 212 213 214 215
			216 217 218 219 220 221 222 223
			224 225 226 227 228 229 230 231
			232 233 234 235 236 237 238 239
			240 241 242 243 244 245 246 247
			248 249 250 251 252 253 254 255>;
		default-brightness-level = <200>;
		enable-gpios = <&gpio7 2 GPIO_ACTIVE_HIGH >; //gpio7_A2  bl_en0
		
		power-supply = <&vcc_bl>;
		pwms = <&pwm1 0 25000 0>;//PWM_POLARITY_INVERTED
	};

&lvds_panel {
	status = "okay";
	compatible ="simple-panel";
	backlight = <&backlight>;    
	enable-gpios = <&gpio7 11 GPIO_ACTIVE_HIGH>;
	enable-delay-ms = <220>;
	bus-format = <MEDIA_BUS_FMT_RGB666_1X7X3_SPWG>;
	rockchip,data-mapping = "vesa";
	rockchip,data-width = <18>; 
	rockchip,output = "lvds";

	display-timings {
		native-mode = <&timing1>;
		timing1: timing1 {
			clock-frequency = <71000000>;
			hactive = <1280>;		
			vactive = <800>;		
			hback-porch = <140>;		
			hfront-porch = <18>;		
			vback-porch = <8>;		
			vfront-porch = <6>;		
			hsync-len = <10>;			
			vsync-len = <2>;			
			hsync-active = <8>;
			vsync-active = <0>;
			de-active = <0>;
			pixelclk-active = <0>;
		};
	};
};

&lvds_in_vopl {
	status = "okay";
};
&lvds_in_vopb {
	status = "disabled";
};
&vopl_out_lvds{
	status = "okay";
};
&vopb_out_lvds{
	status = "disabled";
};

&route_lvds {
    status = "okay";
    connect = <&vopl_out_lvds>;
};

&lvds {
    status = "okay";
};

&hdmi_in_vopb {
	status = "okay";
};
&hdmi_in_vopl {
	status = "disabled";
};
&vopb_out_hdmi{
	status = "okay";
};
&vopl_out_hdmi{
	status = "disabled";
};
&route_hdmi {
	status = "okay";
	connect = <&vopb_out_hdmi>;
};
&hdmi {
	status = "okay";
};

christianhalter, 2021-07-21:
Thank you very much! It is very helpful for me. 
I'm using kernel 4.4. 
My email is chalter@plantium.com
Thanks again

altimmons, 2022-07-16:
Ok I am back further- I have rockchipn3288 runnning an interactive tc display-  I wanted to use it as a large touch screen. But information is lacking.  Where would this go?

christianhalter, 2022-07-18:
> Ok I am back further- I have rockchipn3288 runnning an interactive tc display- I wanted to use it as a large touch screen. But information is lacking. Where would this go?

I don't think this is the right issue, as LVDS has the display information of the screen. Usually the touchscreen is a glass attached to the display, and the touch information goes through another connector, using for example I2C.
Do you have any information of the display and the touchscreen?