rockchip-linux/kernel #245: IMX415 different sensor modes (60+fps)

Consti10, 2021-05-13:
Hello,
The current driver for imx415 doesn't support anything else than 4K30fps.
However, the sensor is capable of much more resolution/framerate(s),like
1080p60
720p120

Do you know how to add these modes to the imx415 driver ?

Consti10, 2021-05-18:
Does anybody know how to use 30 fps with imx415 ?
It defaults to 25fps@4k, which is suboptimal.
The imx415 driver at least supports some 30fps modes, but I do not know how to tell either mpp or rkmedia to use 30fps.

yy834277137, 2022-06-14:
> Hello, The current driver for imx415 doesn't support anything else than 4K30fps. However, the sensor is capable of much more resolution/framerate(s),like 1080p60 720p120
> 
> Do you know how to add these modes to the imx415 driver ?

did u solve it?

bambooCZ, 2026-10-04:
https://github.com/bambooCZ/rk3588-imx415-4k60