rockchip-linux/kernel #250: list_add_valid

MannixGu, 2021-07-09:
rk3399 android 10  kernel 4.19.111 
mipi-dsi(vopl) + hdmi (vopb)

[    1.489640] Call trace:
[    1.489862]  __list_add_valid+0x70/0x90
[    1.490210]  rockchip_drm_register_sub_dev+0x40/0x70
[    1.490647]  dw_hdmi_rockchip_bind+0x510/0x658
[    1.491039]  component_bind_all+0x134/0x278
[    1.491407]  rockchip_drm_bind+0x2a8/0x1930
[    1.491775]  try_to_bring_up_master+0x144/0x1a8
[    1.492177]  component_master_add_with_match+0xd4/0x130
[    1.492635]  rockchip_drm_platform_probe+0x178/0x2e8
[    1.493072]  platform_drv_probe+0x50/0xa8
[    1.493429]  really_probe+0x200/0x2b0
[    1.493752]  driver_probe_device+0x58/0x100
[    1.494121]  __device_attach_driver+0x90/0xc0
[    1.494512]  bus_for_each_drv+0x70/0xc8
[    1.494857]  __device_attach+0xdc/0x138
[    1.495203]  device_initial_probe+0x10/0x18
[    1.495571]  bus_probe_device+0x94/0xa0
[    1.495917]  deferred_probe_work_func+0x80/0xb8
[    1.496321]  process_one_work+0x1a0/0x418
[    1.496678]  worker_thread+0x1e4/0x3e0
[    1.497014]  kthread+0x12c/0x158
[    1.497304]  ret_from_fork+0x10/0x18

notime2d8, 2022-01-25:
Did you ever find a solution to this? I'm having the same issue on Android 11 and 4.19-rt87.

MannixGu, 2022-01-25:
Maybe you need recheck the dts config. I delete some node cause this error!