rockchip-linux/kernel #261: [kernel 4.19.206] Error building rockhip_wlan

avafinger, 2021-11-20:
I get this error building latest 4.19 while no error building the previous version:

```
drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_pno.c: In function '_dhd_pno_get_for_batch':
drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_pno.c:3172:23: error: passing argument 1 of 'waitqueue_active' from incompatible pointer type [-Werror=incompatible-pointer-types]
  if (waitqueue_active(&_pno_state->get_batch_done.wait))
                       ^
In file included from ./include/linux/pid.h:7:0,
                 from ./include/linux/sched.h:14,
                 from ./arch/arm64/include/asm/compat.h:25,
                 from ./arch/arm64/include/asm/stat.h:24,
                 from ./include/linux/stat.h:6,
                 from ./include/linux/module.h:10,
                 from drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/include/linuxver.h:55,
                 from drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/include/linux_osl.h:295,
                 from drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/include/osl.h:48,
                 from drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_pno.c:37:
./include/linux/wait.h:126:19: note: expected 'struct wait_queue_head *' but argument is of type 'struct swait_queue_head *'
 static inline int waitqueue_active(struct wait_queue_head *wq_head)
                   ^~~~~~~~~~~~~~~~
drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_pno.c: At top level:
cc1: warning: unrecognized command line option '-Wno-non-literal-null-conversion'
cc1: warning: unrecognized command line option '-Wno-parentheses-equality'
cc1: some warnings being treated as errors
scripts/Makefile.build:333: recipe for target 'drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_pno.o' failed
make[6]: *** [drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_pno.o] Error 1
scripts/Makefile.build:637: recipe for target 'drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd' failed
make[5]: *** [drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd] Error 2
scripts/Makefile.build:637: recipe for target 'drivers/net/wireless/rockchip_wlan/rkwifi' failed
make[4]: *** [drivers/net/wireless/rockchip_wlan/rkwifi] Error 2
scripts/Makefile.build:637: recipe for target 'drivers/net/wireless/rockchip_wlan' failed
make[3]: *** [drivers/net/wireless/rockchip_wlan] Error 2

```

How to fix this?

jocker-xie, 2021-12-15:
Hello, have you solved this problem yet？
Temporarily shut down this driver to test its real-time performance. Is it the same with you

```
./cyclictest -t20 -p 80 -n -i 10000 -l 10000
# /dev/cpu_dma_latency set to 0us
policy: fifo: loadavg: 2.60 1.21 0.56 1/219 828          

T: 0 (  807) P:80 I:10000 C:  10000 Min:     21 Act:   63 Avg:   67 Max:     177
T: 1 (  808) P:80 I:10500 C:   9524 Min:     19 Act:   64 Avg:   68 Max:     190
T: 2 (  809) P:80 I:11000 C:   9091 Min:     20 Act:   64 Avg:   65 Max:     233
T: 3 (  810) P:80 I:11500 C:   8695 Min:     18 Act:   74 Avg:   85 Max:     250
T: 4 (  811) P:80 I:12000 C:   8333 Min:     22 Act:   62 Avg:   65 Max:     167
T: 5 (  812) P:80 I:12500 C:   8000 Min:     20 Act:   64 Avg:   67 Max:     167
T: 6 (  813) P:80 I:13000 C:   7692 Min:     21 Act:   61 Avg:   67 Max:     177
T: 7 (  814) P:80 I:13500 C:   7407 Min:     20 Act:   65 Avg:   66 Max:     231
...
```

avafinger, 2021-12-16:
No.
The driver seems to be leftover from a non rt kernel.

jocker-xie, 2021-12-17:
Have you tried xenomai transplantation?

MehmetTektek, 2021-12-25:
With the transplantation IPIPE patch is critual. Check the kernel sometimes little tweaks with kernel solves the problem

avafinger, 2021-12-25:
> With the transplantation IPIPE patch is critual

Can you point to the patch? I am not sure it's so trivial. Do you have rockhip_wlan working?

NHellFire, 2022-01-03:
As it's using simple wait queue, it should call `swait_active` and not `waitqueue_active`

[rockchip_wlan-rt.patch.txt](https://github.com/rockchip-linux/kernel/files/7800155/rockchip_wlan-rt.patch.txt)

avafinger, 2022-01-16:
@NHellFire ,

Thank you for the fix and sorry for the long delay to get back to you.

closed 2022-01-16 by avafinger

ych325608, 2024-01-27:
Hello! After I replaced the RT kernel, the WiFi driver was lost. Do you know the reason