rockchip-linux/kernel #270: Hi, is there any support kernel source of mainline kernel lts version? (LTS version is 5.15.67 when I post)

TurnOffNOD, 2022-09-10:
Hi, is there any support latest LTS version kernel source of rk3568? Which is 5.15.67 when I post this.
Official distribution latest LTS kernel will have NPU, GPU, VPU and hardware encoding/decoding working, so I post the issue here.

mo123, 2022-10-13:
@TurnOffNOD Have you looked at Armbian?
They use the 5.19, 6.x kernel for several Rockchip Development boards.
https://github.com/armbian/build

TurnOffNOD, 2022-10-15:
> @TurnOffNOD Have you looked at Armbian? They use the 5.19, 6.x kernel for several Rockchip Development boards. https://github.com/armbian/build

Armbian has no rk3568 npu support, perhaps has no rk3568 arm support. That's why I am asking here.

wtarreau, 2022-10-29:
Note that anyway the Rockchip kernel has very little to do with the Linux kernel. They both diverged 13 years ago, at 2.6.32, and while they periodically *try* to remerge new stuff, there are many conflicts which are not always resolved correctly and when running a diff between the two (which is as large as a regular kernel), you see lots of places where bugs were not correctly fixed, and will never be since there's no reliable way to spot them.

Rockchip's kernel team ought to *rebase* their work on top of new kernels, not *merge* it. They're unintentionally hiding and burying  bugs forever and their kernel very likely contains thousands of bugs and vulnerabilities that will never get fixed. Nobody must ever use them, they're unsafe and unauditable. The only way to use their hardware reliably is by waiting a year or two that it gets supported by mainline. It's painful but I'm hardly seeing any other option.