rockchip-linux/kernel #271: filesystem hangs with kernel: Internal error: Oops: 96000004 [#1] PREEMPT SMP during copy operation

Jimi-James, 2022-10-03:
On Armbian Buster, Rock64 (RK3328). This was also happening on the latest backported version of 5.10, I think 5.10.72? I tried upgrading to fix it to no avail.

I have a couple 8TB HDDs that I use as backups. All I did was reformat one of them as btrfs and start copying the data back over from the other (still EXT4). I tried rsync and simple cp. Without fail, a few minutes into the operation, the copy operation will hang indefinitely, and all open terminals receive these kernel messages:

```
Message from syslogd@localhost at Oct  1 16:59:57 ...
 kernel:[  339.539365] Internal error: Oops: 96000004 [#1] PREEMPT SMP

Message from syslogd@localhost at Oct  1 16:59:57 ...
 kernel:[  339.560387] Code: cb0100a3 cb0d015c 7100019f 9b167c00 (9adb2400) 
```

After that, every subsequent copy operation from the EXT4 drive to this new btrfs partition will hang until I reboot the rock64.

However, the drive otherwise continues to work fine. fsck and btrfs scrub find no problems. The super weird thing is I can still write files, and copy a small shell script from the home folder, all just fine--but copying from the other 8TB drive will immediately hang. I have had zero issues with either drive until trying btrfs now. I'm able to rsync between them just fine when they're both EXT4.

[journalctl.txt](https://github.com/rockchip-linux/kernel/files/9694439/journalctl.txt)
[specific-error.txt](https://github.com/rockchip-linux/kernel/files/9694440/specific-error.txt)

Kimi-Jimi-James, 2023-01-15:
This error has finally just now occurred on ext4... So I guess it's not exclusive to btrfs. Googling it now, without looking for just btrfs hits, it seems to be widespread across rock64 boards. Maybe even a hardware issue? I haven't finished researching.

demaniak, 2024-04-07:
Hit this problem on oDroid C2, Armbian Bookwork (happened on bullseye also).

Got a mix of disks and filesystems attached, with things running on top.
- ext4
- xfs

Samba, NFS and GlusterFS.

I saw the hang before I installed GlusterFS, so I don't believe that is involved, but the rest was present.

The hints I got from a terminal that was open before the hand happened:
```
Message from syslogd@files at Apr  7 10:11:19 ...
 kernel:[213832.467099] Internal error: Oops: 96000005 [#1] PREEMPT SMP

Message from syslogd@files at Apr  7 10:11:19 ...
 kernel:[213832.647127] Process .NET ThreadPool (pid: 489238, stack limit = 0x00000000bf005c5e)

Message from syslogd@files at Apr  7 10:11:19 ...
 kernel:[213832.730445] Code: a903e7b8 b9402262 b9401020 11000400 (f8626ad8) 

Message from syslogd@files at Apr  7 10:11:19 ...
 kernel:[213832.467099] Internal error: Oops: 96000005 [#1] PREEMPT SMP

Message from syslogd@files at Apr  7 10:11:19 ...
 kernel:[213832.647127] Process .NET ThreadPool (pid: 489238, stack limit = 0x00000000bf005c5e)

Message from syslogd@files at Apr  7 10:11:19 ...
 kernel:[213832.730445] Code: a903e7b8 b9402262 b9401020 11000400 (f8626ad8) 
 ```
 
 Point being, this does not seem to be isolated to just one platform/OS