rockchip-linux/kernel #273: I can't install kubernetes because module br_netfilter not found in ubuntu server 20.04

jikunchong, 2022-10-15:
embedded board: Radxa Rock 5b
system image: [rock-5b-ubuntu-focal-server-arm64-20220930-1451-gpt.img.xz](https://github.com/radxa/debos-radxa/releases/tag/20220930-1224)
ubuntu server version: 20.04.5

step docs:  https://kubesphere.io/docs/v3.3/quick-start/all-in-one-on-linux/
installer: [kubekey v2.2.2](https://github.com/kubesphere/kubekey/tree/v2.2.2)

```shell
./kk create cluster --with-kubernetes v1.23.8
```

```shell
22:44:25 CST [InitKubernetesModule] Init cluster using kubeadm
22:44:26 CST stdout: [rock-5b]
W1015 22:44:25.913328   13561 utils.go:69] The recommended value for "clusterDNS" in "KubeletConfiguration" is: [10.233.0.10]; the provided value is: [169.254.25.10]
[init] Using Kubernetes version: v1.23.8
[preflight] Running pre-flight checks
        [WARNING SystemVerification]: missing optional cgroups: hugetlb
error execution phase preflight: [preflight] Some fatal errors occurred:
        [ERROR FileContent--proc-sys-net-bridge-bridge-nf-call-iptables]: /proc/sys/net/bridge/bridge-nf-call-iptables does not exist
[preflight] If you know what you are doing, you can make a check non-fatal with `--ignore-preflight-errors=...`
To see the stack trace of this error execute with --v=5 or higher

```

```shell
modprobe br_netfilter
```

```shell
modprobe: FATAL: Module br_netfilter not found in directory /lib/modules/5.10.66-24-rockchip-gcb09ad15af75
```

opsiff, 2022-10-16:
> embedded board: Radxa Rock 5b system image: [rock-5b-ubuntu-focal-server-arm64-20220930-1451-gpt.img.xz](https://github.com/radxa/debos-radxa/releases/tag/20220930-1224) ubuntu server version: [...]

Can you install this kernel and give me a feedback that it works?

https://github.com/opsiff/linux-rk3588/releases/tag/0.1-1

jikunchong, 2022-10-16:
install success.  but calico components init error.
I'm not sure if this is relevant.

```shell
uname -a
Linux rock-5b 5.10.66-rk3588 #525a51f34 SMP Sun Oct 16 06:55:12 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
```

```shell
./kk create cluster --with-kubernetes v1.23.8

..............
23:28:03 CST [ConfigureOSModule] Exec init os script
23:28:04 CST stdout: [rock-5b]
net.ipv4.ip_forward = 1
net.bridge.bridge-nf-call-arptables = 1
net.bridge.bridge-nf-call-ip6tables = 1
net.bridge.bridge-nf-call-iptables = 1
net.ipv4.ip_local_reserved_ports = 30000-32767
vm.max_map_count = 262144
vm.swappiness = 1
fs.inotify.max_user_instances = 524288
kernel.pid_max = 65535
23:28:04 CST success: [rock-5b]

.........

Installation is complete.

Please check the result using the command:

        kubectl get pod -A
```

```shell
kubectl get pod -A
NAMESPACE     NAME                                       READY   STATUS       RESTARTS      AGE
kube-system   calico-kube-controllers-676c86494f-v292k   0/1     Pending      0             2m10s
kube-system   calico-node-7t546                          0/1     Init:Error   4 (58s ago)   2m10s
kube-system   coredns-757cd945b-79vjb                    0/1     Pending      0             2m15s
kube-system   coredns-757cd945b-m8hct                    0/1     Pending      0             2m15s
kube-system   kube-apiserver-rock-5b                     1/1     Running      0             2m26s
kube-system   kube-controller-manager-rock-5b            1/1     Running      0             2m26s
kube-system   kube-proxy-g5gkk                           1/1     Running      0             2m15s
kube-system   kube-scheduler-rock-5b                     1/1     Running      0             2m26s
kube-system   nodelocaldns-l5b6d                         1/1     Running      0             2m15s
```

but, I use https://github.com/ihexon/rock5b_kernel  ,it's  Running.
- add nf_table framework

maybe, it's kubesphere or kubekey bug.

jikunchong, 2022-10-16:
I retry [ihexon/rock5b_kernel](https://github.com/ihexon/rock5b_kernel/releases/tag/v1.0)

```shell
uname -a
Linux rock-5b 5.10.66-gcf277d5eb46e-dirty #4 SMP Thu Oct 13 11:56:55 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
```

and reinstall kubernetes:
```shell
./kk create cluster --with-kubernetes v1.23.8
```

coredns and calico components is running.

```shell
kubectl get pod -A
NAMESPACE     NAME                                       READY   STATUS    RESTARTS   AGE
kube-system   calico-kube-controllers-84897d7cdf-wvdg6   1/1     Running   0          52s
kube-system   calico-node-krfmg                          1/1     Running   0          52s
kube-system   coredns-b7c47bcdc-d84np                    1/1     Running   0          53s
kube-system   coredns-b7c47bcdc-tbrj5                    1/1     Running   0          53s
kube-system   kube-apiserver-rock-5b                     1/1     Running   0          61s
kube-system   kube-controller-manager-rock-5b            1/1     Running   0          61s
kube-system   kube-proxy-pl8jl                           1/1     Running   0          53s
kube-system   kube-scheduler-rock-5b                     1/1     Running   0          61s
kube-system   nodelocaldns-v9jhs                         1/1     Running   0          53s
```

so, please add nf_table framework. thanks.

opsiff, 2022-10-17:
> install success. but calico components init error. I'm not sure if this is relevant.
>
> ```shell
> uname -a
> Linux rock-5b 5.10.66-rk3588 #525a51f34 SMP Sun Oct 16 06:55:12 UTC 2022 aarch64 aarch64 [...]

I update the test kernel with nf_table patchset.
Could you help me test it?

https://github.com/opsiff/linux-rk3588/releases

code:
https://github.com/opsiff/kernel-2/commit/03f45735020cf9b2c53c419d1a758dbcf7ea7da3

jikunchong, 2022-10-17:
calico components init error.

```shell
dpkg -l | grep 'linux-'
ii  binutils-aarch64-linux-gnu                      2.34-6ubuntu1.3                   arm64        GNU binary utilities, for aarch64-linux-gnu target
ii  linux-5.10-rock-5-latest                        5.10.66-24-rockchip               arm64        This package will always depend on linux-image and linux-headers and linux-firmware-image.
ii  linux-base                                      4.5ubuntu3.7                      all          Linux image base package
ri  linux-headers-5.10.66-24-rockchip-gcb09ad15af75 5.10.66-24-rockchip               arm64        Linux kernel headers for 5.10.66-24-rockchip-gcb09ad15af75 on arm64
ii  linux-headers-5.10.66-gcf277d5eb46e-dirty       5.10.66-gcf277d5eb46e-dirty-4     arm64        Linux kernel headers for 5.10.66-gcf277d5eb46e-dirty on arm64
ii  linux-headers-5.10.66-rk3588                    5.10.66-1-525a51f34               arm64        Linux kernel headers for 5.10.66-rk3588 on arm64
ri  linux-image-5.10.66-24-rockchip-gcb09ad15af75   5.10.66-24-rockchip               arm64        Linux kernel, version 5.10.66-24-rockchip-gcb09ad15af75
ii  linux-image-5.10.66-gcf277d5eb46e-dirty         5.10.66-gcf277d5eb46e-dirty-4     arm64        Linux kernel, version 5.10.66-gcf277d5eb46e-dirty
ii  linux-image-5.10.66-rk3588                      5.10.66-1-525a51f34               arm64        Linux kernel, version 5.10.66-rk3588
ii  linux-libc-dev:arm64                            5.10.66-1-525a51f34               arm64        Linux support headers for userspace development
```

```shell
uname -a
Linux rock-5b 5.10.66-rk3588 #525a51f34 SMP Sun Oct 16 19:39:15 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
```

```shell
kubectl get pod -A
NAMESPACE     NAME                                       READY   STATUS                  RESTARTS     AGE
kube-system   calico-kube-controllers-84897d7cdf-pn7p5   0/1     Pending                 0            5m45s
kube-system   calico-node-pnxhs                          0/1     Init:CrashLoopBackOff   5 (2m ago)   5m45s
kube-system   coredns-b7c47bcdc-4zdht                    0/1     Pending                 0            5m49s
kube-system   coredns-b7c47bcdc-vp6gc                    0/1     Pending                 0            5m49s
kube-system   kube-apiserver-rock-5b                     1/1     Running                 0            6m2s
kube-system   kube-controller-manager-rock-5b            1/1     Running                 0            6m3s
kube-system   kube-proxy-jdmqc                           1/1     Running                 0            5m49s
kube-system   kube-scheduler-rock-5b                     1/1     Running                 0            6m2s
kube-system   nodelocaldns-vwx72                         1/1     Running                 0            5m49s
```

```shell
kubectl get events -A
NAMESPACE     LAST SEEN   TYPE      REASON                    OBJECT                                          MESSAGE
default       14m         Normal    NodeHasSufficientMemory   node/rock-5b                                    Node rock-5b status is now: NodeHasSufficientMemory
default       14m         Normal    NodeHasNoDiskPressure     node/rock-5b                                    Node rock-5b status is now: NodeHasNoDiskPressure
default       14m         Normal    NodeHasSufficientPID      node/rock-5b                                    Node rock-5b status is now: NodeHasSufficientPID
default       13m         Normal    Starting                  node/rock-5b                                    Starting kubelet.
default       13m         Normal    NodeHasSufficientMemory   node/rock-5b                                    Node rock-5b status is now: NodeHasSufficientMemory
default       13m         Normal    NodeHasNoDiskPressure     node/rock-5b                                    Node rock-5b status is now: NodeHasNoDiskPressure
default       13m         Normal    NodeHasSufficientPID      node/rock-5b                                    Node rock-5b status is now: NodeHasSufficientPID
default       13m         Normal    NodeAllocatableEnforced   node/rock-5b                                    Updated Node Allocatable limit across pods
default       13m         Normal    RegisteredNode            node/rock-5b                                    Node rock-5b event: Registered Node rock-5b in Controller
default       13m         Normal    Starting                  node/rock-5b                                    
kube-system   16s         Warning   FailedScheduling          pod/calico-kube-controllers-84897d7cdf-pn7p5    0/1 nodes are available: 1 node(s) had taint {node.kubernetes.io/not-ready: }, that the pod didn't tolerate.
kube-system   13m         Warning   FailedCreate              replicaset/calico-kube-controllers-84897d7cdf   Error creating: pods "calico-kube-controllers-84897d7cdf-" is forbidden: error looking up service account kube-system/calico-kube-controllers: serviceaccount "calico-kube-controllers" not found
kube-system   13m         Normal    SuccessfulCreate          replicaset/calico-kube-controllers-84897d7cdf   Created pod: calico-kube-controllers-84897d7cdf-pn7p5
kube-system   13m         Normal    ScalingReplicaSet         deployment/calico-kube-controllers              Scaled up replica set calico-kube-controllers-84897d7cdf to 1
[... cut ...]
kube-system   13m         Normal    Pulled                    pod/nodelocaldns-vwx72                          Container image "kubesphere/k8s-dns-node-cache:1.15.12" already present on machine
kube-system   13m         Normal    Created                   pod/nodelocaldns-vwx72                          Created container node-cache
kube-system   13m         Normal    Started                   pod/nodelocaldns-vwx72                          Started container node-cache
kube-system   13m         Normal    SuccessfulCreate          daemonset/nodelocaldns                          Created pod: nodelocaldns-vwx72

```

opsiff, 2022-10-17:
> calico components init error.
>
> ```shell
> dpkg -l | grep 'linux-'
> ii  binutils-aarch64-linux-gnu                      2.34-6ubuntu1.3                   arm64        GNU binary utilities, for [...]

sorry for a mistake that the image is not update but i am not saw the nf_tables.ko in that deb .

https://github.com/opsiff/linux-rk3588/releases/tag/0.1-2 is the new one that inclue bith br_netfilter and nf_tables set.

the mistake:
https://github.com/opsiff/linux-rk3588/releases/tag/0.1-1 is the older one enable br_netfilter.

jikunchong, 2022-10-17:
same, coredns & calico  can't running.

```shell
uname -a
Linux rock-5b 5.10.66-rk3588 #03f457350 SMP Mon Oct 17 04:47:52 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
```

steps:

```shell
dpkg -r  <old package......5.10.66-1-525a51f34>
dpkg -i <new package......5.10.66-2-03f457350>
reboot
uname -a
./kk delete cluster
./kk create cluster --with-kubernetes v1.23.8
kubectl get pod -A
```

```shell
NAMESPACE     NAME                                       READY   STATUS                  RESTARTS        AGE
kube-system   calico-kube-controllers-84897d7cdf-srvr6   0/1     Pending                 0               14m
kube-system   calico-node-rcfpg                          0/1     Init:CrashLoopBackOff   7 (3m20s ago)   14m
kube-system   coredns-b7c47bcdc-lvjr9                    0/1     Pending                 0               14m
kube-system   coredns-b7c47bcdc-rbg9m                    0/1     Pending                 0               14m
kube-system   kube-apiserver-rock-5b                     1/1     Running                 0               15m
kube-system   kube-controller-manager-rock-5b            1/1     Running                 0               15m
kube-system   kube-proxy-k2zcg                           1/1     Running                 0               14m
kube-system   kube-scheduler-rock-5b                     1/1     Running                 0               15m
kube-system   nodelocaldns-wxxcv                         1/1     Running                 0               14m
```

jikunchong, 2022-10-17:
I retry [ihexon/rock5b_kernel](https://github.com/ihexon/rock5b_kernel/releases/tag/v1.0)

```shell
dpkg -l |grep linux-
ii  binutils-aarch64-linux-gnu                      2.34-6ubuntu1.3                   arm64        GNU binary utilities, for aarch64-linux-gnu target
ii  linux-5.10-rock-5-latest                        5.10.66-24-rockchip               arm64        This package will always depend on linux-image and linux-headers and linux-firmware-image.
ii  linux-base                                      4.5ubuntu3.7                      all          Linux image base package
ri  linux-headers-5.10.66-24-rockchip-gcb09ad15af75 5.10.66-24-rockchip               arm64        Linux kernel headers for 5.10.66-24-rockchip-gcb09ad15af75 on arm64
ii  linux-headers-5.10.66-gcf277d5eb46e-dirty       5.10.66-gcf277d5eb46e-dirty-4     arm64        Linux kernel headers for 5.10.66-gcf277d5eb46e-dirty on arm64
ri  linux-image-5.10.66-24-rockchip-gcb09ad15af75   5.10.66-24-rockchip               arm64        Linux kernel, version 5.10.66-24-rockchip-gcb09ad15af75
ii  linux-image-5.10.66-gcf277d5eb46e-dirty         5.10.66-gcf277d5eb46e-dirty-4     arm64        Linux kernel, version 5.10.66-gcf277d5eb46e-dirty
rc  linux-image-5.10.66-rk3588                      5.10.66-2-03f457350               arm64        Linux kernel, version 5.10.66-rk3588
ii  linux-libc-dev:arm64                            5.10.66-gcf277d5eb46e-dirty-4     arm64        Linux support headers for userspace development
```

steps:

```shell
# libc => headers => image
dpkg -r <new packages......5.10.66-2-03f457350>
dpkg -i <packages......>
reboot
uname -a
./kk delete cluster
./kk create cluster --with-kubernetes v1.23.8
kubectl get pod -A
```

```shell
NAMESPACE     NAME                                       READY   STATUS    RESTARTS   AGE
kube-system   calico-kube-controllers-84897d7cdf-8b5r4   1/1     Running   0          54s
kube-system   calico-node-m8t4d                          1/1     Running   0          54s
kube-system   coredns-b7c47bcdc-hhnfz                    1/1     Running   0          57s
kube-system   coredns-b7c47bcdc-klr62                    1/1     Running   0          57s
kube-system   kube-apiserver-rock-5b                     1/1     Running   0          63s
kube-system   kube-controller-manager-rock-5b            1/1     Running   0          63s
kube-system   kube-proxy-gt5ll                           1/1     Running   0          56s
kube-system   kube-scheduler-rock-5b                     1/1     Running   0          63s
kube-system   nodelocaldns-dkjpq                         1/1     Running   0          56s
```

I doubt it is related to the specific configuration of the two modules.

opsiff, 2022-10-17:
> ```shell
> Init:CrashLoopBackOff
> ```

yeh, the modules are not automatically loaded.
Could you try commands 'modprobe br_netfilter' and 'modprobe nf_tables' ?

jikunchong, 2022-10-17:
lsmod compare:

lsmod-dirty (ihexon):

```shell
Module                  Size  Used by
bpfilter               16384  1
ipt_rpfilter           16384  1
xt_multiport           16384  2
ip_set_hash_ip         32768  1
ip_set_hash_net        36864  3
veth                   24576  0
ipip                   16384  0
tunnel4                16384  1 ipip
ip_tunnel              24576  1 ipip
xt_set                 24576  11
ip_set_hash_ipportnet    36864  2
ip_set_hash_ipportip    32768  4
ip_set_hash_ipport     32768  16
ip_set_bitmap_port     20480  10
dummy                  16384  0
nf_tables             163840  0
iptable_raw            16384  1
ip_set                 40960  7 ip_set_hash_ipportnet,ip_set_bitmap_port,ip_set_hash_ip,xt_set,ip_set_hash_net,ip_set_hash_ipport,ip_set_hash_ipportip
ip_vs_sh               16384  0
ip_vs_wrr              16384  0
ip_vs_rr               16384  4
ip_vs                 139264  10 ip_vs_rr,ip_vs_sh,ip_vs_wrr
xt_CT                  16384  8
xt_tcpudp              16384  11
ip6table_mangle        16384  1
iptable_mangle         16384  1
ip6table_filter        16384  1
ip6table_nat           16384  1
ip6_tables             24576  3 ip6table_filter,ip6table_nat,ip6table_mangle
xt_comment             16384  115
xt_mark                16384  43
xt_conntrack           16384  7
xt_MASQUERADE          16384  4
nf_conntrack_netlink    40960  0
nfnetlink              16384  4 nf_conntrack_netlink,nf_tables,ip_set
xt_addrtype            16384  6
iptable_filter         16384  1
iptable_nat            16384  1
nf_nat                 53248  3 ip6table_nat,iptable_nat,xt_MASQUERADE
fuse                  110592  3
sch_fq_codel           20480  3
overlay               102400  18
nf_conntrack          135168  6 xt_conntrack,nf_nat,nf_conntrack_netlink,xt_CT,xt_MASQUERADE,ip_vs
nf_defrag_ipv6         20480  2 nf_conntrack,ip_vs
nf_defrag_ipv4         16384  1 nf_conntrack
br_netfilter           24576  0
bridge                184320  1 br_netfilter
stp                    16384  1 bridge
llc                    16384  2 bridge,stp
bpf_preload            16384  0
ip_tables              24576  4 iptable_filter,iptable_raw,iptable_nat,iptable_mangle
x_tables               28672  17 ip6table_filter,xt_conntrack,iptable_filter,xt_multiport,xt_tcpudp,xt_addrtype,xt_comment,xt_set,ip6_tables,ipt_rpfilter,xt_CT,iptable_raw,ip_tables,ip6table_mangle,xt_MASQUERADE,iptable_mangle,xt_mark
ipv6                  413696  189 bridge,br_netfilter,ip6table_mangle,ip_vs
r8125                 135168  0
```

---

lsmod-offical (your)

```shell
Module                  Size  Used by
xt_set                 20480  4
ip_set_bitmap_port     20480  10
[... 48 lines cut ...]
gpio_rockchip          20480  16
pinctrl_rockchip       65536  81
phy_rockchip_naneng_combphy    20480  3
spi_rockchip           20480  0
dw_mmc_rockchip        16384  0
spi_rockchip_sfc       20480  0
phy_rockchip_inno_usb2    57344  17
rockchipdrm           413696  1
dw_hdmi                53248  1 rockchipdrm
panel_simple           77824  1 rockchipdrm
analogix_dp            45056  1 rockchipdrm
dw_hdmi_qp             32768  3 dw_hdmi_qp_i2s_audio,dw_hdmi_qp_cec,rockchipdrm
dw_mipi_dsi            20480  1 rockchipdrm
drm_kms_helper        163840  5 dw_mipi_dsi,rockchipdrm,dw_hdmi,analogix_dp,dw_hdmi_qp
cfbfillrect            16384  1 drm_kms_helper
syscopyarea            16384  1 drm_kms_helper
cfbimgblt              16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
cfbcopyarea            16384  1 drm_kms_helper
drm                   380928  8 drm_kms_helper,dw_mipi_dsi,panel_simple,rockchipdrm,dw_hdmi,analogix_dp,dw_hdmi_qp
drm_panel_orientation_quirks    20480  1 drm
rockchip_cpuinfo       16384  5 rockchip_thermal,pinctrl_rockchip,rockchipdrm,dw_mmc_rockchip,nvmem_rockchip_otp
```

opsiff, 2022-10-17:
need open:
ip6_udp_tunnel
ip_set_hash_ip
ip_set_hash_net
nf_conntrack_ipv4
nf_nat_ipv4
nf_nat_masquerade_ipv4
udp_tunnel
veth
vxlan
xt_multiport
xt_nat
xt_recent
xt_statistic

ref link:https://www.rancher.co.jp/docs/rke/latest/en/os/

jikunchong, 2022-10-17:
By comparison, the following modules are missing:
```shell
modprobe bpfilter
modprobe bpf_preload
modprobe xt_CT
modprobe
modprobe: FATAL: Module ...... not found in directory /lib/modules/5.10.66-rk3588
```

opsiff, 2022-10-17:
> By comparison, the following modules are missing:
> 
> ```shell
> modprobe bpfilter
> modprobe bpf_preload
> modprobe xt_CT
> modprobe nf_nat_ipv4
> modprobe nf_nat_masquerade_ipv4
> modprobe: FATAL: Module ...... not found in directory /lib/modules/5.10.66-rk3588
> ```

bpfilter nf_nat_ipv4 nf_nat_masquerade_ipv4 had been opened before,so I open the config 'NETFILTER_XT_TARGET_CT' 

link:https://github.com/opsiff/linux-rk3588/releases/tag/0.1-3

jikunchong, 2022-10-18:
same, coredns & calico can't running.
I checked, only xt_CT is available.

```shell
modprobe bpfilter
modprobe bpf_preload
modprobe nf_nat_ipv4
modprobe nf_nat_masquerade_ipv4
modprobe: FATAL: Module ...... not found in directory /lib/modules/5.10.66-rk3588
```

opsiff, 2022-10-18:
I believe that as we use cross-compile environment,if we open CONFIG_BPF_PRELOAD ,we need to cross-compile libelf and so on lib with cross-compile before we compile kernel finally or We download with arm deb repo... very complex.

opsiff, 2022-10-18:
> can't running. I checked, only xt_CT is available. but,
> 
> ```shell
> uname -a
> Linux rock-5b 5.10.66-rk3588 #e9b410e47 SMP Mon Oct 17 17:11:48 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
> ```
> 
> ```shell
> # Cilium network components may require
> modprobe bpfilter
> modprobe: FATAL: Module bpfilter not found in directory /lib/modules/5.10.66-rk3588
> modprobe bpf_preload
> modprobe: FATAL: Module bpf_preload not found in directory /lib/modules/5.10.66-rk3588
> # 
> modprobe nf_nat_ipv4
> modprobe: FATAL: Module nf_nat_ipv4 not found in directory /lib/modules/5.10.66-rk3588
> modprobe nf_nat_masquerade_ipv4
> modprobe: FATAL: Module nf_nat_masquerade_ipv4 not found in directory /lib/modules/5.10.66-rk3588
> ```
> 
> Please update it again, thank you. ♡♡♡♡♡♡ @opsiff

With @CodeChenL Help,We found that We don not need BPF_PRELOAD just now.

I hope that kubernetes will run finally with the following kernel now.

https://github.com/opsiff/linux-rockpi5-rk3588/releases/tag/0.1-6

BRs

jikunchong, 2022-10-19:
can't running. 

```shell
uname -a
Linux rock-5b 5.10.66-rk3588 #06c1cc1b9 SMP Tue Oct 18 18:25:35 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
```

```shell
modprobe nf_nat_ipv4
modprobe: FATAL: Module nf_nat_ipv4 not found in directory /lib/modules/5.10.66-rk3588
modprobe nf_nat_masquerade_ipv4
modprobe: FATAL: Module nf_nat_masquerade_ipv4 not found in directory /lib/modules/5.10.66-rk3588
```

---
I refer to the configuration of ihexon, and the main remaining problem should be the NF_ NAT configuration. 
Please confirm whether the corresponding configuration is open again. Thanks.

```
# Add nf_tables framework and bridge_netfilter which needed by docker
CONFIG_NF_CONNTRACK_BRIDGE=m
CONFIG_BRIDGE_NF_EBTABLES=m
CONFIG_NF_NAT_SNMP_BASIC=m
CONFIG_NF_NAT_PPTP=m
CONFIG_NF_NAT_H323=m
CONFIG_IP_NF_IPTABLES=m
CONFIG_IP_NF_MATCH_AH=m
CONFIG_IP_NF_MATCH_ECN=m
CONFIG_IP_NF_MATCH_RPFILTER=m
CONFIG_IP_NF_MATCH_TTL=m
CONFIG_IP_NF_FILTER=m
CONFIG_IP_NF_TARGET_REJECT=m
CONFIG_IP_NF_TARGET_SYNPROXY=m
CONFIG_IP_NF_NAT=m
CONFIG_IP_NF_TARGET_MASQUERADE=m
CONFIG_IP_NF_TARGET_NETMAP=m
CONFIG_IP_NF_TARGET_REDIRECT=m
CONFIG_IP_NF_MANGLE=m
CONFIG_IP_NF_TARGET_CLUSTERIP=m
CONFIG_IP_NF_TARGET_ECN=m
CONFIG_IP_NF_TARGET_TTL=m
CONFIG_IP_NF_RAW=m

# [option] Open Security options to support eBPF trace code, Cilium network components may require
CONFIG_KEYS=y
CONFIG_TRUSTED_KEYS=m
CONFIG_ENCRYPTED_KEYS=m
CONFIG_KEY_DH_OPERATIONS=y
CONFIG_SECURITY_DMESG_RESTRICT=y
CONFIG_SECURITY=y
CONFIG_SECURITYFS=y
CONFIG_SECURITY_NETWORK=y
CONFIG_SECURITY_NETWORK_XFRM=y
CONFIG_SECURITY_PATH=y
CONFIG_HAVE_HARDENED_USERCOPY_ALLOCATOR=y
```

> refs: https://github.com/ihexon/rock5b_kernel#readme

opsiff, 2022-10-19:
> can't running.
>
> ```shell
> uname -a
> Linux rock-5b 5.10.66-rk3588 #06c1cc1b9 SMP Tue Oct 18 18:25:35 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
> ```
>
> ```shell
> modprobe nf_nat_ipv4
> modprobe: FATAL: [...]

nf_nat_ipv4 and nf_nat_masquerade_ipv4 are be renamed as 'nf_nat_ipv4 'and 'nf_nat_masquerade_ipv4' now
as link:https://github.com/k3s-io/k3s/pull/2358

before my first commit they are set by default
https://github.com/opsiff/kernel-2/blob/03f45735020cf9b2c53c419d1a758dbcf7ea7da3/arch/arm64/configs/rockchip_linux_defconfig#L267

opsiff, 2022-10-19:
I think that we need more logs , and less guess.

jikunchong, 2022-10-27:
can't running.

Forgive me for not knowing much about the kernel. However, the support for Docker&Kubernetes is a typical scenario of the rk3588. I hope Rockchip can have special tests and verifications on the kernel to meet our actual needs. Thank you~

opsiff, 2022-10-29:
> can't running.
> 
> Forgive me for not knowing much about the kernel. However, the support for Docker&Kubernetes is a typical scenario of the rk3588. I hope Rockchip can have special tests and verifications on the kernel to meet our actual needs. Thank you~

sorry for that I have notime to test in My rock pi 5b ,I test kubekey installer with the following kernel,and 'kubectl get pod -A' show all in running.

jikunchong, 2022-11-01:
Running!
Thank you for your support. 

[image]

closed 2022-11-01 by jikunchong