rockchip-linux/kernel #274: develop-5.10 内核在RK3399上 开源驱动 使用wayland模式 屏幕闪烁。

IamHuskar, 2022-10-24:
由于4.4内核版本较低，想在rk3399上用develop-5.10内核 用开源驱动+wayland 提高性能。但是发现有闪烁。但是5.4的主线内核没有这个问题。能否修复一下这个屏幕闪烁的问题呢

IamHuskar, 2022-10-24:
更新一下 测试结果。直接用主线的 5.10.110的drm 也可以正常启动

用develop-5.10的drm  内部会报错。
采用的内核 commit id : a4423c56f430bfc7c75dbb68d4a9a1a0ef62b371
采用的rootfs :  ubuntu22.04 base  +  palsma-desktop +wayland

报错信息有两种情况：
第一种
`[   34.876612] Workqueue: events_unbound commit_work
[   34.876639] pstate: 40000005 (nZcv daif -PAN -UAO -TCO BTYPE=--)
[   34.876659] pc : vop_crtc_load_lut+0x790/0x898
[   34.876677] lr : vop_crtc_load_lut+0x3c/0x898
[   34.876689] sp : ffffffc01337bbe0
[   34.876703] x29: ffffffc01337bbe0 x28: 0000000000000465 
[   34.876736] x27: 0000000000000780 x26: 0000000000000438 
[   34.876759] x25: ffffff8004025818 x24: ffffff8004025800 
[   34.876784] x23: 0000000000000438 x22: ffffff80041c8800 
[   34.876810] x21: ffffff800437b800 x20: ffffff8004025800 
[   34.876834] x19: ffffff800a390080 x18: 0000000000000000 
[   34.876860] x17: 0000000000000000 x16: 0000000000000000 
[   34.876884] x15: 0000000000000000 x14: 0000000000000000 
[   34.876907] x13: 0048000000000465 x12: ffffffffffffffff 
[   34.876931] x11: 0000000000000040 x10: ffffffc011290860 
[   34.876954] x9 : 0000000000c00840 x8 : ffffff8004daa058 
[   34.876977] x7 : 0000000080008001 x6 : ffffff800a389080 
[   34.877001] x5 : 00000000000000ff x4 : 0000000000000100 
[   34.877024] x3 : 0000000000000100 x2 : 00000000000000ff 
[   34.877046] x1 : 00000000000000ff x0 : 0000000000000000 
[   34.877082] Call trace:
[   34.877106]  vop_crtc_load_lut+0x790/0x898
[   34.877123]  vop_crtc_atomic_flush+0x674/0x21c8
[   34.877156]  drm_atomic_helper_commit_planes+0x144/0x1e8
[   34.877177]  rockchip_drm_atomic_helper_commit_tail_rpm+0xf0/0x150
[   34.877190]  commit_tail+0xa4/0x170
[   34.877204]  commit_work+0x20/0x30
[   34.877235]  process_one_work+0x204/0x4b0
[   34.877252]  worker_thread+0x178/0x530
[   34.877272]  kthread+0xfc/0x128
[   34.877296]  ret_from_fork+0x10/0x1c`

第二种
[image]

woqidaideshi, 2023-02-09:
你好，你测试的时候 wifi 正常吗？