rockchip-linux/kernel #287: RK3568 使用develop-5.10-rt53分支，使能pcie后报ITS queue timeout错误

Mleaf, 2023-01-31:
[  103.275476] pcieport 0000:00:00.0: PME: Signaling with IRQ 114
[  103.275815] sysfs: cannot create duplicate filename '/devices/platform/3c0000000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/config'
[  103.275825] CPU: 2 PID: 115 Comm: rk-pcie Not tainted 5.10.110-rt53 #1
[  103.275832] Hardware name: Mrkaio M68S (DT)
[  103.275838] Call trace:
[  103.275842]  dump_backtrace+0x0/0x1b4
[  103.275857]  show_stack+0x24/0x30
[  103.275863]  dump_stack_lvl+0xc8/0xf8
[  103.275878]  dump_stack+0x18/0x34
[  103.275882]  sysfs_warn_dup+0x70/0x8c
[  103.275893]  sysfs_add_file_mode_ns+0xd4/0x174
[  103.275898]  sysfs_create_bin_file+0x90/0xa4
[  103.275902]  pci_create_sysfs_dev_files+0x5c/0x22c
[  103.275914]  pci_bus_add_device+0x3c/0xa8
[  103.275922]  pci_bus_add_devices+0x64/0x88
[  103.275926]  pci_bus_add_devices+0x80/0x88
[  103.275930]  pci_host_probe+0x7c/0xb0
[  103.275936]  dw_pcie_host_init+0x248/0x3d4
[  103.275945]  rk_pcie_really_probe+0x644/0xc4c
[  103.275951]  kthread+0x120/0x130
[  103.275960]  ret_from_fork+0x10/0x30
[  103.276286] r8125 2.5Gigabit Ethernet driver 9.011.00-NAPI-RSS loaded
[  103.276438] r8125 0000:01:00.0: enabling device (0000 -> 0003)
[  103.276561] pwm-backlight backlight-0: supply power not found, using dummy regulator
[  103.278143] ITS queue timeout (9088 513)
[  103.278150] ITS cmd its_build_mapti_cmd failed
[  104.738308] ITS queue timeout (9120 513)
[  104.738317] ITS cmd its_build_mapd_cmd failed
[  104.739676] ITS queue timeout (9184 513)
[  104.739682] ITS cmd its_build_mapti_cmd failed
[  106.200644] ITS queue timeout (9248 513)
[  106.200652] ITS cmd its_build_mapti_cmd failed
[  106.201018] ITS queue timeout (9312 513)
[  106.201022] ITS cmd its_build_mapti_cmd failed
[  107.662011] ITS queue timeout (9376 513)
[  107.662017] ITS cmd its_build_mapti_cmd failed
[  107.662348] ITS queue timeout (9440 513)
[  107.662353] ITS cmd its_build_mapti_cmd failed
[  109.123369] ITS queue timeout (9504 513)
[  109.123374] ITS cmd its_build_mapti_cmd failed
[  109.123688] ITS queue timeout (9568 513)
[  109.123692] ITS cmd its_build_mapti_cmd failed
[  110.584705] ITS queue timeout (9632 513)
[  110.584710] ITS cmd its_build_mapti_cmd failed
[  110.585006] ITS queue timeout (9696 513)
[  110.585011] ITS cmd its_build_mapti_cmd failed
[  112.046062] ITS queue timeout (9760 513)

ling020feng, 2023-06-06:
有解决方法吗？问题原因是什么

flylngPenguin, 2023-06-06:
> 有解决方法吗？问题原因是什么

5.10-rt这边解决了，同步了openwrt那边内核加了实时补丁后的gic/irq后，解决了。
不过这个内核还有其他问题，就没用了。
现在直接用的openwrt(5.15.107)的，比较稳定，自己手动合了一个实时补丁。

stoneForRock, 2023-06-20:
> > 有解决方法吗？问题原因是什么
> 
> 5.10-rt这边解决了，同步了openwrt那边内核加了实时补丁后的gic/irq后，解决了。 不过这个内核还有其他问题，就没用了。 现在直接用的openwrt(5.15.107)的，比较稳定，自己手动合了一个实时补丁。

你好，请问你的实时补丁是哪里下载的呢

flylngPenguin, 2023-06-20:
> > > 有解决方法吗？问题原因是什么
> > 
> > 
> > 5.10-rt这边解决了，同步了openwrt那边内核加了实时补丁后的gic/irq后，解决了。 不过这个内核还有其他问题，就没用了。 现在直接用的openwrt(5.15.107)的，比较稳定，自己手动合了一个实时补丁。
> 
> 你好，请问你的实时补丁是哪里下载的呢

去官方(https://mirrors.edge.kernel.org/pub/linux/kernel/)下载一个对应版本的内核, 并打上实时补丁(https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/)。然后相互比较, 手动合并

shicheng-hou, 2024-11-07:
> > > > > 有解决方法吗？问题原因是什么
> > > > 
> > > > 
> > > > 5.10-rt这边解决了，同步了openwrt那边内核加了实时补丁后的gic/irq后，解决了。 不过这个内核还有其他问题，就没用了。 现在直接用的openwrt(5.15.107)的，比较稳定，自己手动合了一个实时补丁。
> > > 
> > > 
> > > 你好，请问你的实时补丁是哪里下载的呢
> > 
> > 
> > 去官方( https://mirrors.edge.kernel.org/pub/linux/kernel )下载一个对应版本的内核, 并打上实时补丁( https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt )。然后相互比较, 手动合并
> 
> 好的 谢谢 感谢指点

你好，后续你打补丁试了吗？解决这个问题了吗

flylngPenguin, 2024-11-07:
> 你好，后续你打补丁试了吗？解决这个问题了吗

早就解决了，现在主线都支持3568了

shicheng-hou, 2024-11-07:
我这边情况是这样的：我没有用主线内核，我是openharmony V4.1版本，内核是5.10.184，跟当时您遇到的问题一样，使能两路PCIE后，就开始报ITS queue timeout的错误，所以请问您如果要解决的话，需要打那些patch？您那边还有的话，麻烦发我一份。非常感谢！

安静
***@***.***

flylngPenguin, 2024-11-07:
> 我这边情况是这样的：我没有用主线内核，我是openharmony V4.1版本，内核是5.10.184，跟当时您遇到的问题一样，使能两路PCIE后，就开始报ITS queue timeout的错误，所以请问您如果要解决的话，需要打那些patch？您那边还有的话，麻烦发我一份。非常感谢！ 安静 ***@***.***  
> […](#)

参考以下这里的提交(同级目录下其他相关的patch), 之前在5.10上解决过, 不过记得好像不稳定
https://github.com/coolsnowwolf/lede/blob/44cccd68f53775121af1c37fd73677b55aff27ee/target/linux/rockchip/patches-5.15/110-arm64-rk3568-update-gicv3-its-and-pci-msi-map.patch

shicheng-hou, 2024-11-07:
好的，收到，谢谢！

安静
***@***.***

  

  
 https://github.com/coolsnowwolf/lede/blob/44cccd68f53775121af1c37fd73677b55aff27ee/target/linux/rockchip/patches-5.15/110-arm64-rk3568-update-gicv3-its-and-pci-msi-map.patch