rockchip-linux/kernel #316: rk3588在docker内使用opencl报错

d597797974, 2024-01-22:
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded

开发板环境opencl是可以使用的，docker内部不能使用
docker run -it --name test --device=/dev/dri -v /tmp/.X11-unix:/tmp/.X11-unix -v /usr/lib/librknnrt.so:/usr/lib/librknnrt.so -e DISPLAY=:0.0
我是按照下面帖子的方法操作的，但是报错
https://clehaxze.tw/gemlog/2023/06-17-setting-up-opencl-on-rk3588-using-libmali.gmi

root@508b149c2de8:/lib# clinfo
Number of platforms                               1
  Platform Name                                   ARM Platform
  Platform Vendor                                 ARM
  Platform Version                                OpenCL 2.1 v1.g6p0-01eac0.2819f9d4dbe0b5a2f89c835d8484f9cd
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_3d_image_writes cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_fp16 cl_khr_icd cl_khr_egl_image cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_subgroups cl_khr_subgroup_extended_types cl_khr_subgroup_non_uniform_vote cl_khr_subgroup_ballot cl_khr_il_program cl_khr_priority_hints cl_khr_create_command_queue cl_khr_spirv_no_integer_wrap_decoration cl_khr_extended_versioning cl_khr_device_uuid cl_arm_core_id cl_arm_printf cl_arm_non_uniform_work_group_size cl_arm_import_memory cl_arm_import_memory_dma_buf cl_arm_import_memory_host cl_arm_integer_dot_product_int8 cl_arm_integer_dot_product_accumulate_int8 cl_arm_integer_dot_product_accumulate_saturate_int8 cl_arm_scheduling_controls cl_arm_controlled_kernel_termination cl_ext_cxx_for_opencl
  Platform Host timer resolution                  1ns
  Platform Extensions function suffix             ARM

  Platform Name                                   ARM Platform
Number of devices                                 1
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '7'.
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
  Device Name                                     <printDeviceInfo:0: get CL_DEVICE_NAME size : error -6>
  Device Vendor                                   ARM
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '7'.
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
  Device Vendor ID                                <printDeviceInfo:2: get CL_DEVICE_VENDOR_ID : error -6>
  Device Version                                  OpenCL 2.1 v1.g6p0-01eac0.2819f9d4dbe0b5a2f89c835d8484f9cd
  Driver Version                                  2.1
  Device OpenCL C Version                         OpenCL C 2.0 v1.g6p0-01eac0.2819f9d4dbe0b5a2f89c835d8484f9cd
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '7'.
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
  Max compute units                               <printDeviceInfo:17: get CL_DEVICE_MAX_COMPUTE_UNITS : error -6>
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '7'.
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
  Max clock frequency                             <printDeviceInfo:21: get CL_DEVICE_MAX_CLOCK_FREQUENCY : error -6>
  Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     None
    Supported affinity domains                    (n/a)
  Max work item dimensions                        3
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '7'.
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
  Max work item sizes                             <printDeviceInfo:36: get number of CL_DEVICE_MAX_WORK_ITEM_SIZES : error -6>
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '7'.
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
  Max work group size                             <printDeviceInfo:37: get CL_DEVICE_MAX_WORK_GROUP_SIZE : error -6>
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '7'.
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
  Preferred work group size multiple              <getWGsizes:1171: create context : error -6>
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '7'.
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
  Max sub-groups per work group                   <printDeviceInfo:43: get CL_DEVICE_MAX_NUM_SUB_GROUPS : error -6>
  Preferred / native vector sizes                 
    char                                                16 / 4       
    short                                                8 / 2       
    int                                                  4 / 1       
    long                                                 2 / 1       
    half                                                 8 / 2        (cl_khr_fp16)
    float                                                4 / 1       
    double                                               0 / 0        (n/a)
  Half-precision Floating-point support           (cl_khr_fp16)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
  Single-precision Floating-point support         (core)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  No
  Double-precision Floating-point support         (n/a)
  Address bits                                    64, Little-Endian
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '7'.
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
  Global memory size                              <printDeviceInfo:77: get CL_DEVICE_GLOBAL_MEM_SIZE : error -6>
  Error Correction support                        No
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '7'.
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
  Max memory allocation                           <printDeviceInfo:83: get CL_DEVICE_MAX_MEM_ALLOC_SIZE : error -6>
  Unified memory for Host and Device              Yes
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '7'.
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
Segmentation fault (core dumped)

nyanmisaka, 2024-01-24:
Mali GPU是通过`/dev/mali0`访问的

d597797974, 2024-01-25:
已经解决了，谢谢

closed 2024-01-25 by d597797974

wohaiaini, 2025-03-01:
> Mali GPU是通过`/dev/mali0`访问的

也需要libmali.so才行吧

nyanmisaka, 2025-03-01:
https://github.com/JeffyCN/mirrors/tree/libmali/lib/aarch64-linux-gnu

wohaiaini, 2025-03-03:
> https://github.com/JeffyCN/mirrors/tree/libmali/lib/aarch64-linux-gnu

你好再请教下，下载了3个G52对应的库试了下，报错找不到 opencl platform，这个是啥意思？

rk3576 mali-G52 环境 试了 libmali-bifrost-g52-g2p0-gbm.so  libmali-bifrost-g52-g13p0-dummy-gbm.so  libmali-bifrost-g52-g24p0-gbm.so
是这个几个都不满足吗？还是有啥操作方式需要注意？谢谢