rockchip-linux/kernel #330: IMX708 Driver

NIKovachev, 2024-06-11:
Hello,
can we get any help on porting IMX708 driver to this kernel? 

Here is the actual driver: https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/media/i2c/imx708.c

shingoxy, 2024-06-28:
I could try, but could you link the camera module? Then I could test it after porting the driver.

NIKovachev, 2024-06-28:
much appreciated!

here it is:
https://datasheets.raspberrypi.com/camera/camera-module-3-product-brief.pdf

shingoxy, 2024-06-29:
Sorry, I mean do you have the   .c file for imx708 from raspberrypi? If so I could have a try. Because I need the file to check the sensor settings.

shingoxy, 2024-07-09:
Sorry for the delay, I checked and found that it was supported by https://joshua-riek.github.io/ubuntu-rockchip-download/
please take a look~

NIKovachev, 2024-07-09:
yes I've seen that but there are numerous issues with the 3A server and the iqfiles. Also the driver is very limited. can you have a look?

shingoxy, 2024-07-10:
> yes I've seen that but there are numerous issues with the 3A server and the iqfiles. Also the driver is very limited. can you have a look?

Could you find the driver from Joshua-riek? Thanks~