rockchip-linux/kernel #349: USB3 Vision Camera Not Streaming on RK3588 (DWC3 USB3 Controller Issue)

openedev, 2025-06-06:
We are encountering a compatibility issue with USB3 Vision cameras on RK3588-based boards. The same camera works flawlessly on an x86_64 AMD laptop but fails to stream or even get detected reliably on RK3588.

Hardware Setup:
```
Board: RK3588 (Edgeble-6A), ROCK-5B, ITX-RK3588 Firefly
Camera: Cypress-based USB3 Vision camera
VendorID:ProductID = 04b4:d003
Model: MV-518UC85-GS
Port: USB3.0 port (verified SuperSpeed in DT)
Cable: USB 3.0 certified (0.5m to 2m tested)
Kernel: 5.10.110, 6.1.115, Mainline Linux
```

Host System (Works):
```
Laptop: AMD Ryzen Cezanne (x86_64)
USB Controller: AMD USB 3.1 xHCI
OS: Ubuntu 22.04
Kernel: 6.11
Software: aravis and gstreamer pipelines work reliably.
```

X86_64 Host: working
```
gst-launch-1.0 -v   aravissrc do-timestamp=true !   video/x-bayer,format=gbrg,width=2448,height=2048 !   bayer2rgb !   videoconvert n-threads=4 !   fpsdisplaysink sync=false text-overlay=true
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstAutoVideoSink:fps-display-video_sink/GstXvImageSink:fps-display-video_sink-actual-sink-xvimage: sync = false
/GstPipeline:pipeline0/GstAravis:aravis0: camera = (null)
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstAravis:aravis0: stream = (null)
/GstPipeline:pipeline0/GstAravis:aravis0.GstPad:src: caps = video/x-bayer, format=(string)gbrg, width=(int)2448, height=(int)2048, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-bayer, format=(string)gbrg, width=(int)2448, height=(int)2048, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstBayer2RGB:bayer2rgb0.GstPad:src: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)20/1, format=(string)RGBx
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:src: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)20/1, format=(string)YV12
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)20/1, format=(string)YV12
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstTextOverlay:fps-display-text-overlay.GstPad:src: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)20/1, format=(string)YV12
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstAutoVideoSink:fps-display-video_sink.GstGhostPad:sink.GstProxyPad:proxypad1: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)20/1, format=(string)YV12
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstAutoVideoSink:fps-display-video_sink/GstXvImageSink:fps-display-video_sink-actual-sink-xvimage.GstPad:sink: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)20/1, format=(string)YV12
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstAutoVideoSink:fps-display-video_sink.GstGhostPad:sink: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)20/1, format=(string)YV12
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstTextOverlay:fps-display-text-overlay.GstPad:video_sink: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)20/1, format=(string)YV12
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)20/1, format=(string)YV12
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:sink: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)20/1, format=(string)RGBx
/GstPipeline:pipeline0/GstBayer2RGB:bayer2rgb0.GstPad:sink: caps = video/x-bayer, format=(string)gbrg, width=(int)2448, height=(int)2048, framerate=(fraction)20/1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-bayer, format=(string)gbrg, width=(int)2448, height=(int)2048, framerate=(fraction)20/1
Redistribute latency...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstAutoVideoSink:fps-display-video_sink/GstXvImageSink:fps-display-video_sink-actual-sink-xvimage: sync = false
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstTextOverlay:fps-display-text-overlay: text = rendered: 12, dropped: 0, current: 22.19, average: 22.19
```

RK3588 failing:
```
gst-launch-1.0 -v aravissrc do-timestamp=true ! video/x-bayer,format=gbrg,width=2448,height=2048 ! bayer2rgb ! videoconvert ! fpsdisplaysink sync=false text-overlay=true video-sink="kmssink"
Setting pipeline to PAUSED ...
[19:06:27.334] ð interface> Found 1 USB3Vision device (among 11 USB devices)
[19:06:27.356] ð device> [UvDevice::new] GUID    = 04B400000209
[19:06:27.358] ð device> [UvDevice::new] USB 3.0
[19:06:27.358] ð device> [UvDevice::new] Using control endpoint 2, interface 0
[19:06:27.358] ð device> [UvDevice::new] Using data endpoint 1, interface 1
[19:06:27.358] ð device> Get genicam
[19:06:27.358] ð device> MANUFACTURER_NAME =        'Hifly Technology'
[19:06:27.359] ð device> MAX_DEVICE_RESPONSE_TIME = 0x000001f4
[19:06:27.359] ð device> DEVICE_CAPABILITY        = 0x0000000000000a09
[19:06:27.359] ð device> SRBM_ADDRESS =             0x0000000002000000
[19:06:27.359] ð device> MANIFEST_TABLE_ADDRESS =   0x0000000004000000
[19:06:27.360] ð device> U3VCP_CAPABILITY =         0x0000000000000001
[19:06:27.360] ð device> MAX_CMD_TRANSFER =         0x00004000
[19:06:27.360] ð device> MAX_ACK_TRANSFER =         0x00004000
[... 45 lines cut ...]
/GstPipeline:pipeline0/GstBayer2RGB:bayer2rgb0.GstPad:src: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)10/1, format=(string)BGRA
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:src: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)10/1, format=(string)BGRA
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)10/1, format=(string)BGRA
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstTextOverlay:fps-display-text-overlay.GstPad:src: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)10/1, format=(string)BGRA
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstKMSSink:kmssink0.GstPad:sink: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)10/1, format=(string)BGRA
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstTextOverlay:fps-display-text-overlay.GstPad:video_sink: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)10/1, format=(string)BGRA
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)10/1, format=(string)BGRA
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:sink: caps = video/x-raw, width=(int)2448, height=(int)2048, framerate=(fraction)10/1, format=(string)BGRA
/GstPipeline:pipeline0/GstBayer2RGB:bayer2rgb0.GstPad:sink: caps = video/x-bayer, format=(string)gbrg, width=(int)2448, height=(int)2048, framerate=(fraction)10/1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-bayer, format=(string)gbrg, width=(int)2448, height=(int)2048, framerate=(fraction)10/1
0:00:00.0 / 99:99:99.Redistribute latency...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstKMSSink:kmssink0: sync = false
0:00:00.5 / 99:99:99.0:00:00.6 / 99:99:99.[19:06:28.233] ð stream-thread> Payload transfer failed (LIBUSB_TRANSFER_STALL)
[19:06:28.233] ð stream-thread> Trailer transfer failed (LIBUSB_TRANSFER_ERROR)
[19:06:28.233] ð stream-thread> Leader transfer failed (LIBUSB_TRANSFER_ERROR)
[19:06:28.234] ð stream-thread> Payload transfer failed (LIBUSB_TRANSFER_ERROR)
[19:06:28.235] ð stream-thread> Payload transfer failed (LIBUSB_TRANSFER_ERROR)
[19:06:28.235] ð stream-thread> Payload transfer failed (LIBUSB_TRANSFER_ERROR)
[19:06:28.236] ð stream-thread> Payload transfer failed (LIBUSB_TRANSFER_ERROR)
```

Tried setting up lower resolutions, used mpp, rga pipelines. all showing similar issues.

Any help?
Jagan.

SisroCodeke, 2026-10-27:
Do you found any solution for this problem?
I'm facing the same problem (same soc different usb3 camera)

hjtroteclaser, 2026-10-27:
We also encountered problems with USB3 camera.
We were able to fix our issues by disable RAM frequency changes. As it turned out, RAM frequency changes imply retraining and this takes too long so that internal FIFO overruns when USB3 transfers are in progress.

<p>rk3588-rock-5b.dts

``` c
&dmc {
	center-supply = <&vdd_ddr_s0>;
	mem-supply = <&vdd_log_s0>;
	/*
	 * Keep the DDR frequency fixed at the highest OPP. A runtime DMC
	 * frequency change briefly halts memory traffic while the DRAM is
	 * retrained, stalling USB3 DMA for tens of milliseconds. A streaming
	 * USB3 camera overruns its internal FIFO in that window and halts its
	 * stream endpoint, which breaks the stream.
	 */
	auto-freq-en = <0>;
	system-status-level = <
		/*system status         freq level*/
		SYS_STATUS_NORMAL       DMC_FREQ_LEVEL_HIGH
		SYS_STATUS_REBOOT       DMC_FREQ_LEVEL_HIGH
		SYS_STATUS_SUSPEND      DMC_FREQ_LEVEL_LOW
		SYS_STATUS_VIDEO_4K     DMC_FREQ_LEVEL_HIGH
		SYS_STATUS_VIDEO_4K_10B DMC_FREQ_LEVEL_HIGH
		SYS_STATUS_VIDEO_SVEP   DMC_FREQ_LEVEL_HIGH
		SYS_STATUS_BOOST        DMC_FREQ_LEVEL_HIGH
		SYS_STATUS_ISP          DMC_FREQ_LEVEL_HIGH
		SYS_STATUS_PERFORMANCE  DMC_FREQ_LEVEL_HIGH
		SYS_STATUS_DUALVIEW     DMC_FREQ_LEVEL_HIGH
		SYS_STATUS_HDMIRX       DMC_FREQ_LEVEL_HIGH
		SYS_STATUS_DEEP_SUSPEND DMC_FREQ_LEVEL_HIGH
	>;
	status = "okay";
};
```
</p>

<p>rk3588.dtsi

``` c
	usbdrd3_1: usbdrd3_1 {
		compatible = "rockchip,rk3588-dwc3", "rockchip,rk3399-dwc3";
		clocks = <&cru REF_CLK_USB3OTG1>, <&cru SUSPEND_CLK_USB3OTG1>,
			 <&cru ACLK_USB3OTG1>;
		clock-names = "ref", "suspend", "bus";
		#address-cells = <2>;
		#size-cells = <2>;
		ranges;
		status = "disabled";

		usbdrd_dwc3_1: usb@fc400000 {
			compatible = "snps,dwc3";
			reg = <0x0 0xfc400000 0x0 0x400000>;
			interrupts = <GIC_SPI 221 IRQ_TYPE_LEVEL_HIGH>;
			power-domains = <&power RK3588_PD_USB>;
			resets = <&cru SRST_A_USB3OTG1>;
			reset-names = "usb3-otg";
			dr_mode = "host";
			phys = <&u2phy1_otg>, <&usbdp_phy1_u3>;
			phy-names = "usb2-phy", "usb3-phy";
			phy_type = "utmi_wide";
			clocks = <&cru REF_CLK_USB3OTG1>;
			clock-names = "ref";
			snps,dis_enblslpm_quirk;
			snps,dis-u1-entry-quirk;
			snps,dis-u2-entry-quirk;
			snps,dis-u2-freeclk-exists-quirk;
			snps,dis-del-phy-power-chg-quirk;
			snps,dis-tx-ipgap-linecheck-quirk;
			snps,parkmode-disable-hs-quirk;
			snps,parkmode-disable-ss-quirk;
			status = "disabled";
		};
	};
```

SisroCodeke, 2026-11-07:
thanks
it worked!